x-wrt/x-wrt #157: Can't access LuCi since last update: Xiaomi Router R3

Noruas29, 2021-01-31:
I updated my Xiaomi Router R3 with x-wrt-9.0-b202101251550-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin version... and now I can't access to my router with the web interface.

My DHCP server see it:
[image]

I can ping it:
[image]

I can access with SSH:
[image]

But I can't access now with web interface:
[image]

I tried it with Vivaldi, Firefox, Chrome on Desktop, Laptop and Mobile, same results.

I appear with the last update, any suggestion ?

ptpt52, 2021-01-31:
access via wan? or via lan?
default 192.168.15.1

ptpt52, 2021-01-31:
I can confirm: no issue on my miwifi-r3

Noruas29, 2021-01-31:
I tried both, wan and lan... same results.

ptpt52, 2021-02-01:
try reset default config?

Noruas29, 2021-02-01:
Before trying to reset default, I tried to install again the last firmware via sysupgrade in SSH... no problem... and I can access to LuCi.
Cool !
I updated for my language... and it crash again.
I installed again and just keep the minimal french packages:
luci-i18n-aria2-fr luci-i18n-base-fr luci-i18n-commands-fr luci-i18n-ddns-fr luci-i18n-diag-core-fr luci-i18n-firewall-fr luci-i18n-ksmbd-fr luci-i18n-mwan3-fr luci-i18n-nft-qos-fr luci-i18n-openvpn-fr luci-i18n-opkg-fr luci-i18n-p910nd-fr luci-i18n-sqm-fr luci-i18n-upnp-fr luci-i18n-wireguard-fr

My router isn't full in french, but a lots of modules are translate and LuCi is OK... I will invistigate later to understand which package crash the web interface.
Thanks for you advices.

closed 2021-02-01 by Noruas29