x-wrt/x-wrt #160: can't update package for kernel issue

r1riyad, 2021-02-19:
**Describe the bug**
Several package can't be update due to kernel dependency error.
some of the packages are:
+-----+---------------------------+------------------------------+------------------------------+
|   # | Package                   | Current                      | Update                       |
+-----+---------------------------+------------------------------+------------------------------+
|   1 | kmod-cfg80211             | 5.4.95+5.8.18-1-5            | 5.4.99+5.10.16-1-1           |
|   2 | kmod-crypto-aead          | 5.4.95-1                     | 5.4.99-1                     |
|   3 | kmod-crypto-arc4          | 5.4.95-1                     | 5.4.99-1                     |
|   4 | kmod-crypto-ccm           | 5.4.95-1                     | 5.4.99-1                     |
|   5 | kmod-crypto-cmac          | 5.4.95-1                     | 5.4.99-1                     |
and the error is:
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-crypto-ghash:
 * 	kernel (= 5.4.99-1-1378356ad65d12ce4e0407a8f6766044)
 * opkg_install_cmd: Cannot install package kmod-crypto-ghash.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-crypto-hash:
 * 	kernel (= 5.4.99-1-1378356ad65d12ce4e0407a8f6766044)

**To Reproduce**
Steps to reproduce the behavior:
1. check opkg list-upgradable
2. then try to upgrade the packages outdated package
3. for error check there is a custom command opkg list-upgradable | cut -f 1 -d ' ' | xargs opkg upgrade
4. See error

**Expected behavior**

The packages should be  updated as its  from opkg repository.

 - Device: Xiaomi Mi Router 3[R3]
 - Build: x-wrt-9.0-b202102081450

ptpt52, 2021-02-19:
sure it cannot be update you have to rebuild from source
dont try to update the kmod package

r1riyad, 2021-02-19:
> sure it cannot be update you have to rebuild from source
> dont try to update the kmod package
Well kinda thought this.then closing the issue..thanks fpr clarification

closed 2021-02-19 by r1riyad