x-wrt/x-wrt #167: luci software page working with luci-app-opkg from openwrt not with built in version

r1riyad, 2021-05-17:
**Describe the bug**
whenever i install a luci theme there is a option called software where user can manage all the openwrt packages using browser, the link for access https://192.168.* .* /cgi-bin/luci/admin/system/opkg
but the page does not work with xwrt default luci-app-opkg_****.opkg .if i force downgrade this package with the version from here https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/ 
then it works i will add the snapshots below

**To Reproduce**
Steps to reproduce the behavior:
1. install a luci theme
2. goto '192.168.* .* /cgi-bin/luci/admin/system/opkg'
3. it will show blank page

**Expected behavior**
it was expected to show the opkgs that are installed or can be updated check 2nd ss.

**Screenshots**
1st ss - with default 'luci-app-opkg'
[image]

2nd ss - with 'luci-app-opkg' from https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/ (installed by force downgrade)

[image]

ptpt52, 2021-05-17:
not a bug, I disable it.

ptpt52, 2021-05-17:
The target users of x-wrt are not people who use opkg luci web, it makes the interface complicated, and unexpected problems occur after installing other software

r1riyad, 2021-05-17:
Thanks for the clearance. Then i m closing the issue.thank u again

closed 2021-05-17 by r1riyad