x-wrt/x-wrt #171: Netgear r6260 刷进去无法打开192.168.15.1

causlayer, 2021-08-21:
我下载的是x-wrt-9.0-b202108141018-ramips-mt7621-netgear_r6260-squashfs-factory.img，可以刷，然后也可以连接上，但是无法进入192.168.15.1进入到luci进行配置，请问如何处理，谢谢
[image]

ptpt52, 2021-08-21:
找一个U盘，fat32格式，u盘创建一个空白文件，名字叫 x-wrt-ssh.txt

ptpt52, 2021-08-21:
然后插入U 盘，路由器启动，ssh就可以登陆了。

ptpt52, 2021-08-21:
界面无法访问，不知道是不是有什么问题bug

causlayer, 2021-08-21:
请问能看看是什么问题吗？我头一次刷机，不会成为转头吧？

ptpt52, 2021-08-21:
应该就是web界面不工作（可能有问题），你进ssh可以升级刷别的openwrt

ptpt52, 2021-08-21:
可以进入ssh，看看web服务有没有

causlayer, 2021-08-21:
请问刷别的openwrt是什么意思，头一次刷路由器，还不太懂，是刷别的型号吗？

ptpt52, 2021-08-21:
进ssh看看再说
logread 命令，结果发出来

ps auxww | grep nginx

causlayer, 2021-08-21:
我手上没有u盘，请问有其它办法吗？我可以把电脑和路由器做连接，这是我自己手动设置的en10.
ifconfig
lo0: flags=8049<UP,LOOPBACK,RUNNING,MULTICAST> mtu 16384
	options=1203<RXCSUM,TXCSUM,TXSTATUS,SW_TIMESTAMP>
	inet 127.0.0.1 netmask 0xff000000
	inet6 ::1 prefixlen 128
	inet6 fe80::1%lo0 prefixlen 64 scopeid 0x1
	nd6 options=201<PERFORMNUD,DAD>
gif0: flags=8010<POINTOPOINT,MULTICAST> mtu 1280
stf0: flags=0<> mtu 1280
XHC20: flags=0<> mtu 0
XHC1: flags=0<> mtu 0
XHC0: flags=0<> mtu 0
en5: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	ether ac:de:48:00:11:22
	inet6 fe80::aede:48ff:fe00:1122%en5 prefixlen 64 scopeid 0x7
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect
	status: active
en0: flags=8823<UP,BROADCAST,SMART,SIMPLEX,MULTICAST> mtu 1500
	options=400<CHANNEL_IO>
	ether 78:4f:43:8d:b0:47
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect (<unknown type>)
	status: inactive
en3: flags=8963<UP,BROADCAST,SMART,RUNNING,PROMISC,SIMPLEX,MULTICAST> mtu 1500
	options=460<TSO4,TSO6,CHANNEL_IO>
	ether 82:dc:88:41:44:01
	media: autoselect <full-duplex>
	status: inactive
en1: flags=8963<UP,BROADCAST,SMART,RUNNING,PROMISC,SIMPLEX,MULTICAST> mtu 1500
	options=460<TSO4,TSO6,CHANNEL_IO>
	ether 82:dc:88:41:44:00
	media: autoselect <full-duplex>
	status: inactive
en4: flags=8963<UP,BROADCAST,SMART,RUNNING,PROMISC,SIMPLEX,MULTICAST> mtu 1500
	options=460<TSO4,TSO6,CHANNEL_IO>
	ether 82:dc:88:41:44:05
	media: autoselect <full-duplex>
	status: inactive
en2: flags=8963<UP,BROADCAST,SMART,RUNNING,PROMISC,SIMPLEX,MULTICAST> mtu 1500
	options=460<TSO4,TSO6,CHANNEL_IO>
	ether 82:dc:88:41:44:04
	media: autoselect <full-duplex>
	status: inactive
en10: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
	options=6467<RXCSUM,TXCSUM,VLAN_MTU,TSO4,TSO6,CHANNEL_IO,PARTIAL_CSUM,ZEROINVERT_CSUM>
	ether 88:d7:f6:db:bd:8e
	inet6 fe80::4e1:e4dc:ebec:1258%en10 prefixlen 64 secured scopeid 0xd
	inet 192.168.15.106 netmask 0xffffff00 broadcast 192.168.15.255
	nd6 options=201<PERFORMNUD,DAD>
	media: autoselect (1000baseT <full-duplex>)
	status: active

[image]

ptpt52, 2021-08-21:
正常流程是，进入界面，开启ssh登陆，就可以登陆ssh
现在界面无法进入，只能u盘方式开启ssh了
然后去解决界面的问题

causlayer, 2021-08-21:
那只能搁置两天我弄个u盘试试，凭你的经验来看我这个新买的路由器不会成为砖头吧？

ptpt52, 2021-08-21:
还没砖，上面你的信息显示，就是ssh 拒绝登陆而已，说明ssh在运行。
现在的状况就是web界面没有，web服务不正常，这个不知道问题。等进ssh可以分析，可以救回来。

ptpt52, 2021-08-21:
你可以尝试一下，开机正常运行的状态下，摁住reset保持6秒，放开，恢复默认配置一次，看看情况如何。
如果不能解决，搞u盘开ssh吧。

causlayer, 2021-08-21:
试过，不行，就像现在的wifi名称是X-WRT_6C7B，只是每次的wifi名称后四位在变化，也不能恢复到出厂设置。

causlayer, 2021-08-22:
我刚刚找了一个u盘，已经格式化是fat32格式，u盘里创建了一个x-wrt-ssh.txt空白文件，然后重启路由器，wifi 连接到 X-WRT_6C7B，还是不能连接上啊，请问哪里操作错误了吗？
ssh -v -o PreferredAuthentications=password -o PubkeyAuthentication=no root@192.168.15.1
OpenSSH_8.1p1, LibreSSL 2.7.3
debug1: Reading configuration data /Users/slayer/.ssh/config
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 47: Applying options for *
debug1: Connecting to 192.168.15.1 [192.168.15.1] port 22.
debug1: Connection established.
debug1: identity file /Users/slayer/.ssh/id_rsa type 0
debug1: identity file /Users/slayer/.ssh/id_rsa-cert type -1
debug1: identity file /Users/slayer/.ssh/id_dsa type -1
debug1: identity file /Users/slayer/.ssh/id_dsa-cert type -1
debug1: identity file /Users/slayer/.ssh/id_ecdsa type -1
debug1: identity file /Users/slayer/.ssh/id_ecdsa-cert type -1
debug1: identity file /Users/slayer/.ssh/id_ed25519 type -1
debug1: identity file /Users/slayer/.ssh/id_ed25519-cert type -1
debug1: identity file /Users/slayer/.ssh/id_xmss type -1
debug1: identity file /Users/slayer/.ssh/id_xmss-cert type -1
debug1: Local version string SSH-2.0-OpenSSH_8.1
debug1: Remote protocol version 2.0, remote software version dropbear
debug1: no match: dropbear
debug1: Authenticating to 192.168.15.1:22 as 'root'
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: algorithm: curve25519-sha256
debug1: kex: host key algorithm: ssh-ed25519
debug1: kex: server->client cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: kex: client->server cipher: chacha20-poly1305@openssh.com MAC: <implicit> compression: none
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug1: Server host key: ssh-ed25519 SHA256:uNxyj/856QPQ4UV42/VkbtDd0pzd2lhu0pr4nWjik4s
debug1: Host '192.168.15.1' is known and matches the ED25519 host key.
debug1: Found key in /Users/slayer/.ssh/known_hosts:15
debug1: rekey out after 134217728 blocks
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: SSH2_MSG_NEWKEYS received
debug1: rekey in after 134217728 blocks
debug1: Will attempt key: /Users/slayer/.ssh/id_rsa RSA SHA256:og568IeOfn3zTs9obLafOAGT+XSBv+jmnx6j8bJcmrI
debug1: Will attempt key: /Users/slayer/.ssh/id_dsa
debug1: Will attempt key: /Users/slayer/.ssh/id_ecdsa
debug1: Will attempt key: /Users/slayer/.ssh/id_ed25519
debug1: Will attempt key: /Users/slayer/.ssh/id_xmss
debug1: SSH2_MSG_EXT_INFO received
debug1: kex_input_ext_info: server-sig-algs=<ssh-ed25519,rsa-sha2-256,ssh-rsa>
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentications that can continue: publickey
debug1: No more authentication methods to try.
root@192.168.15.1: Permission denied (publickey).

ptpt52, 2021-08-23:
应该是行的，不行就不得而知了。

causlayer, 2021-08-23:
请问x-wrt修改了路由器原厂的CFE了吗？

ptpt52, 2021-08-23:
不会，这个设备，和openwrt官方保持一致，没有任何特别的修改。

causlayer, 2021-08-23:
终于用官方的程序搞定了，我直接刷了netgear最新的固件。
以后不会再出问题了。

closed 2021-08-23 by causlayer