x-wrt/x-wrt #172: TP-LINK WMC181/WTC181能使配吗？

daiaji, 2021-08-21:
https://www.mydigit.cn/thread-263542-2-1.html
运营商定制
ROM和RAM看起来挺大的。
而且是MT7621A方案。
似乎是和CR660X差不多。
https://post.smzdm.com/p/akmg58dk/
TP的WEB页面似乎能够升级固件。
所以TP是怎么验证固件完整性的。
华硕的固件为什么能拿来用？

daiaji, 2021-08-31:
@ptpt52 我有两个，一个暂时拿来用，另外一个我寄给你，适配完了发给我就行。

hhh245645, 2022-08-24:
老哥你的wtc181刷了吗~我也想刷，找了一圈没教程[捂脸]

daiaji, 2022-08-24:
由于迟迟没有固件，所以我拿起了我的K2P。
反正带宽差距不大。

hhh245645, 2022-08-24:
噢噢~

daiaji, 2023-02-24:
@ptpt52 哥，问个适配的事。
https://www.mydigit.cn/thread-303510-1-1.html
我有一个一样的路由器，MT7621DAT的。
但闪存是8脚的，编程器能刷。
https://forum.openwrt.org/t/does-the-mainline-openwrt-support-mt7621da
和这边说的一样，我焊上了TTL针脚但是没输出。
是不是直接把固件从闪存里读出来，然后找个通用的固件混合一下就行？
涉及到破解TP的固件就比较痛苦。
如果从源码编译移植，还要DTS吧。
有通用的吗？
这个路由器貌似没有无线功能。
通用的DTS就能用吧？
Breed能刷吗？
热心黑客把配置加密方法黑出来了。
https://github.com/sta-c0000/tpconf_bin_xml/issues/32
但距离魔改固件获得ROOT SHELL还有有点远。