x-wrt/x-wrt #178: Latest release  X-wrt 9.0_b202109131311 randomly restarts on Xiaomi Miwifi mini

newmem12, 2021-09-14:
Latest release  X-wrt 9.0_b202109131311 randomly restarts on Xiaomi Miwifi mini
with default settings and random restart every few hours!
Thanks for the support

ptpt52, 2021-09-14:
what restart means?
reboot system? or what

newmem12, 2021-09-14:
Thanks ptpt52!
i think it's reboot system because the output of uptime command is "1 min"

ptpt52, 2021-09-14:
Please try this firmware:
```
https://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109131311-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade-v1.bin
```

and if the crash/reboot happens
login ssh and get the crash log in
```
/sys/fs/pstore/
```

newmem12, 2021-09-14:
thanks, i will report the results to you soon

newmem12, 2021-09-14:
Hi ptpt52!
the system just randomly rebooted!
but i found nothing in /sys/fs/pfstore folder, it's empty!
here's log:
root@X-WRT:~# logread -l 100
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
Tue Sep 14 15:52:00 2021 daemon.notice hostapd: ACS: Survey is missing noise floor
[... cut ...]
Tue Sep 14 15:53:19 2021 daemon.notice hostapd: wlan1: AP-STA-CONNECTED fa:87:db:98:88:ee
Tue Sep 14 15:53:19 2021 daemon.info hostapd: wlan1: STA fa:87:db:98:88:ee RADIUS: starting accounting session 0CBC168247D5DD5F
Tue Sep 14 15:58:08 2021 authpriv.notice dropbear[6741]: Password auth succeeded for 'root' from 192.168.15.7:58364
Tue Sep 14 15:58:15 2021 authpriv.notice dropbear[6796]: Password auth succeeded for 'root' from 192.168.15.7:58376
root@X-WRT:~# ^C
root@X-WRT:~# uptime
 16:01:38 up 11 min,  load average: 0.02, 0.15, 0.17
root@X-WRT:~#

ptpt52, 2021-09-14:
post `dmesg` result
and if `/sys/fs/pstore/` empty
it is likely not system crash but (maybe) somewhat hardware power issue

newmem12, 2021-09-14:
hi!
```
root@X-WRT:~# dmesg
[    0.000000] Linux version 5.4.145 (ptpt52@SC-GAME) (gcc version 10.3.0 (OpenWrt GCC 10.3.0 r17500-cf4094acec)) #0 Tue Sep 14 07:46:22 2021
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
[    0.000000] MIPS: machine is Xiaomi MiWiFi Mini
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 256 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=0004c7f0
[    0.000000] Readback ErrCtl register=0004c7f0
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 121644K/131072K available (4965K kernel code, 186K rwdata, 1080K rodata, 1248K init, 204K bss, 9428K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x350/0x51c with crng_init=0
[    0.000000] CPU Clock: 580MHz
[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
[    0.000000] systick: enable autosleep mode
[    0.000000] systick: running - mult: 214748, shift: 32
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
[    0.000009] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
[    0.007599] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.073558] pid_max: default: 32768 minimum: 301
[    0.078189] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.085242] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.094815] devtmpfs: initialized
[    0.102487] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.112030] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.118645] pinctrl core: initialized pinctrl subsystem
[    0.124257] NET: Registered protocol family 16
[    0.391493] PCI host bridge /pcie@10140000 ranges:
[    0.396105]  MEM 0x0000000020000000..0x000000002fffffff
[    0.401156]   IO 0x0000000010160000..0x000000001016ffff
[    0.427934] rt2880_gpio 10000600.gpio: registering 24 gpios
[    0.433336] rt2880_gpio 10000600.gpio: registering 24 irq handlers
[    0.440464] rt2880_gpio 10000638.gpio: registering 16 gpios
[    0.445854] rt2880_gpio 10000638.gpio: registering 16 irq handlers
[    0.452996] rt2880_gpio 10000660.gpio: registering 32 gpios
[    0.458385] rt2880_gpio 10000660.gpio: registering 32 irq handlers
[    0.466079] PCI host bridge to bus 0000:00
[    0.470009] pci_bus 0000:00: root bus resource [mem 0x20000000-0x2fffffff]
[    0.476749] pci_bus 0000:00: root bus resource [io  0x10160000-0x1016ffff]
[    0.483362] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
[    0.489964] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.497697] pci 0000:00:00.0: [1814:0801] type 01 class 0x060400
[    0.503520] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[    0.509566] pci 0000:00:00.0: reg 0x14: [mem 0x20200000-0x2020ffff]
[    0.515739] pci 0000:00:00.0: supports D1
[... 233 lines cut ...]
[   67.310974] ieee80211 phy1: rt2800_iq_search: Info - IQCalibration Done! CH = 1, (gain= 3, phase=3e)
[   67.320317] ieee80211 phy1: rt2800_loft_iq_calibration: Info - TX IQ Calibration Done!
[   67.513259] ieee80211 phy1: rt2800_rxiq_calibration: Info - RXIQ G_imb=-1, Ph_rx=-1
[   67.531354] ieee80211 phy1: rt2800_rxiq_calibration: Info - RXIQ G_imb=-2, Ph_rx=-3
[   67.579812] warning: (zone_netdev_event:345)dev=wlan1 set zone=127 type=0
[   67.586821] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=wlan1, add ingress hook
[   67.746885] br-lan: port 2(wlan1) entered blocking state
[   67.752406] br-lan: port 2(wlan1) entered disabled state
[   67.758336] device wlan1 entered promiscuous mode
[   68.463348] warning: (zone_netdev_event:345)dev=wlan0 set zone=127 type=0
[   68.470310] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=wlan0, add ingress hook
[   68.589023] br-lan: port 3(wlan0) entered blocking state
[   68.594533] br-lan: port 3(wlan0) entered disabled state
[... cut ...]
[   68.610863] br-lan: port 3(wlan0) entered forwarding state
[   68.620659] br-lan: port 3(wlan0) entered disabled state
[   76.963928] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   76.970769] br-lan: port 2(wlan1) entered blocking state
[   76.976192] br-lan: port 2(wlan1) entered forwarding state
[   80.910399] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   80.917203] br-lan: port 3(wlan0) entered blocking state
[   80.922663] br-lan: port 3(wlan0) entered forwarding state
root@X-WRT:~#
```

ptpt52, 2021-09-14:
seems pstore log not working
post `find /proc/device-tree/ | grep ramoops` result

newmem12, 2021-09-14:
root@X-WRT:~# find /proc/device-tree/ | grep ramoops
/proc/device-tree/reserved-memory/ramoops@3f00000
/proc/device-tree/reserved-memory/ramoops@3f00000/compatible
/proc/device-tree/reserved-memory/ramoops@3f00000/record-size
/proc/device-tree/reserved-memory/ramoops@3f00000/console-size
/proc/device-tree/reserved-memory/ramoops@3f00000/ftrace-size
/proc/device-tree/reserved-memory/ramoops@3f00000/reg
/proc/device-tree/reserved-memory/ramoops@3f00000/name
root@X-WRT:~#
``

ptpt52, 2021-09-14:
@newmem12 
hi
Please test this one:
```
http://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109141708-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade.bin
```

this trying to make the pstore log works and if crash again, login ssh, and get the log in `/sys/fs/pstore/`

ptpt52, 2021-09-14:
to verify the pstore working
`dmesg | grep pstore`
get the output

newmem12, 2021-09-14:
Thanks ptpt52!
I will report the results to you soon

newmem12, 2021-09-14:
> to verify the pstore working
> `dmesg | grep pstore`
> get the output

output:
```
root@X-WRT:~# dmesg | grep pstore
[    0.407015] pstore: Registered ramoops as persistent store backend
[    1.048859] pstore: Using crash dump compression: deflate
root@X-WRT:~#

```

newmem12, 2021-09-15:
> post `dmesg` result
> and if `/sys/fs/pstore/` empty
> **it is likely not system crash but (maybe) somewhat hardware power issue**

i think you are right, I changed another adapter and the device is working fine

```
root@X-WRT:~# uptime
 12:48:44 up 16:16,  load average: 0.00, 0.00, 0.00
root@X-WRT:~#

```

newmem12, 2021-09-15:
Hi ptpt52!
The system has just been rebooted
Here's the log:
dmesg-ramoops-0
`Oops#1 Part1
<5>[    0.000000] Linux version 5.4.145 (ptpt52@SC-GAME) (gcc version 10.3.0 (OpenWrt GCC 10.3.0 r17500-cf4094acec)) #0 Tue Sep 14 12:28:44 2021
<6>[    0.000000] Board has DDR2
<6>[    0.000000] Analog PMU set to hw control
<6>[    0.000000] Digital PMU set to hw control
<6>[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
<6>[    0.000000] printk: bootconsole [early0] enabled
<6>[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
<6>[    0.000000] MIPS: machine is Xiaomi MiWiFi Mini
<6>[    0.000000] Initrd not found or empty - disabling initrd
<4>[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
<4>[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
<6>[    0.000000] Zone ranges:
<6>[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Movable zone start for each node
<6>[    0.000000] Early memory node ranges
<6>[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
<7>[    0.000000] On node 0 totalpages: 32768
<7>[    0.000000]   Normal zone: 256 pages used for memmap
<7>[    0.000000]   Normal zone: 0 pages reserved
<7>[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
<7>[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
<7>[    0.000000] pcpu-alloc: [0] 0 
<6>[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
<5>[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
<6>[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
<6>[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
<6>[    0.000000] Writing ErrCtl register=0001af10
<6>[    0.000000] Readback ErrCtl register=0001af10
<6>[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
<6>[    0.000000] Memory: 121644K/131072K available (4981K kernel code, 190K rwdata, 1084K rodata, 1224K init, 204K bss, 9428K reserved, 0K cma-reserved)
<6>[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
<6>[    0.000000] NR_IRQS: 256
<5>[    0.000000] random: get_random_bytes called from start_kernel+0x350/0x51c with crng_init=0
<6>[    0.000000] CPU Clock: 580MHz
<6>[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
<6>[    0.000000] systick: enable autosleep mode
<6>[    0.000000] systick: running - mult: 214748, shift: 32
<6>[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
<6>[    0.000011] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
<6>[    0.007600] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
<6>[    0.073553] pid_max: default: 32768 minimum: 301
<6>[    0.078189] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.085239] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.094812] devtmpfs: initialized
<6>[    0.102504] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
<6>[    0.112047] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
<6>[    0.118662] pinctrl core: initialized pinctrl subsystem
<6>[    0.124275] NET: Registered protocol family 16
<6>[    0.391646] PCI host bridge /pcie@10140000 ranges:
<6>[    0.396262]  MEM 0x0000000020000000..0x000000002fffffff
<6>[    0.401310]   IO 0x0000000010160000..0x000000001016ffff
<6>[    0.407015] pstore: Registered ramoops as persistent store backend
<6>[    0.412954] ramoops: using 0x80000@0x3f00000, ecc: 0
<6>[    0.441905] rt2880_gpio 10000600.gpio: registering 24 gpios
<6>[    0.447307] rt2880_gpio 10000600.gpio: registering 24 irq handlers
<6>[    0.454461] rt2880_gpio 10000638.gpio: registering 16 gpios
<6>[    0.459850] rt2880_gpio 10000638.gpio: registering 16 irq handlers
<6>[    0.467024] rt2880_gpio 10000660.gpio: registering 32 gpios
<6>[    0.472413] rt2880_gpio 10000660.gpio: registering 32 irq handlers
<6>[    0.480106] PCI host bridge to bus 0000:00
<6>[    0.484037] pci_bus 0000:00: root bus resource [mem 0x20000000-0x2fffffff]
<6>[    0.490774] pci_bus 0000:00: root bus resource [io  0x10160000-0x1016ffff]
<6>[    0.497387] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
<6>[    0.503987] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[... 638 lines cut ...]
<4>[25743.644104] Lo    : 2e48f657
<4>[25743.647077] epc   : 861f4670 rt2x00queue_flush_queues+0x51c/0x6d0 [rt2x00lib]
<4>[25743.654363] ra    : 861f4370 rt2x00queue_flush_queues+0x21c/0x6d0 [rt2x00lib]
<4>[25743.661635] Status: 1100f403	KERNEL EXL IE 
<4>[25743.665903] Cause : 40800008 (ExcCode 02)
<4>[25743.669986] BadVA : 00000008
<4>[25743.672919] PrId  : 00019650 (MIPS 24KEc)
<4>[25743.677001] Modules linked in: ksmbd rt2800soc rt2800mmio rt2800lib qcserial pppoe ppp_async option cdc_mbim batman_adv wireguard usb_wwan sierra_net sierra rt2x00soc rt2x00mmio rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport  [... cut ...]
<4>[25744.005491]         8634f048 861f2620 00000001 85dad600 85dad600 87c9a810 00000000 00000000
<4>[25744.014019]         ...
<4>[25744.016517] Call Trace:
<4>[25744.019028] [<861f4670>] rt2x00queue_flush_queues+0x51c/0x6d0 [rt2x00lib]
<4>[25744.025960] Code: 7c420000  14400053  00002825 <92630008> a6030008  92220021  000211c2  1040ff6d  34630008 
<4>[25744.035916] 
<4>[25744.037494] ---[ end trace 1a8a598014c9a841 ]---
<0>[25744.066068] Kernel panic - not syncing: Fatal exception in interrupt
`

ptpt52, 2021-09-15:
Please try: login ssh and `rmmod natflow`
and watch/wait next time crash or not.

newmem12, 2021-09-15:
> Please try: login ssh and `rmmod natflow`
> and watch/wait next time crash or not.

`
root@X-WRT:~# rmmod natflow
root@X-WRT:~# rmmod natflow
module is not loaded
root@X-WRT:~# rmmod natflow
module is not loaded
root@X-WRT:~#`

I'm waiting and will let you know the results soon

newmem12, 2021-09-16:
hi ptpt52!
The system has just been rebooted
`root@X-WRT:~# uptime
 21:11:35 up 38 min,  load average: 0.00, 0.00, 0.00
root@X-WRT:~#
`

ptpt52, 2021-09-16:
and any new crash log?

newmem12, 2021-09-16:
> and any new crash log?

no, there is no new crash log!

ptpt52, 2021-09-16:
post file dmesg-ramoops-0 out
and which version is good?

newmem12, 2021-09-16:
dmesg-ramoops-0
`Oops#1 Part1
<5>[    0.000000] Linux version 5.4.145 (ptpt52@SC-GAME) (gcc version 10.3.0 (OpenWrt GCC 10.3.0 r17500-cf4094acec)) #0 Tue Sep 14 12:28:44 2021
<6>[    0.000000] Board has DDR2
<6>[    0.000000] Analog PMU set to hw control
<6>[    0.000000] Digital PMU set to hw control
<6>[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
<6>[    0.000000] printk: bootconsole [early0] enabled
<6>[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
<6>[    0.000000] MIPS: machine is Xiaomi MiWiFi Mini
<6>[    0.000000] Initrd not found or empty - disabling initrd
<4>[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
<4>[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
<6>[    0.000000] Zone ranges:
<6>[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Movable zone start for each node
<6>[    0.000000] Early memory node ranges
<6>[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
<7>[    0.000000] On node 0 totalpages: 32768
<7>[    0.000000]   Normal zone: 256 pages used for memmap
<7>[    0.000000]   Normal zone: 0 pages reserved
<7>[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
<7>[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
<7>[    0.000000] pcpu-alloc: [0] 0 
<6>[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
<5>[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
<6>[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
<6>[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
<6>[    0.000000] Writing ErrCtl register=0001af10
<6>[    0.000000] Readback ErrCtl register=0001af10
<6>[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
<6>[    0.000000] Memory: 121644K/131072K available (4981K kernel code, 190K rwdata, 1084K rodata, 1224K init, 204K bss, 9428K reserved, 0K cma-reserved)
<6>[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
<6>[    0.000000] NR_IRQS: 256
<5>[    0.000000] random: get_random_bytes called from start_kernel+0x350/0x51c with crng_init=0
<6>[    0.000000] CPU Clock: 580MHz
<6>[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
<6>[    0.000000] systick: enable autosleep mode
<6>[    0.000000] systick: running - mult: 214748, shift: 32
<6>[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
<6>[    0.000011] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
<6>[    0.007600] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
<6>[    0.073541] pid_max: default: 32768 minimum: 301
<6>[    0.078177] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.085227] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.094800] devtmpfs: initialized
<6>[    0.102492] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
<6>[    0.112034] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
<6>[    0.118650] pinctrl core: initialized pinctrl subsystem
<6>[    0.124262] NET: Registered protocol family 16
<6>[    0.391633] PCI host bridge /pcie@10140000 ranges:
<6>[    0.396250]  MEM 0x0000000020000000..0x000000002fffffff
<6>[    0.401297]   IO 0x0000000010160000..0x000000001016ffff
<6>[    0.407267] pstore: Registered ramoops as persistent store backend
<6>[    0.413207] ramoops: using 0x80000@0x3f00000, ecc: 0
<6>[    0.442152] rt2880_gpio 10000600.gpio: registering 24 gpios
<6>[    0.447554] rt2880_gpio 10000600.gpio: registering 24 irq handlers
<6>[    0.454710] rt2880_gpio 10000638.gpio: registering 16 gpios
<6>[    0.460099] rt2880_gpio 10000638.gpio: registering 16 irq handlers
<6>[    0.467272] rt2880_gpio 10000660.gpio: registering 32 gpios
<6>[    0.472662] rt2880_gpio 10000660.gpio: registering 32 irq handlers
<6>[    0.480356] PCI host bridge to bus 0000:00
<6>[    0.484286] pci_bus 0000:00: root bus resource [mem 0x20000000-0x2fffffff]
<6>[    0.491023] pci_bus 0000:00: root bus resource [io  0x10160000-0x1016ffff]
<6>[    0.497636] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
<6>[    0.504236] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
<6>[    0.511962] pci 0000:00:00.0: [1814:0801] type 01 class 0x060400
<6>[    0.517796] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
<6>[    0.523846] pci 0000:00:00.0: reg 0x14: [mem 0x20200000-0x2020ffff]
[... 676 lines cut ...]
<4>[25795.165710] Lo    : 56464c3b
<4>[25795.168682] epc   : 86324670 rt2x00queue_flush_queues+0x51c/0x6d0 [rt2x00lib]
<4>[25795.175966] ra    : 86324370 rt2x00queue_flush_queues+0x21c/0x6d0 [rt2x00lib]
<4>[25795.183237] Status: 1100f403	KERNEL EXL IE 
<4>[25795.187507] Cause : 40800008 (ExcCode 02)
<4>[25795.191589] BadVA : 00000008
<4>[25795.194524] PrId  : 00019650 (MIPS 24KEc)
<4>[25795.198605] Modules linked in: ksmbd rt2800soc rt2800mmio rt2800lib qcserial pppoe ppp_async option cdc_mbim batman_adv wireguard usb_wwan sierra_net sierra rt2x00soc rt2x00mmio rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT huawei_cdc_ncm [... cut ...]
<4>[25795.528601]         86379048 86322620 00001080 00000a20 85f66900 861f5600 00000000 00000000
<4>[25795.537128]         ...
<4>[25795.539625] Call Trace:
<4>[25795.542138] [<86324670>] rt2x00queue_flush_queues+0x51c/0x6d0 [rt2x00lib]
<4>[25795.549068] Code: 7c420000  14400053  00002825 <92630008> a6030008  92220021  000211c2  1040ff6d  34630008 
<4>[25795.559025] 
<4>[25795.560616] ---[ end trace f6b94ccfb8d9cbfc ]---
<0>[25795.590699] Kernel panic - not syncing: Fatal exception in interrupt
`

newmem12, 2021-09-16:
I don't remember exactly which version, just before 2020

ptpt52, 2021-09-16:
```
[18139.815065] br-lan: port 1(eth0.1) entered forwarding state 
<1>[25795.100968] CPU 0 Unable to handle kernel paging request at virtual address 00000008, epc == 86324670, ra ==86324370 
<4>[25795.111858] Oops[#1]: <4>[25795.114182] CPU: 0 PID: 0 Comm: swapper Not tainted 5.4.145 #0 
<4>[25795.120126] $ 0 : 00000000 80750000 00000000 00000000 
<4>[25795.125462] $ 4 : 87d48c5c 00000005 00000080 8776e764 
<4>[25795.130793] $ 8 : 8776e000 87d48c5c 000000d0 86fa4460 
<4>[25795.136125] $12 : 000000d0 00000001 00000000 0003790d 
<4>[25795.141454] $16 : 87c0bdd4 85d7a3c0 86331ec0 00000000 
<4>[25795.146785] $20 : 8776e764 00000000 ffffffbf 00000001 
<4>[25795.152115] $24 : 805f6240 80247174 
<4>[25795.157446] $28 : 805f0000 87c0bd90 86278748 86324370 
<4>[25795.162777] Hi : 0018c219 
<4>[25795.165710] Lo : 56464c3b 
<4>[25795.168682] epc : 86324670 rt2x00queue_flush_queues+0x51c/0x6d0 [rt2x00lib] 
<4>[25795.175966] ra : 86324370 rt2x00queue_flush_queues+0x21c/0x6d0 [rt2x00lib] 
<4>[25795.183237] Status: 1100f403 KERNEL EXL IE <4>[25795.187507] Cause : 40800008 (ExcCode 02) 
<4>[25795.191589] BadVA : 00000008 
<4>[25795.194524] PrId : 00019650 (MIPS 24KEc) <4>[25795.198605] Modules linked in: ksmbd rt2800soc
```

@nbd168  sorry to spam you here
but we get some issue with rt2x00 wifi driver
crash log shows `Unable to handle kernel paging request at virtual address 00000008`
in fucntion `rt2x00queue_flush_queues`

I am not familiar with this wifi driver or related, maybe you could give us some help?
there is no issue in version of 2020 year before, according to @newmem12  report.

ptpt52, 2021-09-16:
in function `rt2x00queue_create_tx_descriptor()`
```
 458         /*
 459          * Determine rate modulation.
 460          */
 461         if (txrate->flags & IEEE80211_TX_RC_GREEN_FIELD)
 462                 txdesc->rate_mode = RATE_MODE_HT_GREENFIELD;
 463         else if (txrate->flags & IEEE80211_TX_RC_MCS)
 464                 txdesc->rate_mode = RATE_MODE_HT_MIX;
 465         else {
 466                 rate = ieee80211_get_tx_rate(rt2x00dev->hw, tx_info);
 467                 hwrate = rt2x00_get_rate(rate->hw_value);
 468                 if (hwrate->flags & DEV_RATE_OFDM)
 469                         txdesc->rate_mode = RATE_MODE_OFDM;
 470                 else
 471                         txdesc->rate_mode = RATE_MODE_CCK;
 472         }
 473 
 474         /*
 475          * Apply TX descriptor handling by components
 476          */
 477         rt2x00crypto_create_tx_descriptor(rt2x00dev, skb, txdesc);
 478         rt2x00queue_create_tx_descriptor_seq(rt2x00dev, skb, txdesc);
 479 
 480         if (rt2x00_has_cap_flag(rt2x00dev, REQUIRE_HT_TX_DESC))
 481                 rt2x00queue_create_tx_descriptor_ht(rt2x00dev, skb, txdesc,
 482                                                    sta, hwrate);
 483         else
 484                 rt2x00queue_create_tx_descriptor_plcp(rt2x00dev, skb, txdesc,
 485                                                       hwrate);
 486 }
```
var `hwrate` is possible to be NULL

ptpt52, 2021-09-16:
okay, Let me try to fix it
@newmem12  Please try test this new firmware:
```
http://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109141708-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade.bin
```

newmem12, 2021-09-16:
> okay, Let me try to fix it
> @newmem12 Please try test this new firmware:
> 
> ```
> http://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109141708-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade.bin
> ```

thanks ptpt52!
I'm testing and will show you the results soon

newmem12, 2021-09-16:
hi ptpt52!
I add more information: Xiaomi R3 has the same problem!
It also rebooted after 8 hours!
So do these 2 devices have the same hardware?

```
root@OpenWrt:~# uptime
 21:30:23 up  8:29,  load average: 0.27, 0.13, 0.04
root@OpenWrt:~# iwi
login as: root

BusyBox v1.34.0 (2021-09-13 05:14:26 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r17651+1-6d256b4
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# amdin
-ash: amdin: not found
root@OpenWrt:~# uptime
 21:30:30 up 1 min,  load average: 0.56, 0.24, 0.08
root@OpenWrt:~#

```

ptpt52, 2021-09-16:
sure, miwifi-r3 has the same wifi 2.4G
also do test on miwifi-r3:
```
http://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109141708-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin
```

newmem12, 2021-09-16:
> sure, miwifi-r3 has the same wifi 2.4G
> also do test on miwifi-r3:
> 
> ```
> http://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109141708-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin
> ```
thanks!
I'm testing and will show you the results of both soon!

newmem12, 2021-09-16:
I'm suspecting my other 3 devices have the same problem!
Thanks to ptpt52 to see if they have the same hardware!
If they have the same hardware, it will be easier for us to find and fix bugs
`https://openwrt.org/toh/hwdata/hiwifi_gee/hiwifi_gee_hc5761`
`https://openwrt.org/toh/hwdata/phicomm/phicomm_k2_psg1218`
`https://openwrt.org/toh/hwdata/lenovo/lenovo_newifi_mini_y1`

ptpt52, 2021-09-16:
they are all mt7620 soc and the same wifi 2.4G

newmem12, 2021-09-16:
> they are all mt7620 soc and the same wifi 2.4G

yes, I have noticed that sometimes when I connect to wifi 2.4 it will rebooted

newmem12, 2021-09-16:
Hi ptpt52, Xiaomi R3 has just been rebooted!
dmesg-ramoops-0
`Oops#1 Part1
<5>[    0.000000] Linux version 5.4.145 (ptpt52@SC-GAME) (gcc version 10.3.0 (OpenWrt GCC 10.3.0 r17500-cf4094acec)) #0 Thu Sep 16 06:11:35 2021
<6>[    0.000000] Board has DDR2
<6>[    0.000000] Analog PMU set to hw control
<6>[    0.000000] Digital PMU set to hw control
<6>[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
<6>[    0.000000] printk: bootconsole [early0] enabled
<6>[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
<6>[    0.000000] MIPS: machine is Xiaomi Mi Router R3
<6>[    0.000000] Initrd not found or empty - disabling initrd
<4>[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
<4>[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
<6>[    0.000000] Zone ranges:
<6>[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Movable zone start for each node
<6>[    0.000000] Early memory node ranges
<6>[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
<7>[    0.000000] On node 0 totalpages: 32768
<7>[    0.000000]   Normal zone: 256 pages used for memmap
<7>[    0.000000]   Normal zone: 0 pages reserved
<7>[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
<7>[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
<7>[    0.000000] pcpu-alloc: [0] 0 
<6>[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
<5>[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
<6>[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
<6>[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
<6>[    0.000000] Writing ErrCtl register=0005e670
<6>[    0.000000] Readback ErrCtl register=0005e670
<6>[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
<6>[    0.000000] Memory: 121644K/131072K available (4981K kernel code, 190K rwdata, 1084K rodata, 1224K init, 204K bss, 9428K reserved, 0K cma-reserved)
<6>[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
<6>[    0.000000] NR_IRQS: 256
<5>[    0.000000] random: get_random_bytes called from start_kernel+0x350/0x51c with crng_init=0
<6>[    0.000000] CPU Clock: 580MHz
<6>[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
<6>[    0.000000] systick: enable autosleep mode
<6>[    0.000000] systick: running - mult: 214748, shift: 32
<6>[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
<6>[    0.000010] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
<6>[    0.007600] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
<6>[    0.073547] pid_max: default: 32768 minimum: 301
<6>[    0.078182] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.085232] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.094808] devtmpfs: initialized
<6>[    0.102519] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
<6>[    0.112061] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
<6>[    0.118668] pinctrl core: initialized pinctrl subsystem
<6>[    0.124274] NET: Registered protocol family 16
<6>[    0.391339] PCI host bridge /pcie@10140000 ranges:
<6>[    0.395957]  MEM 0x0000000020000000..0x000000002fffffff
<6>[    0.401002]   IO 0x0000000010160000..0x000000001016ffff
<6>[    0.406710] pstore: Registered ramoops as persistent store backend
<6>[    0.412650] ramoops: using 0x80000@0x3f00000, ecc: 0
<6>[    0.441531] rt2880_gpio 10000600.gpio: registering 24 gpios
<6>[    0.446933] rt2880_gpio 10000600.gpio: registering 24 irq handlers
<6>[    0.454078] rt2880_gpio 10000638.gpio: registering 16 gpios
<6>[    0.459467] rt2880_gpio 10000638.gpio: registering 16 irq handlers
<6>[    0.467196] PCI host bridge to bus 0000:00
<6>[    0.471126] pci_bus 0000:00: root bus resource [mem 0x20000000-0x2fffffff]
<6>[    0.477855] pci_bus 0000:00: root bus resource [io  0x10160000-0x1016ffff]
<6>[    0.484469] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
<6>[    0.491076] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
<6>[    0.498795] pci 0000:00:00.0: [1814:0801] type 01 class 0x060400
<6>[    0.504626] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
<6>[    0.510678] pci 0000:00:00.0: reg 0x14: [mem 0x00000000-0x0000ffff]
<6>[    0.516847] pci 0000:00:00.0: supports D1
[... 727 lines cut ...]
<4>[  581.186676] Lo    : ec4e4000
<4>[  581.189627] epc   : 862a4388 rt2x00queue_flush_queues+0x234/0x700 [rt2x00lib]
<4>[  581.196912] ra    : 862a4718 rt2x00queue_flush_queues+0x5c4/0x700 [rt2x00lib]
<4>[  581.204184] Status: 1100f403	KERNEL EXL IE 
<4>[  581.208454] Cause : 40800008 (ExcCode 02)
<4>[  581.212536] BadVA : 00000006
<4>[  581.215471] PrId  : 00019650 (MIPS 24KEc)
<4>[  581.219552] Modules linked in: ksmbd rt2800soc rt2800mmio rt2800lib qcserial pppoe ppp_async option cdc_mbim batman_adv wireguard usb_wwan sierra_net sierra rt2x00soc rt2x00mmio rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp [... cut ...]
<4>[  581.548001]         85c44c48 862a2620 00001080 00000a20 85d923c0 862e9760 00000000 00000000
<4>[  581.556529]         ...
<4>[  581.559025] Call Trace:
<4>[  581.561536] [<862a4388>] rt2x00queue_flush_queues+0x234/0x700 [rt2x00lib]
<4>[  581.568462] Code: 000210c0  00031942  30440010 <92730006> 1080007f  00431025  24020003  ae020018  02003025 
<4>[  581.578418] 
<4>[  581.579980] ---[ end trace 400f705a147802fd ]---
<0>[  581.615053] Kernel panic - not syncing: Fatal exception in interrupt
`

ptpt52, 2021-09-16:
thanks report, please re-download and test again, I just update with new one
```
https://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109141708-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade.bin
https://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109141708-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin
```

PS
to make the log viewable
post them with ``` quote mark

newmem12, 2021-09-16:
> thanks report, please re-download and test again, I just update with new one
>
> ``` [...]

thanks ptpt52, I'm testing and will show you the results of both soon!

ptpt52, 2021-09-16:
update again!!
re-download the firmware.

```
https://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109141708-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade.bin
https://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109141708-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin
```

[image]

newmem12, 2021-09-16:
i will update again, because Xiaomi R3 has just rebooted!
Thanks

newmem12, 2021-09-16:
Hi ptpt52!
Xiaomi R3 has just been rebooted!
dmesg-ramoops-0
````
Oops#1 Part1
<5>[    0.000000] Linux version 5.4.145 (ptpt52@SC-GAME) (gcc version 10.3.0 (OpenWrt GCC 10.3.0 r17500-cf4094acec)) #0 Thu Sep 16 07:47:37 2021
<6>[    0.000000] Board has DDR2
<6>[    0.000000] Analog PMU set to hw control
<6>[    0.000000] Digital PMU set to hw control
<6>[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
<6>[    0.000000] printk: bootconsole [early0] enabled
<6>[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
<6>[    0.000000] MIPS: machine is Xiaomi Mi Router R3
<6>[    0.000000] Initrd not found or empty - disabling initrd
<4>[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
<4>[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
<6>[    0.000000] Zone ranges:
<6>[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Movable zone start for each node
<6>[    0.000000] Early memory node ranges
<6>[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
<7>[    0.000000] On node 0 totalpages: 32768
<7>[    0.000000]   Normal zone: 256 pages used for memmap
<7>[    0.000000]   Normal zone: 0 pages reserved
<7>[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
<7>[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
<7>[    0.000000] pcpu-alloc: [0] 0 
<6>[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
<5>[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
<6>[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
<6>[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
<6>[    0.000000] Writing ErrCtl register=0005e670
<6>[    0.000000] Readback ErrCtl register=0005e670
<6>[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
<6>[    0.000000] Memory: 121644K/131072K available (4981K kernel code, 190K rwdata, 1084K rodata, 1224K init, 204K bss, 9428K reserved, 0K cma-reserved)
<6>[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
<6>[    0.000000] NR_IRQS: 256
<5>[    0.000000] random: get_random_bytes called from start_kernel+0x350/0x51c with crng_init=0
<6>[    0.000000] CPU Clock: 580MHz
<6>[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
<6>[    0.000000] systick: enable autosleep mode
<6>[    0.000000] systick: running - mult: 214748, shift: 32
<6>[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
<6>[    0.000010] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
<6>[    0.007600] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
<6>[    0.073541] pid_max: default: 32768 minimum: 301
<6>[    0.078177] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.085226] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.094802] devtmpfs: initialized
<6>[    0.102513] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
<6>[    0.112056] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
<6>[    0.118662] pinctrl core: initialized pinctrl subsystem
<6>[    0.124268] NET: Registered protocol family 16
<6>[    0.391338] PCI host bridge /pcie@10140000 ranges:
<6>[    0.395956]  MEM 0x0000000020000000..0x000000002fffffff
<6>[    0.401001]   IO 0x0000000010160000..0x000000001016ffff
<6>[    0.406709] pstore: Registered ramoops as persistent store backend
<6>[    0.412649] ramoops: using 0x80000@0x3f00000, ecc: 0
<6>[    0.441529] rt2880_gpio 10000600.gpio: registering 24 gpios
<6>[    0.446932] rt2880_gpio 10000600.gpio: registering 24 irq handlers
<6>[    0.454077] rt2880_gpio 10000638.gpio: registering 16 gpios
<6>[    0.459466] rt2880_gpio 10000638.gpio: registering 16 irq handlers
<6>[    0.467194] PCI host bridge to bus 0000:00
<6>[    0.471125] pci_bus 0000:00: root bus resource [mem 0x20000000-0x2fffffff]
<6>[    0.477854] pci_bus 0000:00: root bus resource [io  0x10160000-0x1016ffff]
<6>[    0.484468] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
<6>[    0.491075] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
<6>[    0.498794] pci 0000:00:00.0: [1814:0801] type 01 class 0x060400
<6>[    0.504625] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[... 327 lines cut ...]
<4>[  377.749669] ra    : 862a4edc rt2x00queue_map_txskb+0x888/0x96c [rt2x00lib]
<4>[  377.756674] Status: 1100f403	KERNEL EXL IE 
<4>[  377.760944] Cause : c0800008 (ExcCode 02)
<4>[  377.765026] BadVA : 00000006
<4>[  377.767961] PrId  : 00019650 (MIPS 24KEc)
<4>[  377.772042] Modules linked in: ksmbd rt2800soc rt2800mmio rt2800lib qcserial pppoe ppp_async option cdc_mbim batman_adv wireguard usb_wwan sierra_net sierra rt2x00soc rt2x00mmio rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack  [... cut ...]
<4>[  378.114052] [<862a4edc>] rt2x00queue_map_txskb+0x888/0x96c [rt2x00lib]
<4>[  378.120727] [<862a5650>] rt2x00queue_write_tx_frame+0x3c/0x4d4 [rt2x00lib]
<4>[  378.127754] [<862a2620>] rt2x00mac_tx+0xb0/0x3f4 [rt2x00lib]
<4>[  378.133680] [<86334a8c>] ieee80211_beacon_update_cntdwn+0x157c/0x21d8 [mac80211]
<4>[  378.141376] Code: 248485a4  0c008cb7  a04391d0 <1000ffbc> 92620006  2484ffff  24020020  00042302  70842020 
<4>[  378.151336] 
<4>[  378.152904] ---[ end trace fe44f29ab17de4e5 ]---
````

newmem12, 2021-09-16:
hi ptpt52, please don't make any more changes!
I just replaced the adapter for Xiaomi R3

ptpt52, 2021-09-16:
md5 for each firmware, make sure u install the correct one.
```
90175936b32e6538fd09e7bff9825199  x-wrt-9.0-b202109141708-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade.bin
19bfa5d5c07bd203281b4887414d8d2f  x-wrt-9.0-b202109141708-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin
```

newmem12, 2021-09-16:
Hi!
Please give me the link again!
Thanks!

ptpt52, 2021-09-16:
the same link above.

newmem12, 2021-09-16:
thanks! i'm testing...

newmem12, 2021-09-16:
hi ptpt52!
Xiaomi Mini has just been rebooted!
dmesg-ramoops-0
````
Oops#1 Part1
<5>[    0.000000] Linux version 5.4.145 (ptpt52@SC-GAME) (gcc version 10.3.0 (OpenWrt GCC 10.3.0 r17500-cf4094acec)) #0 Thu Sep 16 07:47:37 2021
<6>[    0.000000] Board has DDR2
<6>[    0.000000] Analog PMU set to hw control
<6>[    0.000000] Digital PMU set to hw control
<6>[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
<6>[    0.000000] printk: bootconsole [early0] enabled
<6>[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
<6>[    0.000000] MIPS: machine is Xiaomi MiWiFi Mini
<6>[    0.000000] Initrd not found or empty - disabling initrd
<4>[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
<4>[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
<6>[    0.000000] Zone ranges:
<6>[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Movable zone start for each node
<6>[    0.000000] Early memory node ranges
<6>[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
<7>[    0.000000] On node 0 totalpages: 32768
<7>[    0.000000]   Normal zone: 256 pages used for memmap
<7>[    0.000000]   Normal zone: 0 pages reserved
<7>[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
<7>[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
<7>[    0.000000] pcpu-alloc: [0] 0 
<6>[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
<5>[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
<6>[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
<6>[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
<6>[    0.000000] Writing ErrCtl register=0001af10
<6>[    0.000000] Readback ErrCtl register=0001af10
<6>[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
<6>[    0.000000] Memory: 121644K/131072K available (4981K kernel code, 190K rwdata, 1084K rodata, 1224K init, 204K bss, 9428K reserved, 0K cma-reserved)
<6>[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
<6>[    0.000000] NR_IRQS: 256
<5>[    0.000000] random: get_random_bytes called from start_kernel+0x350/0x51c with crng_init=0
<6>[    0.000000] CPU Clock: 580MHz
<6>[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
<6>[    0.000000] systick: enable autosleep mode
<6>[    0.000000] systick: running - mult: 214748, shift: 32
<6>[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
<6>[    0.000011] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
<6>[    0.007600] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
<6>[    0.073546] pid_max: default: 32768 minimum: 301
<6>[    0.078181] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.085231] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.094805] devtmpfs: initialized
<6>[    0.102496] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
<6>[    0.112039] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
<6>[    0.118655] pinctrl core: initialized pinctrl subsystem
<6>[    0.124267] NET: Registered protocol family 16
<6>[    0.391639] PCI host bridge /pcie@10140000 ranges:
<6>[    0.396255]  MEM 0x0000000020000000..0x000000002fffffff
<6>[    0.401302]   IO 0x0000000010160000..0x000000001016ffff
<6>[    0.407007] pstore: Registered ramoops as persistent store backend
<6>[    0.412946] ramoops: using 0x80000@0x3f00000, ecc: 0
<6>[    0.441898] rt2880_gpio 10000600.gpio: registering 24 gpios
<6>[    0.447300] rt2880_gpio 10000600.gpio: registering 24 irq handlers
<6>[    0.454453] rt2880_gpio 10000638.gpio: registering 16 gpios
<6>[    0.459842] rt2880_gpio 10000638.gpio: registering 16 irq handlers
<6>[    0.467016] rt2880_gpio 10000660.gpio: registering 32 gpios
<6>[    0.472405] rt2880_gpio 10000660.gpio: registering 32 irq handlers
<6>[    0.480099] PCI host bridge to bus 0000:00
<6>[    0.484029] pci_bus 0000:00: root bus resource [mem 0x20000000-0x2fffffff]
<6>[    0.490766] pci_bus 0000:00: root bus resource [io  0x10160000-0x1016ffff]
<6>[    0.497379] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
<6>[    0.503979] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[... 684 lines cut ...]
<4>[  330.365143] Status: 1100f403	KERNEL EXL IE 
<4>[  330.369413] Cause : c0800008 (ExcCode 02)
<4>[  330.373495] BadVA : 00000006
<4>[  330.376430] PrId  : 00019650 (MIPS 24KEc)
<4>[  330.380511] Modules linked in: ksmbd rt2800soc rt2800mmio rt2800lib qcserial pppoe ppp_async option cdc_mbim batman_adv wireguard usb_wwan sierra_net sierra rt2x00soc rt2x00mmio rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_RE [... cut ...]
<4>[  330.729200] [<86305650>] rt2x00queue_write_tx_frame+0x3c/0x4d4 [rt2x00lib]
<4>[  330.736227] [<86302620>] rt2x00mac_tx+0xb0/0x3f4 [rt2x00lib]
<4>[  330.742150] [<86234a8c>] ieee80211_beacon_update_cntdwn+0x157c/0x21d8 [mac80211]
<4>[  330.749851] Code: 248485a4  0c008cb7  a04391d0 <1000ffbc> 92620006  2484ffff  24020020  00042302  70842020 
<4>[  330.759812] 
<4>[  330.761373] ---[ end trace fb52990dbe50320b ]---
<0>[  330.794747] Kernel panic - not syncing: Fatal exception in interrupt
````

ptpt52, 2021-09-16:
ok I get the log, let me analyze

newmem12, 2021-09-17:
Thanks ptpt52!
In the meantime, i will back to the previous firmware!

ptpt52, 2021-09-18:
```
https://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109172238-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade.bin
https://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109172238-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin
```

@newmem12  test this new firmware. this would crash and get the crash log please.
it is hard to determinate the cause of the issue. so more test and log is needed

newmem12, 2021-09-18:
thanks ptpt52!
i'm testing...

newmem12, 2021-09-18:
hi ptpt52!
Xiaomi R3 has just been rebooted!
dmesg-ramoops-0
````
Oops#1 Part1
<5>[    0.000000] Linux version 5.4.145 (ptpt52@SC-GAME) (gcc version 10.3.0 (OpenWrt GCC 10.3.0 r17500-cf4094acec)) #0 Fri Sep 17 18:57:45 2021
<6>[    0.000000] Board has DDR2
<6>[    0.000000] Analog PMU set to hw control
<6>[    0.000000] Digital PMU set to hw control
<6>[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
<6>[    0.000000] printk: bootconsole [early0] enabled
<6>[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
<6>[    0.000000] MIPS: machine is Xiaomi Mi Router R3
<6>[    0.000000] Initrd not found or empty - disabling initrd
<4>[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
<4>[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
<6>[    0.000000] Zone ranges:
<6>[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Movable zone start for each node
<6>[    0.000000] Early memory node ranges
<6>[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
<7>[    0.000000] On node 0 totalpages: 32768
<7>[    0.000000]   Normal zone: 256 pages used for memmap
<7>[    0.000000]   Normal zone: 0 pages reserved
<7>[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
<7>[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
<7>[    0.000000] pcpu-alloc: [0] 0 
<6>[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
<5>[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
<6>[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
<6>[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
<6>[    0.000000] Writing ErrCtl register=0002eb00
<6>[    0.000000] Readback ErrCtl register=0002eb00
<6>[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
<6>[    0.000000] Memory: 121644K/131072K available (5000K kernel code, 184K rwdata, 1088K rodata, 1208K init, 204K bss, 9428K reserved, 0K cma-reserved)
<6>[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
<6>[    0.000000] NR_IRQS: 256
<5>[    0.000000] random: get_random_bytes called from start_kernel+0x350/0x51c with crng_init=0
<6>[    0.000000] CPU Clock: 580MHz
<6>[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
<6>[    0.000000] systick: enable autosleep mode
<6>[    0.000000] systick: running - mult: 214748, shift: 32
<6>[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
<6>[    0.000011] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
<6>[    0.007599] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
<6>[    0.073548] pid_max: default: 32768 minimum: 301
<6>[    0.078179] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.085232] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.094779] devtmpfs: initialized
<6>[    0.102484] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
<6>[    0.112025] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
<6>[    0.118635] pinctrl core: initialized pinctrl subsystem
<6>[    0.124245] NET: Registered protocol family 16
<6>[    0.391475] PCI host bridge /pcie@10140000 ranges:
<6>[    0.396094]  MEM 0x0000000020000000..0x000000002fffffff
<6>[    0.401138]   IO 0x0000000010160000..0x000000001016ffff
<6>[    0.406853] pstore: Registered ramoops as persistent store backend
<6>[    0.412796] ramoops: using 0x80000@0x3f00000, ecc: 0
<6>[    0.441659] rt2880_gpio 10000600.gpio: registering 24 gpios
<6>[    0.447063] rt2880_gpio 10000600.gpio: registering 24 irq handlers
<6>[    0.454216] rt2880_gpio 10000638.gpio: registering 16 gpios
<6>[    0.459608] rt2880_gpio 10000638.gpio: registering 16 irq handlers
<6>[    0.467336] PCI host bridge to bus 0000:00
<6>[    0.471267] pci_bus 0000:00: root bus resource [mem 0x20000000-0x2fffffff]
<6>[    0.477997] pci_bus 0000:00: root bus resource [io  0x10160000-0x1016ffff]
<6>[    0.484613] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
<6>[    0.491223] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
<6>[    0.498937] pci 0000:00:00.0: [1814:0801] type 01 class 0x060400
<6>[    0.504772] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[... 742 lines cut ...]
<4>[  272.321231] Status: 1100f403	KERNEL EXL IE 
<4>[  272.325499] Cause : c0800008 (ExcCode 02)
<4>[  272.329582] BadVA : 00000006
<4>[  272.332516] PrId  : 00019650 (MIPS 24KEc)
<4>[  272.336597] Modules linked in: ksmbd rt2800soc rt2800mmio rt2800lib qcserial pppoe ppp_async option cdc_mbim batman_adv wireguard usb_wwan sierra_net sierra rt2x00soc rt2x00mmio rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS  [... cut ...]
<4>[  272.685283] [<862a5670>] rt2x00queue_write_tx_frame+0x3c/0x4d4 [rt2x00lib]
<4>[  272.692310] [<862a2620>] rt2x00mac_tx+0xb0/0x3f4 [rt2x00lib]
<4>[  272.698234] [<86334a8c>] ieee80211_beacon_update_cntdwn+0x157c/0x21d8 [mac80211]
<4>[  272.705827] Code: 248485c4  0c008cb7  a04391f0 <1000ffbc> 92620006  2484ffff  24020020  00042302  70842020 
<4>[  272.715784] 
<4>[  272.717357] ---[ end trace 46460e36da0c2de4 ]---
<0>[  272.752448] Kernel panic - not syncing: Fatal exception in interrupt
````

ptpt52, 2021-09-18:
ok @newmem12 
test this new one:
```
https://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109172238-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade.bin
https://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109172238-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin
```

it should fix

newmem12, 2021-09-18:
hi ptpt52!
Xiaomi Mini has just been rebooted!
dmesg-ramoops-0
````
Oops#1 Part1
<5>[    0.000000] Linux version 5.4.145 (ptpt52@SC-GAME) (gcc version 10.3.0 (OpenWrt GCC 10.3.0 r17500-cf4094acec)) #0 Sat Sep 18 02:18:48 2021
<6>[    0.000000] Board has DDR2
<6>[    0.000000] Analog PMU set to hw control
<6>[    0.000000] Digital PMU set to hw control
<6>[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
<6>[    0.000000] printk: bootconsole [early0] enabled
<6>[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
<6>[    0.000000] MIPS: machine is Xiaomi MiWiFi Mini
<6>[    0.000000] Initrd not found or empty - disabling initrd
<4>[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
<4>[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
<6>[    0.000000] Zone ranges:
<6>[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Movable zone start for each node
<6>[    0.000000] Early memory node ranges
<6>[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
<6>[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
<7>[    0.000000] On node 0 totalpages: 32768
<7>[    0.000000]   Normal zone: 256 pages used for memmap
<7>[    0.000000]   Normal zone: 0 pages reserved
<7>[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
<7>[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
<7>[    0.000000] pcpu-alloc: [0] 0 
<6>[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
<5>[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
<6>[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
<6>[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
<6>[    0.000000] Writing ErrCtl register=0001e770
<6>[    0.000000] Readback ErrCtl register=0001e770
<6>[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
<6>[    0.000000] Memory: 121644K/131072K available (4981K kernel code, 190K rwdata, 1084K rodata, 1224K init, 204K bss, 9428K reserved, 0K cma-reserved)
<6>[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
<6>[    0.000000] NR_IRQS: 256
<5>[    0.000000] random: get_random_bytes called from start_kernel+0x350/0x51c with crng_init=0
<6>[    0.000000] CPU Clock: 580MHz
<6>[    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
<6>[    0.000000] systick: enable autosleep mode
<6>[    0.000000] systick: running - mult: 214748, shift: 32
<6>[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
<6>[    0.000011] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
<6>[    0.007600] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
<6>[    0.073543] pid_max: default: 32768 minimum: 301
<6>[    0.078178] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.085228] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
<6>[    0.094801] devtmpfs: initialized
<6>[    0.102493] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
<6>[    0.112036] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
<6>[    0.118652] pinctrl core: initialized pinctrl subsystem
<6>[    0.124264] NET: Registered protocol family 16
<6>[    0.391636] PCI host bridge /pcie@10140000 ranges:
<6>[    0.396252]  MEM 0x0000000020000000..0x000000002fffffff
<6>[    0.401299]   IO 0x0000000010160000..0x000000001016ffff
<6>[    0.407004] pstore: Registered ramoops as persistent store backend
<6>[    0.412943] ramoops: using 0x80000@0x3f00000, ecc: 0
<6>[    0.441895] rt2880_gpio 10000600.gpio: registering 24 gpios
<6>[    0.447297] rt2880_gpio 10000600.gpio: registering 24 irq handlers
<6>[    0.454451] rt2880_gpio 10000638.gpio: registering 16 gpios
<6>[    0.459840] rt2880_gpio 10000638.gpio: registering 16 irq handlers
<6>[    0.467015] rt2880_gpio 10000660.gpio: registering 32 gpios
<6>[    0.472404] rt2880_gpio 10000660.gpio: registering 32 irq handlers
<6>[    0.480097] PCI host bridge to bus 0000:00
<6>[    0.484028] pci_bus 0000:00: root bus resource [mem 0x20000000-0x2fffffff]
<6>[    0.490765] pci_bus 0000:00: root bus resource [io  0x10160000-0x1016ffff]
<6>[    0.497378] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
<6>[    0.503978] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[... 632 lines cut ...]
<4>[ 2299.873366] Lo    : f4e831de
<4>[ 2299.876340] epc   : 86304594 rt2x00queue_flush_queues+0x440/0x75c [rt2x00lib]
<4>[ 2299.883624] ra    : 86304254 rt2x00queue_flush_queues+0x100/0x75c [rt2x00lib]
<4>[ 2299.890894] Status: 1100f403	KERNEL EXL IE 
<4>[ 2299.895164] Cause : 40800008 (ExcCode 02)
<4>[ 2299.899247] BadVA : 00000006
<4>[ 2299.902181] PrId  : 00019650 (MIPS 24KEc)
<4>[ 2299.906263] Modules linked in: ksmbd rt2800soc rt2800mmio rt2800lib qcserial pppoe ppp_async option cdc_mbim batman_adv wireguard usb_wwan sierra_net sierra rt2x00soc rt2x00mmio rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt [... cut ...]
<4>[ 2300.234750]         863b2048 86302620 00001080 00000a20 85fdbb40 861ed660 00000000 00000000
<4>[ 2300.243277]         ...
<4>[ 2300.245775] Call Trace:
<4>[ 2300.248287] [<86304594>] rt2x00queue_flush_queues+0x440/0x75c [rt2x00lib]
<4>[ 2300.255218] Code: 00021080  8c830004  00621021 <90420006> 02003025  00029880  02629821  00131040  3c138630 
<4>[ 2300.265175] 
<4>[ 2300.266761] ---[ end trace 077fcc0be08ebfe0 ]---
<0>[ 2300.295334] Kernel panic - not syncing: Fatal exception in interrupt
````

ptpt52, 2021-09-18:
ok, now this one:
```
https://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109181630-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin
https://downloads.x-wrt.com/rom/Downloads/x-wrt-9.0-b202109181630-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade.bin
```

ptpt52, 2021-09-18:
it should fix....

newmem12, 2021-09-18:
thank you so much ptpt52, i'm testing...

newmem12, 2021-09-18:
hi ptpt52!
This firmware is running stable!

newmem12, 2021-09-19:
hi ptpt52!
This firmware is uptime for 14 hours without being rebooted!
How can i apply to other devices with MT7620 chipset and disable pfstore?
Thank you!

ptpt52, 2021-09-19:
firmwares for other device would be in next release
or you could build them from latest code

and pstore would not be disabled in future version.

newmem12, 2021-09-19:
hi ptpt52!
pstore for recording crash files when diagnostics are needed, will it write to flash rom? and is it ok to write to the router's flash rom regularly?

ptpt52, 2021-09-19:
this pstore write to ram memory.

[3 more replies]

closed 2021-09-20 by newmem12