x-wrt/x-wrt #186: crash on mt7621 swconfig driver

ptpt52, 2021-09-27:
```
<6>[21489.208199] ralink_soc_eth 1e100000.ethernet eth0: port 3 link down
<6>[21492.175642] ralink_soc_eth 1e100000.ethernet eth0: port 3 link up
<6>[21493.775503] ralink_soc_eth 1e100000.ethernet eth0: port 3 link down
<6>[21496.663596] ralink_soc_eth 1e100000.ethernet eth0: port 3 link up
<1>[21538.054857] BUG: Bad page state in process swapper/0  pfn:03261
<4>[21538.066649] page:80911c20 refcount:-1 mapcount:0 mapping:00000000 index:0x0
<4>[21538.080490] flags: 0x0()
<4>[21538.085527] raw: 00000000 00000000 80910301 00000000 00000000 00000000 ffffffff ffffffff
<4>[21538.101610] page dumped because: nonzero _refcount
<4>[21538.111125] Modules linked in: ksmbd qcserial pppoe ppp_async option cdc_mbim wireguard usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NFQUEUE xt_NETMAP xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usbnet usblp ts_fsm ts_bm tcp_hybla tcp_bbr slhc sch_cake r8152 poly1305_mips nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash
<4>[21538.111318]  nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c libblake2s_generic ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables crc_ccitt compat_xtables compat chacha_mips cdc_wdm br_netfilter asn1_decoder natflow natcap fuse sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet
<4>[21538.284641]  ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos bonding ip6_gre ip_gre gre ifb ip6_udp_tunnel udp_tunnel sit ip6_tunnel tunnel6 tunnel4 ip_tunnel rpcsec_gss_krb5 auth_rpcgss oid_registry tun vfat fat lockd sunrpc grace autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac ghash_generic gf128mul gcm ecb des_generic libdes cts ctr cmac ccm cbc arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ohci_platform ohci_hcd
<4>[21538.457991]  softdog fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat usbcore nls_base usb_common mii crc32c_generic
<4>[21538.662353] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.138 #0
<4>[21538.674293] Stack : 00000003 8007337c 806c0000 806c2114 80730000 806c20dc 806c0f88 87c0bc1c
<4>[21538.690923]         80870000 8070d114 8070cc63 806508d8 00000000 00000001 87c0bbc0 6d8dfd85
<4>[21538.707549]         00000000 00000000 808b0000 00000000 00000030 0000019b 342e3520 3833312e
<4>[21538.724176]         00000000 0000042f 00000000 000a1b51 00000000 80730000 00000000 808a0000
<4>[21538.740803]         80655e94 80710000 807308c0 00000008 00000003 8033e4ec 00000000 80870000
<4>[21538.757434]         ...
<4>[21538.762293] Call Trace:
<4>[21538.767176] [<8000b61c>] show_stack+0x30/0x100
<4>[21538.776041] [<80595618>] dump_stack+0xa4/0xdc
<4>[21538.784727] [<80125fec>] bad_page+0x120/0x154
<4>[21538.793395] [<80128ff8>] get_page_from_freelist+0x438/0xcb4
<4>[21538.804482] [<80129f64>] __alloc_pages_nodemask+0xf0/0xd00
<4>[21538.815394] [<8012ad14>] page_frag_alloc+0xb0/0x1a8
<4>[21538.825109] [<803cc36c>] fe_poll+0x314/0x930
<4>[21538.833620] [<804232c4>] __napi_poll+0x3c/0x10c
<4>[21538.842635] [<80423538>] net_rx_action+0x114/0x28c
<4>[21538.852170] [<805b7074>] __do_softirq+0x16c/0x334
<4>[21538.861547] [<800302fc>] irq_exit+0x98/0xb0
<4>[21538.869877] [<802dd410>] plat_irq_dispatch+0x64/0x104
<4>[21538.879925] [<80006d08>] except_vec_vi_end+0xb8/0xc4
<4>[21538.889798] [<805b6728>] r4k_wait_irqoff+0x1c/0x24
<4>[21538.899329] Disabling lock debugging due to kernel taint
<1>[21539.252750] CPU 0 Unable to handle kernel paging request at virtual address c0010000, epc == 80023d50, ra == 800214c4
<4>[21539.273904] Oops[#1]:
<4>[21539.278425] CPU: 0 PID: 1735 Comm: mt76-tx phy1 Tainted: G    B             5.4.138 #0
<4>[21539.294177] $ 0   : 00000000 00000001 c0010000 00000020
<4>[21539.304584] $ 4   : 80000000 0408032d 0207fb2d c0010020
<4>[21539.314975] $ 8   : 00000100 00000040 00000000 ffffffff
<4>[21539.325369] $12   : 832f98e0 00000bf4 000000ff 8491ae30
<4>[21539.335767] $16   : 80000000 81006590 85be1f80 854fbd1c
<4>[21539.346163] $20   : 87fce470 0000000c 854fbcbc 856dbf94
<4>[21539.356557] $24   : 848d5494 800212cc                  
<4>[21539.366947] $28   : 854fa000 854fbc50 808a0000 800214c4
<4>[21539.377351] Hi    : 00000000
<4>[21539.383084] Lo    : 0408032d
<4>[21539.388821] epc   : 80023d50 mips_sc_wback_inv+0x4c/0x114
<4>[21539.399560] ra    : 800214c4 r4k_dma_cache_wback_inv+0x1f8/0x340
<4>[21539.411498] Status: 11008403	KERNEL EXL IE 
<4>[21539.419819] Cause : 40800008 (ExcCode 02)
<4>[21539.427787] BadVA : c0010000
<4>[21539.433506] PrId  : 0001992f (MIPS 1004Kc)
<4>[21539.441640] Modules linked in: ksmbd qcserial pppoe ppp_async option cdc_mbim wireguard usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NFQUEUE xt_NETMAP xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usbnet usblp ts_fsm ts_bm tcp_hybla tcp_bbr slhc sch_cake r8152 poly1305_mips nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash
<4>[21539.441949]  nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c libblake2s_generic ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables crc_ccitt compat_xtables compat chacha_mips cdc_wdm br_netfilter asn1_decoder natflow natcap fuse sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet
[... cut ...]
<4>[21540.103178] [<80023d50>] mips_sc_wback_inv+0x4c/0x114
<4>[21540.113231] [<800214c4>] r4k_dma_cache_wback_inv+0x1f8/0x340
<4>[21540.124506] [<800897e8>] dma_direct_map_page+0xfc/0x130
<4>[21540.134935] [<85bb1d88>] mt76_dma_rx_poll+0xabc/0xf04 [mt76]
<4>[21540.146193] Code: 00003012  00643821  00801025 <bc570000> bcf70000  01221021  bc570000  00431021  bc570000 
<4>[21540.165597] 
<4>[21540.169233] ---[ end trace 4e8034be77d9e25e ]---
<0>[21540.209019] Kernel panic - not syncing: Fatal exception in interrupt

```

kokroo, 2021-10-04:
Looks like a clock timing issue

ptpt52, 2022-05-28:
should be fixed in lastest code

closed 2022-05-28 by ptpt52