x-wrt/x-wrt #189: Tenbay T-MB5EU cannot  use in 5GHZ master <-> slave

F5BJR, 2021-10-16:
as in the Tittle:

Tenbay T-MB5EU cannot  use in 5GHZ master <-> slave

No problem in 2.4GHz

ptpt52, 2021-10-17:
post your config `/etc/config/wireless`
and `dmesg` log

ptpt52, 2021-10-17:
also `logread`

F5BJR, 2021-10-17:
thanks for reply

Master infos

``[    0.000000] Linux version 5.4.152 (user@INSPIRON-7791) (gcc version 11.2.0 (OpenWrt GCC 11.2.0 r17757-3a93704a63)) #0 SMP Thu Oct 14 23:12:30 2021
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Tenbay T-MB5EU-32M
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x000000008fe99fff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   node   0: [mem 0x000000008fe97000-0x000000008fe99fff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000008fe99fff]
[    0.000000] On node 0 totalpages: 65539
[    0.000000]   Normal zone: 576 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 3 pages, LIFO batch:0
[    0.000000] percpu: Embedded 14 pages/cpu s26480 r8192 d22672 u57344
[    0.000000] pcpu-alloc: s26480 r8192 d22672 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64963
[    0.000000] Kernel command line: console=ttyS0,115200
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=000101e6
[    0.000000] Readback ErrCtl register=000101e6
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 231852K/262156K available (6137K kernel code, 204K rwdata, 1248K rodata, 1228K init, 230K bss, 30304K reserved, 0K cma-reserved, 12K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x368/0x580 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007804] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.073968] pid_max: default: 32768 minimum: 301
[    0.078757] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.085957] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.096329] rcu: Hierarchical SRCU implementation.
[    0.101650] smp: Bringing up secondary CPUs ...
[    9.039656] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    9.039665] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    9.039677] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    9.039771] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.166422] Synchronize counters for CPU 1: done.
[    8.443493] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    8.443501] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    8.443507] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    8.443557] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.257125] Synchronize counters for CPU 2: done.
[    8.532869] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    8.532876] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    8.532884] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    8.532937] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.342304] Synchronize counters for CPU 3: done.
[    0.372160] smp: Brought up 1 node, 4 CPUs
[    0.380720] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[... 824 lines cut ...]
	option cell_density '0'

config wifi-device 'radio1'
	option type 'mac80211'
	option path '1e140000.pcie/pci0000:00/0000:00:01.0/0000:02:00.0+1'
	option channel '36'
	option band '5g'
	option htmode 'HE80'
	option cell_density '0'

config wifi-iface 'wifinet0'
	option device 'radio0'
	option mode 'ap'
	option encryption 'none'
	option network 'lan'
	option ssid 'WIFI-2.4GHZ'

config wifi-iface 'wifinet1'
	option device 'radio1'
	option mode 'ap'
	option encryption 'none'
	option network 'lan'
	option ssid 'WIFI-5GHz'

``

F5BJR, 2021-10-17:
SLAVE

closed 2021-10-17 by F5BJR

F5BJR, 2021-10-18:
Is this error ?

WARNING: CPU: 2 PID: 1692 at backports-5.10.68-1/net/mac80211/sta_info.c:546 sta_info_insert_rcu+0xac/0x78c [mac80211]