x-wrt/x-wrt #193: wireguard on x-wrt ?

randomername, 2021-11-04:
i am using  X-WRT 9.0_b202110142350 on a mi 3 router

i am trying to use wireguard but i dont see how to.
there is no place to install a package either.

this is what shows under STATUS > WIREGUARD
[image]

please point me in the right direction.

ptpt52, 2021-11-04:
it is ready to use
try Network - interfaces - create new - select protocol

randomername, 2021-11-27:
than

> it is ready to use try Network - interfaces - create new - select protocol

thank you so much.

i have one more issue.

i have a usb stick plugged into my main router that i want to access using the x-wrt router too.

the x-wrt is running a wireguard VPN and has kill switch firewall rules.

i am unable to even ping the host router from which x-wrt draws the internet

closed 2021-11-30 by randomername