x-wrt/x-wrt #196: Random restarts

xesarni, 2021-11-06:
X-WRT 9.0_b202110300939

TOTOLINK X5000R

```
<1>[65876.859154] CPU 1 Unable to handle kernel paging request at virtual address 0000002f, epc == 8e5e1fd4, ra == 8e5e2128
<4>[65876.869783] Oops[#1]:
<4>[65876.872054] CPU: 1 PID: 30429 Comm: kworker/1:2 Not tainted 5.4.155 #0
<4>[65876.878603] Workqueue: events_power_efficient nf_ct_iterate_destroy [nf_conntrack]
<4>[65876.886135] $ 0   : 00000000 00000001 00000000 00641000
<4>[65876.891342] $ 4   : 00000000 8012000d 00000000 8d7b78c0
<4>[65876.896545] $ 8   : f58336d5 0000d0a8 8dfca2a0 80808080
<4>[65876.901748] $12   : fefefeff 00000400 00000000 00000000
<4>[65876.906950] $16   : 8e8d9ab0 00000000 fffffff0 00000000
<4>[65876.912154] $20   : 80720000 00001000 00000f2d 8e5f0000
<4>[65876.917357] $24   : 0000c493 805d1fc4                  
<4>[65876.922560] $28   : 8d958000 8d959e18 80740f80 8e5e2128
<4>[65876.927766] Hi    : 00000af7
<4>[65876.930628] Lo    : 5694d000
<4>[65876.933526] epc   : 8e5e1fd4 nf_ct_iterate_destroy+0x1a4/0xb00 [nf_conntrack]
<4>[65876.940653] ra    : 8e5e2128 nf_ct_iterate_destroy+0x2f8/0xb00 [nf_conntrack]
<4>[65876.947756] Status: 11007c03	KERNEL EXL IE 
<4>[65876.951939] Cause : 40800008 (ExcCode 02)
<4>[65876.955925] BadVA : 0000002f
<4>[65876.958789] PrId  : 0001992f (MIPS 1004Kc)
<4>[65876.962860] Modules linked in: pppoe ppp_async batman_adv wireguard pptp pppox ppp_mppe ppp_generic mt7915e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NFQUEUE xt_NETMAP xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY ts_fsm ts_bm tcp_hybla tcp_bbr slhc sch_cake poly1305_mips nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp
<4>[65876.963025]  nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c libblake2s_generic ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables hwmon crc_ccitt compat_xtables compat chacha_mips br_netfilter asn1_decoder natflow natcap sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port
<4>[65877.050270]  ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 bonding ip6_gre ip_gre gre ifb nat46 ip6_udp_tunnel udp_tunnel sit ip6_tunnel tunnel6 tunnel4 ip_tunnel tun nls_utf8 nls_iso8859_1 nls_base sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gf128mul gcm ecb ctr cmac ccm arc4 leds_gpio softdog gpio_button_hotplug crc32c_generic
<4>[65877.192074] Process kworker/1:2 (pid: 30429, threadinfo=9bd3d539, task=a1e8943b, tls=00000000)
<4>[65877.200647] Stack : 8fc3bb20 8dcecb40 8dcece70 8d959e64 80720000 8e5f0000 8e5f0000 00641001
<4>[65877.208972]         8e5f2164 8e5ef2a4 0000000b 8071ff00 8e5f2164 8de55480 81013940 81016c00
<4>[65877.217294]         00000000 00000000 00000040 fffffffe 80720000 800480a4 81013940 81013940
<4>[65877.225613]         80720000 81013958 81013940 80720000 8de55480 8de55494 81013940 80720000
<4>[65877.233936]         81013958 81013940 80720000 80048924 8de55480 80048790 80660000 80880000
<4>[65877.242259]         ...
<4>[65877.244759] Call Trace:
<4>[65877.247236] [<8e5e1fd4>] nf_ct_iterate_destroy+0x1a4/0xb00 [nf_conntrack]
<4>[65877.254029] Code: 32220001  14400054  00000000 <9222002f> 00028040  02028021  00108100  02508023  02308021 
<4>[65877.263743] 
<4>[65877.265449] ---[ end trace ea13c710515f7a64 ]---
```

[dmesg-ramoops-0.zip](https://github.com/x-wrt/x-wrt/files/7490888/dmesg-ramoops-0.zip)

ptpt52, 2021-11-06:
thanks for the report, Please try the latest version again.
not sure if it is fixed or not.

theorland, 2021-11-07:
same with me on redmi router ac2100 too

simpsss, 2021-11-09:
same problem here.

Model | Xiaomi Mi Router 3G

ptpt52, 2021-11-10:
An new release post here https://downloads.x-wrt.com/rom/
Some features[1] that may cause problems are turned off
Hope your test feedback

[1] user auth functions in natflow kernel module

xesarni, 2021-11-14:
X-WRT 21.10_b202111081727 Impish

Uptime	4d 3h 15m 18s

It seems to be fixed.

ptpt52, 2021-11-27:
still issue in latest version

xesarni, 2021-11-27:
In b202111260705?
I had no problems with b202111081727

xesarni, 2021-11-27:
From b202111260705:
```
<1>[74538.882074] CPU 1 Unable to handle kernel paging request at virtual address 0000002f, epc == 8e4e375c, ra == 8e4e3730
<4>[74538.892686] Oops[#1]:
<4>[74538.894965] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 5.4.159 #0
<4>[74538.900955] $ 0   : 00000000 00000001 00000000 00000000
<4>[74538.906180] $ 4   : 8c545d50 0071479a 00000000 00000000
<4>[74538.911414] $ 8   : 439fb76b 805cbc34 00000000 00000000
<4>[74538.916648] $12   : 00000000 00000000 00000000 00000000
<4>[74538.921869] $16   : 00000000 8d58e8c0 8fc0fa3c fffffff0
<4>[74538.927101] $20   : 80740000 80720000 8e4f0000 00000001
<4>[74538.932320] $24   : 8d58e8c0 804b4948                  
<4>[74538.937537] $28   : 8fc62000 8fc0f978 80740f80 8e4e3730
<4>[74538.942756] Hi    : 00000a5a
<4>[74538.945625] Lo    : 611f3000
<4>[74538.948618] epc   : 8e4e375c nf_conntrack_tuple_taken+0xc8/0x8d8 [nf_conntrack]
<4>[74538.955955] ra    : 8e4e3730 nf_conntrack_tuple_taken+0x9c/0x8d8 [nf_conntrack]
<4>[74538.963233] Status: 11007c03	KERNEL EXL IE 
<4>[74538.967406] Cause : 40800008 (ExcCode 02)
<4>[74538.971394] BadVA : 0000002f
<4>[74538.974260] PrId  : 0001992f (MIPS 1004Kc)
<4>[74538.978338] Modules linked in: pppoe ppp_async batman_adv wireguard pptp pppox ppp_mppe ppp_generic mt7915e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NFQUEUE xt_NETMAP xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_DSCP xt_CT xt_CLASSIFY ts_fsm ts_bm tcp_hybla tcp_bbr slhc sch_cake poly1305_mips nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp
<4>[74538.978559]  nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c libblake2s_generic ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables hwmon crc_ccitt compat_xtables compat chacha_mips br_netfilter asn1_decoder natflow natcap sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port
<4>[74539.065845]  ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 bonding ip6_gre ip_gre gre ifb nat46 ip6_udp_tunnel udp_tunnel sit ip6_tunnel tunnel6 tunnel4 ip_tunnel tun nls_utf8 nls_iso8859_1 nls_base sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gf128mul gcm ecb ctr cmac ccm arc4 leds_gpio softdog gpio_button_hotplug crc32c_generic
<4>[74539.207693] Process swapper/1 (pid: 0, threadinfo=a707bfb2, task=f0751952, tls=00000000)
<4>[74539.215754] Stack : 8fc0fb1c 80516494 8fc0fa54 8fc0fab8 8e4f0000 8e4f0000 00000a5a 807203b0
<4>[74539.224080]         80740740 8072cfd8 8fc0faa0 00000000 8d58e8c0 00000000 80720000 8e9ee000
<4>[74539.232403]         00000000 8fc0fa14 8e65e5f4 8e4d8884 00000000 8fc0fab8 8fc0fa54 00000001
<4>[74539.240726]         00000001 808b0000 0000001f 80720000 808b0000 0101a8c0 00000000 00000000
<4>[74539.249056]         00000000 0002ecd9 ab235968 00000000 00000000 00000000 0006bb01 f1b71d53
<4>[74539.257417]         ...
<4>[74539.259987] Call Trace:
<4>[74539.262560] [<8e4e375c>] nf_conntrack_tuple_taken+0xc8/0x8d8 [nf_conntrack]
<4>[74539.269595] [<8e4d8884>] nf_ct_nat_ext_add+0x730/0xf74 [nf_nat]
<4>[74539.275569] Code: 02001825  2413fff0  24170001 <9202002f> 00022040  00822021  00042100  02642023  02042021 
<4>[74539.285310] 
<4>[74539.287151] ---[ end trace 2b0bed98f1c9b401 ]---
<0>[74539.312622] Kernel panic - not syncing: Fatal exception in interrupt
<4>[74539.318988] ------------[ cut here ]------------
<4>[74539.323641] WARNING: CPU: 1 PID: 0 at kernel/smp.c:433 smp_call_function_many+0x3dc/0x3f8
<4>[74539.331785] Modules linked in: pppoe ppp_async batman_adv wireguard pptp pppox ppp_mppe ppp_generic mt7915e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connb [... cut ...]
<4>[74539.632821] [<800ac8ec>] smp_call_function_many+0x3dc/0x3f8
<4>[74539.638367] [<800ac928>] smp_call_function+0x20/0x2c
<4>[74539.643317] [<8002e050>] panic+0x120/0x310
<4>[74539.647405] [<8000b5a8>] die+0x104/0x110
<4>[74539.651328] [<80017780>] do_page_fault+0x4b8/0x4c8
<4>[74539.656105] [<8001c948>] tlb_do_page_fault_0+0x118/0x120
<4>[74539.661514] [<8e4e3730>] nf_conntrack_tuple_taken+0x9c/0x8d8 [nf_conntrack]
<4>[74539.668552] [<8e4d8884>] nf_ct_nat_ext_add+0x730/0xf74 [nf_nat]
<4>[74539.674486] ---[ end trace 2b0bed98f1c9b402 ]---

```

ptpt52, 2021-11-28:
should be fixed since https://github.com/x-wrt/x-wrt/tree/21.10_b202111280050

ptpt52, 2021-11-29:
fixed

closed 2021-11-29 by ptpt52

ThomasCr, 2022-01-28:
I have also random reboots on 21.10-b202201172351 on my xiaomi Redmi ac 2100.

I upgraded from openwrt with sysuprade without keeping settings. I also did again a reset from Webinterface (when it was available, most time the router reboots)

xesarni, 2022-01-28:
21.10_b202201172351 works stable here.

Uptime  8d 6h 56m 50s