x-wrt/x-wrt #199: x-wrt as VPN gateway

randomername, 2021-11-29:
i have a usb stick plugged into my main router that i want to access using the x-wrt router too.

the x-wrt is running a wireguard VPN and has kill switch firewall rules.

i am unable to even ping the host router from which x-wrt draws the internet

ptpt52, 2021-11-29:
what top?

randomername, 2021-11-29:
> what top?

i dont understand? what is a top?

[Powered by LuCI Master (git-21.286.81213-cd11ef6) / X-WRT 9.0_b202110142350 Hirsute r17911+1-e510ff17e2]

this is what i am using as x-wrt
its wan is connected to the host router that has the usb stick plugged in

ptpt52, 2021-11-30:
or show your config
```
/etc/config/network
```

randomername, 2021-12-01:
> or show your config
> 
> ```
> /etc/config/network
> ```

im sorry for being a pain...but how do i show my config?

im using the LUCI gui interface and have no idea how to find my config.

can u please guide me?

ptpt52, 2021-12-01:
what setup, maybe screenshots for network and firewall

randomername, 2021-12-04:
> what setup, maybe screenshots for network and firewall

ok here it is, please tell me if you need more.
[image]
[image]
[image]
[image]
[image]

[image]