x-wrt/x-wrt #201: usb 4g/5g 网络 mbim协议的，需要关闭软件加速 (natflow)和硬件加速 (hwnat)

ptpt52, 2021-11-29:
usb 4g/5g 网络 mbim协议的，需要关闭软件加速 (natflow)和硬件加速 (hwnat)

目前尚不支持mbim的软件/硬件加速

gsmantenna, 2023-06-09:
现在 X-wrt 支持 高通的 4G/5G QMI  MBIM   的网络加速吗？

ptpt52, 2023-06-09:
不支持。mbim需要关闭natflow

ptpt52, 2023-06-09:
@gsmantenna 或者你有mbim环境吗，可以帮测试一下，我打算支持一下

gsmantenna, 2023-06-10:
我有的， QMI 和 MBIM 都有环境

ptpt52, 2023-06-10:
@gsmantenna  设备型号是什么？

gsmantenna, 2023-06-10:
我手上有 newifi D3  极路由4 （MTK7621A）

ptpt52, 2023-06-10:
@gsmantenna  x-wrt 不支持这个型号呀，只支持 newifi d2

gsmantenna, 2023-06-10:
打错了  不是d3. 是d2

ptpt52, 2023-06-10:
@gsmantenna 
可以测试一下这个固件 https://downloads.x-wrt.com/rom/Downloads/mt7621/x-wrt-23.06-b202306100640-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade.bin
跑 mbim 或者 qmi 跑大流量测速，看看网络通不通。

ssh登陆，echo debug=7 >/dev/natflow_ctl
开启调试，跑流量，dmesg看看日志。

gsmantenna, 2023-06-10:
[image]
现在用的mbim 模式， 移远rm502Q ，联通5G
root@X-WRT:~# echo debug=7 >/dev/natflow_ctl
root@X-WRT:~# dmesg
[    0.000000] Linux version 5.15.114 (ptpt52@SC-GAME) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23276+1-af51b                                                                                          fd67a2) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Fri Jun 9 22:37:20 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is D-Team Newifi D2
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s15376 r8192 d21488 u45056
[    0.000000] pcpu-alloc: s15376 r8192 d21488 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130560
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=0003b23a
[    0.000000] Readback ErrCtl register=0003b23a
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 508028K/524288K available (7071K kernel code, 649K rwdata, 1460K rodata, 1240K init, 251K bss, 16                                                                                          260K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.008050] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066220] pid_max: default: 32768 minimum: 301
[    0.071513] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.078725] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.089734] rcu: Hierarchical SRCU implementation.
[    0.095170] smp: Bringing up secondary CPUs ...
[    0.100524] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.100547] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.100562] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.100612] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.159967] Synchronize counters for CPU 1: done.
[    0.192352] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.192373] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.192384] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.192415] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.251334] Synchronize counters for CPU 2: done.
[    0.281935] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.281954] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.281965] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.282000] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.336531] Synchronize counters for CPU 3: done.
[    0.366381] smp: Brought up 1 node, 4 CPUs
[    0.371547] devtmpfs: initialized
[    0.379640] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.389456] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.396423] pinctrl core: initialized pinctrl subsystem
[    0.403102] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.409727] thermal_sys: Registered thermal governor 'step_wise'
[    0.410311] cpuidle: using governor teo
[    0.432429] FPU Affinity set after 11720 emulations
[... 532 lines cut ...]
[ 1009.048707] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[210.22.155.34:8080->192.168.15.177:57896|ID:67C9,IL:14                                                                                          20|...A....]: dir=1 use hash=3246 outdev=eth0(vlan:1 pppoe=-1)
[ 1009.064825] info: keepalive[4654] nfn[192.168.15.177:57892->153.35.187.2:8080] ct1 diff_jiffies=0 HZ=100 bytes=0 hw=0
[ 1009.080756] info: keepalive[5272] nfn[192.168.15.177:57896->210.22.155.34:8080] ct1 diff_jiffies=0 HZ=100 bytes=0 hw=                                                                                          0
[... cut ...]
[ 1009.496853] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[10.129.180.41:57795->124.130.103.227:8088|ID:F738,IL:4                                                                                          1|...A....]: dir=0 use hash=648 outdev=wwan0(vlan:-1 pppoe=-1)
[ 1009.512877] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[10.129.180.41:57795->124.130.103.227:8088|ID:F738,IL:4                                                                                          1|...A....]: dir=1 use hash=4284 outdev=eth0(vlan:1 pppoe=-1)
root@X-WRT:~#

ptpt52, 2023-06-10:
@gsmantenna  跑流量通的话，就说明没问题。

gsmantenna, 2023-06-10:
我现在5Gnr信号不是很好，极限速度也跑不太快， 我可以用 ECM 模式做环回跑流 iperf3 ，这个模式能offloading 加速吗？

ptpt52, 2023-06-10:
不知道你描述的回环是什么意思，数据从一个网口进，从另外一个网口出，就可以有软件加速。

gsmantenna, 2023-06-10:
就是我这个 移远的 RM500U ECM 模式会有自己的独立的IP地址 192.168.42.1 . 这个模组也可以ssh 进去，然后我 做iperf3 跑流测试 ，newifi-->RM500U （ecm）

ptpt52, 2023-06-10:
可以。电脑---路由---模组 这样跑流量。

gsmantenna, 2023-06-11:
RG200U (RM500U) ---usb3.0 转接板 ---Newifi 3 
RG200U ecm 模式 192.168.42.1 ， 开启iperf3 服务器模式
newifi3 iperf3 客户端 和rg200u打流 
[image]

root@X-WRT:~# echo debug=7 >/dev/natflow_ctl
root@X-WRT:~# dmesg
[    0.000000] Linux version 5.15.114 (ptpt52@SC-GAME) (mipsel-openwrt-linux-mus                                                                                         l-gcc (OpenWrt GCC 12.3.0 r23276+1-af51bfd67a2) 12.3.0, GNU ld (GNU Binutils) 2.                                                                                         40.0) #0 SMP Fri Jun 9 22:37:20 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is D-Team Newifi D2
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 byt                                                                                         es
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s15376 r8192 d21488 u45056
[    0.000000] pcpu-alloc: s15376 r8192 d21488 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130560
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jff                                                                                         s2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, l                                                                                         inear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, lin                                                                                         ear)
[    0.000000] Writing ErrCtl register=0003b2ba
[    0.000000] Readback ErrCtl register=0003b2ba
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 508028K/524288K available (7071K kernel code, 649K rwdata                                                                                         , 1460K rodata, 1240K init, 251K bss, 16260K reserved, 0K cma-reserved, 262144K                                                                                          highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jif                                                                                         fies.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478ab                                                                                         b4, max_idle_ns: 440795247997 ns
[    0.000005] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 43980                                                                                         46511103ns
[    0.008056] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066229] pid_max: default: 32768 minimum: 301
[    0.071531] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linea                                                                                         r)
[    0.078743] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes,                                                                                          linear)
[    0.089861] rcu: Hierarchical SRCU implementation.
[    0.095332] smp: Bringing up secondary CPUs ...
[    0.100731] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.100756] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 byt                                                                                         es
[    0.100772] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.100822] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.160175] Synchronize counters for CPU 1: done.
[    0.193022] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.193041] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 byt                                                                                         es
[    0.193053] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.193082] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.251791] Synchronize counters for CPU 2: done.
[    0.282394] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.282416] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 byt                                                                                         es
[... cut ...]
[  425.212407] info: keepalive[3150] nfn[192.168.15.177:51790->192.168.42.1:22]                                                                                          ct0 diff_jiffies=8 HZ=100 bytes=52 hw=0
[  427.133242] info: keepalive[3150] nfn[192.168.15.177:51790->192.168.42.1:22]                                                                                          ct1 diff_jiffies=192 HZ=100 bytes=196 hw=0
[  427.200955] info: keepalive[3150] nfn[192.168.15.177:51790->192.168.42.1:22]                                                                                          ct0 diff_jiffies=7 HZ=100 bytes=52 hw=0
root@X-WRT:~#

gsmantenna, 2023-06-11:
第二次测试  
pc (iperf3 server 端）-newifi3 -rg200u(ecm)
[dmesg pc-newifi-rg200u.txt](https://github.com/x-
[image]
wrt/x-wrt/files/11714008/dmesg.pc-newifi-rg200u.txt)

ptpt52, 2023-06-11:
不是这样打流吧，应该是客户端iperf3 在电脑上，才行。
电脑(iperf3客户端) --->路由器(x-wrt)---->4G模块（iperf3服务器）

gsmantenna, 2023-06-11:
你说的有道理
新取得log 
打流速度很少能超过30MBps 反向打流反而更高一些

[image]
root@X-WRT:~# dmesg
[12217.046342] info: keepalive[3528] nfn[192.168.15.177:59408->20.242.39.171:443] ct0 diff_jiffies=0 HZ=100 bytes=0 hw=0
[12217.266576] info: keepalive[3528] nfn[192.168.15.177:59408->20.242.39.171:443] ct1 diff_jiffies=22 HZ=100 bytes=0 hw=0
[12217.315683] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59409->52.113.194.132:443|ID:4887,IL:40|...A....]: dir=0 use hash=4816 outdev=usb0(vlan:-1 pppoe=-1)
[12217.331486] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59409->52.113.194.132:443|ID:4887,IL:40|...A....]: dir=1 use hash=5078 outdev=eth0(vlan:1 pppoe=-1)
[12217.347357] info: keepalive[4816] nfn[192.168.15.177:59409->52.113.194.132:443] ct0 diff_jiffies=0 HZ=100 bytes=0 hw=0
[12217.450203] info: keepalive[4816] nfn[192.168.15.177:59409->52.113.194.132:443] ct1 diff_jiffies=10 HZ=100 bytes=0 hw=0
[12217.848460] info: keepalive[7264] nfn[192.168.15.177:49893->155.133.239.51:27031] ct1 diff_jiffies=494 HZ=100 bytes=0 hw=0
[12217.910859] info: keepalive[7264] nfn[192.168.15.177:49893->155.133.239.51:27031] ct0 diff_jiffies=6 HZ=100 bytes=0 hw=0
[12218.453954] info: keepalive[3528] nfn[192.168.15.177:59408->20.242.39.171:443] ct1 diff_jiffies=118 HZ=100 bytes=2766 hw=0
[12218.829132] info: keepalive[3528] nfn[192.168.15.177:59408->20.242.39.171:443] ct0 diff_jiffies=38 HZ=100 bytes=89251 hw=0
[12219.215347] info: keepalive[4294] nfn[192.168.15.177:59397->52.168.112.66:443] ct0 diff_jiffies=622 HZ=100 bytes=1361 hw=0
[12219.510859] info: keepalive[4294] nfn[192.168.15.177:59397->52.168.112.66:443] ct1 diff_jiffies=29 HZ=100 bytes=5655 hw=0
[12219.870006] info: keepalive[1672] nfn[192.168.15.177:57133->192.168.42.1:22] ct1 diff_jiffies=1503 HZ=100 bytes=52 hw=0
[12219.881736] info: keepalive[1672] nfn[192.168.15.177:57133->192.168.42.1:22] ct0 diff_jiffies=1 HZ=100 bytes=0 hw=0
[12220.263097] info: keepalive[3482] nfn[192.168.15.177:51800->39.106.148.116:3478] ct0 diff_jiffies=1003 HZ=100 bytes=0 hw=0
[12220.308711] info: keepalive[3482] nfn[192.168.15.177:51800->39.106.148.116:3478] ct1 diff_jiffies=5 HZ=100 bytes=0 hw=0
[12222.906738] info: keepalive[7264] nfn[192.168.15.177:49893->155.133.239.51:27031] ct1 diff_jiffies=500 HZ=100 bytes=0 hw=0
[12222.973249] info: keepalive[7264] nfn[192.168.15.177:49893->155.133.239.51:27031] ct0 diff_jiffies=6 HZ=100 bytes=0 hw=0
[12223.053363] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:57470->203.208.49.66:443|ID:4709,IL:41|...A....]: dir=0 use hash=3330 outdev=usb0(vlan:-1 pppoe=-1)
[12223.069125] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:57470->203.208.49.66:443|ID:4709,IL:41|...A....]: dir=1 use hash=5645 outdev=eth0(vlan:1 pppoe=-1)
[12223.126649] info: keepalive[3330] nfn[192.168.15.177:57470->203.208.49.66:443] ct1 diff_jiffies=5 HZ=100 bytes=0 hw=0
[12223.595482] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59327->112.228.226.203:8088|ID:00BF,IL:41|...A....]: dir=0 use hash=4327 outdev=usb0(vlan:-1 pppoe=-1)
[12223.611415] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59327->112.228.226.203:8088|ID:00BF,IL:41|...A....]: dir=1 use hash=7616 outdev=eth0(vlan:1 pppoe=-1)
[12223.668623] info: keepalive[4327] nfn[192.168.15.177:59327->112.228.226.203:8088] ct1 diff_jiffies=4 HZ=100 bytes=0 hw=0
[12223.691204] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59326->39.67.228.78:8088|ID:3FA2,IL:41|...A....]: dir=0 use hash=7714 outdev=usb0(vlan:-1 pppoe=-1)
[12223.706866] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59326->39.67.228.78:8088|ID:3FA2,IL:41|...A....]: dir=1 use hash=4358 outdev=eth0(vlan:1 pppoe=-1)
[12223.802116] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59329->112.252.154.209:8088|ID:1206,IL:41|...A....]: dir=0 use hash=2094 outdev=usb0(vlan:-1 pppoe=-1)
[12223.818063] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59329->112.252.154.209:8088|ID:1206,IL:41|...A....]: dir=1 use hash=2312 outdev=eth0(vlan:1 pppoe=-1)
[12224.136600] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59324->112.83.167.1:80|ID:E80C,IL:41|...A....]: dir=0 use hash=7568 outdev=usb0(vlan:-1 pppoe=-1)
[12224.136600] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59325->112.83.167.1:80|ID:E80B,IL:41|...A....]: dir=0 use hash=8082 outdev=usb0(vlan:-1 pppoe=-1)
[12224.136678] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59325->112.83.167.1:80|ID:E80B,IL:41|...A....]: dir=1 use hash=4750 outdev=eth0(vlan:1 pppoe=-1)
[12224.152112] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59324->112.83.167.1:80|ID:E80C,IL:41|...A....]: dir=1 use hash=4237 outdev=eth0(vlan:1 pppoe=-1)
[12224.503227] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59331->39.81.212.38:8088|ID:660C,IL:41|...A....]: dir=0 use hash=6157 outdev=usb0(vlan:-1 pppoe=-1)
[12224.518917] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59331->39.81.212.38:8088|ID:660C,IL:41|...A....]: dir=1 use hash=7464 outdev=eth0(vlan:1 pppoe=-1)
[12224.641208] info: (PCO)[PRE](natflow_path_pre_ct_in_hook:1793)[192.168.42.2:59330->222.133.88.251:8088|ID:805D,IL:41|...A....]: dir=0 use hash=454 outdev=usb0(vlan:-1 pppoe=-1)
[... cut ...]
[12425.009038] info: keepalive[5773] nfn[192.168.15.177:49694->120.55.59.130:443] ct1 diff_jiffies=5 HZ=100 bytes=0 hw=0
[12425.666804] info: keepalive[4290] nfn[192.168.15.177:59638->172.217.160.74:443] ct0 diff_jiffies=298 HZ=100 bytes=0 hw=0
[12426.052883] info: keepalive[7264] nfn[192.168.15.177:49893->155.133.239.51:27031] ct0 diff_jiffies=214 HZ=100 bytes=0 hw=0
[12426.256514] info: keepalive[7264] nfn[192.168.15.177:49893->155.133.239.51:27031] ct1 diff_jiffies=20 HZ=100 bytes=0 hw=0
root@X-WRT:~#

ptpt52, 2023-06-11:
下载速度300Mbps多，是吧。你可以关闭加速 `echo disabled=1 >/dev/natflow_ctl` 测速对比一下。就知道有没有加速了。

gsmantenna, 2023-06-11:
感觉关掉了反而更快一点， 最快有超过36MBps的速度  。红框sh是关闭前的打流

[image]

ptpt52, 2023-06-11:
可以对比cpu使用情况，在路由器上执行
```
mpstat -P ALL 1
```

ptpt52, 2023-06-11:
另外你测试下载速度吧，iperf3 -R

gsmantenna, 2023-06-11:
> 可以对比cpu使用情况，在路由器上执行
> 
> ```
> mpstat -P ALL 1
> ```
cpu 占用来看 natflow 有些效果， 但跑流速度还是上不来 
root@
[image]
X-WRT:~# echo disabled=1  >/dev/natflow_ctl

root@X-WRT:~# mpstat -P ALL 1
Linux 5.15.114 (X-WRT)  06/11/23        _mips_  (4 CPU)

19:44:16     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
19:44:17     all    0.00    0.00    0.00    0.00    0.00    0.50    0.00    0.00   99.50
19:44:17       0    0.00    0.00    0.00    0.00    0.00    0.99    0.00    0.00   99.01
19:44:17       1    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00  100.00
19:44:17       2    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00  100.00
19:44:17       3    0.00    0.00    0.00    0.00    0.00    0.99    0.00    0.00   99.01

19:44:17     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
19:44:18     all    0.50    0.00    1.98    0.00    0.00    0.00    0.00    0.00   97.52
19:44:18       0    0.00    0.00    2.97    0.00    0.00    0.00    0.00    0.00   97.03
19:44:18       1    0.99    0.00    0.00    0.00    0.00    0.00    0.00    0.00   99.01
19:44:18       2    0.00    0.00    1.98    0.00    0.00    0.00    0.00    0.00   98.02
19:44:18       3    0.99    0.00    2.97    0.00    0.00    0.00    0.00    0.00   96.04

19:44:18     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
19:44:19     all    0.00    0.00    0.00    0.00    0.00    0.25    0.00    0.00   99.75
19:44:19       0    0.00    0.00    0.00    0.00    0.00    0.99    0.00    0.00   99.01
19:44:19       1    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00  100.00
19:44:19       2    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00  100.00
19:44:19       3    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00  100.00

19:44:19     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
19:44:20     all    0.00    0.00    4.46    0.00    0.00   17.57    0.00    0.00   77.97
19:44:20       0    0.00    0.00   13.86    0.00    0.00   20.79    0.00    0.00   65.35
19:44:20       1    0.00    0.00    3.96    0.00    0.00    0.00    0.00    0.00   96.04
19:44:20       2    0.00    0.00    0.00    0.00    0.00   49.50    0.00    0.00   50.50
19:44:20       3    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00  100.00

19:44:20     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
19:44:21     all    2.23    0.00    7.67    0.00    0.00   34.41    0.00    0.00   55.69
19:44:21       0    1.98    0.00    7.92    0.00    0.00   43.56    0.00    0.00   46.53
19:44:21       1    2.97    0.00   12.87    0.00    0.00    0.00    0.00    0.00   84.16
19:44:21       2    0.00    0.00    4.95    0.00    0.00   94.06    0.00    0.00    0.99
19:44:21       3    3.96    0.00    4.95    0.00    0.00    0.00    0.00    0.00   91.09

19:44:21     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
19:44:22     all    1.00    0.00    8.50    0.00    0.00   32.00    0.00    0.00   58.50
19:44:22       0    1.00    0.00   24.00    0.00    0.00   31.00    0.00    0.00   44.00
19:44:22       1    2.00    0.00    3.00    0.00    0.00    1.00    0.00    0.00   94.00
19:44:22       2    0.00    0.00    0.00    0.00    0.00   96.00    0.00    0.00    4.00
19:44:22       3    1.00    0.00    7.00    0.00    0.00    0.00    0.00    0.00   92.00

19:44:22     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
19:44:23     all    0.25    0.00    5.69    0.00    0.00   33.66    0.00    0.00   60.40
19:44:23       0    0.00    0.00   11.88    0.00    0.00   41.58    0.00    0.00   46.53
19:44:23       1    0.99    0.00    9.90    0.00    0.00    0.00    0.00    0.00   89.11
19:44:23       2    0.00    0.00    0.00    0.00    0.00   93.07    0.00    0.00    6.93
19:44:23       3    0.00    0.00    0.99    0.00    0.00    0.00    0.00    0.00   99.01

19:44:23     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
19:44:24     all    1.24    0.00    7.43    0.00    0.00   34.41    0.00    0.00   56.93
19:44:24       0    2.97    0.00    4.95    0.00    0.00   44.55    0.00    0.00   47.52
19:44:24       1    0.00    0.00   13.86    0.00    0.00    0.00    0.00    0.00   86.14
19:44:24       2    0.00    0.00    5.94    0.00    0.00   93.07    0.00    0.00    0.99
19:44:24       3    1.98    0.00    4.95    0.00    0.00    0.00    0.00    0.00   93.07

19:44:24     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
[... 334 lines cut ...]
19:46:58       2    0.00    0.00    7.92    0.00    0.00    7.92    0.00    0.00   84.16
19:46:58       3    0.00    0.00    7.92    0.00    0.00    0.00    0.00    0.00   92.08

19:46:58     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
19:46:59     all    0.00    0.00    2.72    0.00    0.00   13.12    0.00    0.00   84.16
19:46:59       0    0.00    0.00    0.00    0.00    0.00   48.51    0.00    0.00   51.49
19:46:59       1    0.00    0.00    0.00    0.00    0.00    0.99    0.00    0.00   99.01
19:46:59       2    0.00    0.00    7.92    0.00    0.00    2.97    0.00    0.00   89.11
19:46:59       3    0.00    0.00    2.97    0.00    0.00    0.00    0.00    0.00   97.03

19:46:59     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
[... cut ...]
19:47:00     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
19:47:01     all    0.00    0.00    0.25    0.00    0.00    0.00    0.00    0.00   99.75
19:47:01       0    0.00    0.00    0.99    0.00    0.00    0.00    0.00    0.00   99.01
19:47:01       1    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00  100.00
19:47:01       2    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00  100.00
19:47:01       3    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00  100.00
^C
root@X-WRT:~#

ptpt52, 2023-06-11:
cpu差别，说明没有加速情况已经达到cpu瓶颈了。 也说明natflow有加速的效果。
另外为什么没有跑上去，瓶颈在哪里，就不好说了，可能是usb，可能是4G模块那边，也可能是其他。

gsmantenna, 2023-06-12:
你这个natflow 用到MTK 的hwnat 吗？
我测过用sdk编译的固件 同一个模块 同一个模式 打流600+ ，现网speedtest实测也可以500
[image]

模组这边也没有瓶颈， 因为模组+usb转接板连接电脑，电脑和模组之间直接打留 可以到1.1G 
[image]

ptpt52, 2023-06-12:
瓶颈可能在usb驱动或者usb模块驱动。

gsmantenna, 2023-06-12:
> 瓶颈可能在usb驱动或者usb模块驱动。

估计你的猜想是正确的，ecm 驱动有瓶颈
我到实网用rm502Q 做了测试， mbim 模式speedtest 可以跑到430Mbps，测速时cpu0占满了

x-wrt newifi 内核版本：5.15.114

4x4 panel antenna ,
root@X-WRT:~# mpstat -P ALL 1
Linux 5.15.114 (X-WRT)  06/12/23        _mips_  (4 CPU)

16:03:19     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest                                               %idle
16:03:20     all    0.25    0.00    1.49    0.00    0.00    1.24    0.00    0.00                                               97.03
16:03:20       0    0.99    0.00    0.99    0.00    0.00    1.98    0.00    0.00                                               96.04
16:03:20       1    0.00    0.00    1.98    0.00    0.00    0.99    0.00    0.00                                               97.03
16:03:20       2    0.00    0.00    0.99    0.00    0.00    0.99    0.00    0.00                                               98.02
16:03:20       3    0.00    0.00    1.98    0.00    0.00    0.99    0.00    0.00                                               97.03

16:03:20     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest                                               %idle
16:03:21     all    0.25    0.00    0.50    0.00    0.00    0.00    0.00    0.00                                               99.26
16:03:21       0    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00
16:03:21       1    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00
16:03:21       2    0.99    0.00    0.99    0.00    0.00    0.00    0.00    0.00                                               98.02
16:03:21       3    0.00    0.00    0.99    0.00    0.00    0.00    0.00    0.00                                               99.01

16:03:21     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest                                               %idle
16:03:22     all    0.00    0.00    0.25    0.00    0.00    0.74    0.00    0.00                                               99.01
16:03:22       0    0.00    0.00    0.99    0.00    0.00    0.99    0.00    0.00                                               98.02
16:03:22       1    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00
16:03:22       2    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00
16:03:22       3    0.00    0.00    0.00    0.00    0.00    1.98    0.00    0.00                                               98.02

16:03:22     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest                                               %idle
16:03:23     all    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00
16:03:23       0    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00
16:03:23       1    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00
16:03:23       2    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00
16:03:23       3    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00

16:03:23     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest                                               %idle
16:03:24     all    0.25    0.00    2.72    0.00    0.00    0.25    0.00    0.00                                               96.78
16:03:24       0    0.00    0.00    1.98    0.00    0.00    0.99    0.00    0.00                                               97.03
16:03:24       1    0.00    0.00    1.98    0.00    0.00    0.00    0.00    0.00                                               98.02
16:03:24       2    0.99    0.00    4.95    0.00    0.00    0.00    0.00    0.00                                               94.06
16:03:24       3    0.00    0.00    1.98    0.00    0.00    0.00    0.00    0.00                                               98.02

16:03:24     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest                                               %idle
16:03:25     all    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00
16:03:25       0    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00
16:03:25       1    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00
16:03:25       2    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00
16:03:25       3    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00

16:03:25     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest                                               %idle
16:03:26     all    0.00    0.00    0.25    0.00    0.00    0.00    0.00    0.00                                               99.75
16:03:26       0    0.00    0.00    0.00    0.00    0.00    0.00    0.00    0.00                                              100.00
[... cut ...]
[  185.644832] warning: (zone_netdev_event:347)dev=wwan0 set zone=127 type=0
[  185.651643] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=wwan0(tso=0,0,hw_csum=0,0), add ingress hook
[  192.466603] ksmbd: The ksmbd server is experimental, use at your own risk.
[  274.899565] mt753x 1e110000.gsw: Port 3 Link is Down
[  279.959682] mt753x 1e110000.gsw: Port 3 Link is Up - 1Gbps/Full
[  295.294572] mt753x 1e110000.gsw: Port 3 Link is Down
[  320.106092] mt753x 1e110000.gsw: Port 3 Link is Up - 1Gbps/Full

[image]

closed 2024-03-04 by ptpt52