x-wrt/x-wrt #208: 小米路由器3大流量就显示ralink_soc_eth 10100000.ethernet eth0: transmit timed out

MortalM, 2021-12-14:
P大你好，我最近在编译小米路由器3的openwrt，期初用lean大的，一开始的时候刷机无限重启，后来是您提交的nand驱动才解决了问题，https://github.com/coolsnowwolf/lede/issues/8405
后来正常刷机后，发现流量大了会有wan超时的情况发生，无论有限还是无线都会超时。
报错信息如下
Thu Dec 9 19:48:30 2021 kern.err kernel: [ 479.229151] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Dec 9 19:48:30 2021 kern.info kernel: [ 479.235757] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Dec 9 19:48:30 2021 kern.info kernel: [ 479.242189] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=064d4000, max=1024, ctx=301, dtx=301, fdx=287, next=301
Thu Dec 9 19:48:30 2021 kern.info kernel: [ 479.253478] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=06434000, max=1024, calc=10, drx=11
Thu Dec 9 19:48:40 2021 kern.err kernel: [ 482.920900] ralink_soc_eth 10100000.ethernet eth0: transmit timed out
Thu Dec 9 19:48:40 2021 kern.info kernel: [ 482.927505] ralink_soc_eth 10100000.ethernet eth0: dma_cfg:80000065
Thu Dec 9 19:48:40 2021 kern.info kernel: [ 482.933936] ralink_soc_eth 10100000.ethernet eth0: tx_ring=0, base=064d4000, max=1024, ctx=255, dtx=255, fdx=252, next=255
Thu Dec 9 19:48:40 2021 kern.info kernel: [ 482.945227] ralink_soc_eth 10100000.ethernet eth0: rx_ring=0, base=05194000, max=1024, calc=528, drx=529

然后我尝试您的固件，发现没有类似的问题，并且我在您这边也找到相关的issue,https://github.com/x-wrt/x-wrt/issues/2
但是我没看懂您到底修复了什么，请问我该如何操作可以解决这个问题。非常感谢。

ptpt52, 2021-12-18:
并不知道什么问题
你可以看看改的记录 
```
git log target/linux/ramips/files/drivers/net/ethernet/ralink
```

MortalM, 2021-12-23:
额你的最新的固件5.10内核也有网口自动link  down的问题，我换到5.4内核的21.10_b202112032317 貌似就正常了，就用这个版本吧。 哎，小米路由器3真是辣鸡啊，不整了。

closed 2021-12-23 by MortalM