x-wrt/x-wrt #215: USB-CDC: config 1 has an invalid interface number: 2 but max is 1

mrninhvn, 2022-01-12:
**Describe the bug**
USB-CDC device not recognized.
X-WRT 21.10 run on Xiaomi router 3.

`dmesg`:
`[ 1689.309297] usb 2-1: new full-speed USB device number 5 using ohci-platform
[ 1689.406129] usb 2-1: config 1 has an invalid interface number: 2 but max is 1
[ 1689.413443] usb 2-1: config 1 has no interface number 0`

USB log:
`T: Bus=02 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#= 6 Spd=12 MxCh= 0
D: Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs= 1
P: Vendor=1915 ProdID=cafe Rev= 1.00
S: Manufacturer=Nordic Semiconductor
S: Product=nRF528xx OpenThread Device
S: SerialNumber=C8DC6B57099E
C:* #Ifs= 2 Cfg#= 1 Atr=c0 MxPwr=500mA
A: FirstIf#= 1 IfCount= 2 Cls=02(comm.) Sub=02 Prot=00
I:* If#= 1 Alt= 0 #EPs= 1 Cls=02(comm.) Sub=02 Prot=00 Driver=(none)
E: Ad=82(I) Atr=03(Int.) MxPS= 64 Ivl=16ms
I:* If#= 2 Alt= 0 #EPs= 2 Cls=0a(data ) Sub=00 Prot=00 Driver=(none)
E: Ad=81(I) Atr=02(Bulk) MxPS= 64 Ivl=0ms
E: Ad=01(O) Atr=02(Bulk) MxPS= 64 Ivl=0ms`

**Screenshots**
[image]

ptpt52, 2022-01-12:
https://forum.openwrt.org/t/usb-cdc-ttyacm-not-found/117129/4
see this?

mrninhvn, 2022-01-13:
> https://forum.openwrt.org/t/usb-cdc-ttyacm-not-found/117129/4 see this?

Yeah, it's me, but it not work.
It not work for me.
`Unknown package 'kmod-usb-acm'.`
`Collected errors:`
`* pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.90-1-6488505ac05478a1b8f4a1fcc993beb8) for kmod-nls-base `
`* pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.90-1-6488505ac05478a1b8f4a1fcc993beb8) for kmod-usb-core `
`* pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.90-1-6488505ac05478a1b8f4a1fcc993beb8) for kmod-usb-acm `
`* pkg_hash_fetch_best_installation_candidate: Packages for kmod-usb-acm found, but incompatible with the architectures configured `
`* opkg_install_cmd: Cannot install package kmod-usb-acm.`

I asked on discord and they said to create an issue.

ptpt52, 2022-01-13:
you have to build firmware from source to package kmod-usb-acm

mrninhvn, 2022-01-13:
> you have to build firmware from source to package kmod-usb-acm

Actually, I just checked again, it should appear as /dev/ttyUSB, any ideas?

mrninhvn, 2022-01-17:
> you have to build firmware from source to package kmod-usb-acm

I am new to openwrt, do you have any tutorial to do this? Thank you agian.

ptpt52, 2022-01-18:
docs about how to build firmware from source:
https://blog.x-wrt.com/docs/build/

mrninhvn, 2022-01-19:
> 

I got that error when building:
`mkdir -p /home/oem/matter/x-wrt/staging_dir/hostpkg/stamp`
`touch /home/oem/matter/x-wrt/build_dir/hostpkg/gettext-0.21/.built`
`touch /home/oem/matter/x-wrt/staging_dir/hostpkg/stamp/.gettext-full_installed`
`make[3]: Leaving directory '/home/oem/matter/x-wrt/package/libs/gettext-full'`
`time: package/libs/gettext-full/host-compile#445.98#77.80#863.79`
`make[2]: Leaving directory '/home/oem/matter/x-wrt'`
`make[1]: *** [package/Makefile:110: /home/oem/matter/x-wrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2`
`make[1]: Leaving directory '/home/oem/matter/x-wrt'`
`make: *** [/home/oem/matter/x-wrt/include/toplevel.mk:230: world] Error 2`

Same for both master code and sdk.

mrninhvn, 2022-01-19:
> > you have to build firmware from source to package kmod-usb-acm
> 
> I am new to openwrt, do you have any tutorial to do this? Thank you agian.

I got that error when building:
`mkdir -p /home/oem/matter/x-wrt/staging_dir/hostpkg/stamp`
`touch /home/oem/matter/x-wrt/build_dir/hostpkg/gettext-0.21/.built`
`touch /home/oem/matter/x-wrt/staging_dir/hostpkg/stamp/.gettext-full_installed`
`make[3]: Leaving directory '/home/oem/matter/x-wrt/package/libs/gettext-full'`
`time: package/libs/gettext-full/host-compile#445.98#77.80#863.79`
`make[2]: Leaving directory '/home/oem/matter/x-wrt'`
`make[1]: *** [package/Makefile:110: /home/oem/matter/x-wrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2`
`make[1]: Leaving directory '/home/oem/matter/x-wrt'`
`make: *** [/home/oem/matter/x-wrt/include/toplevel.mk:230: world] Error 2`

Same for both master code and sdk.

mrninhvn, 2022-01-19:
> docs about how to build firmware from source: https://blog.x-wrt.com/docs/build/

I got that error when building:
`mkdir -p /home/oem/matter/x-wrt/staging_dir/hostpkg/stamp`
`touch /home/oem/matter/x-wrt/build_dir/hostpkg/gettext-0.21/.built`
`touch /home/oem/matter/x-wrt/staging_dir/hostpkg/stamp/.gettext-full_installed`
`make[3]: Leaving directory '/home/oem/matter/x-wrt/package/libs/gettext-full'`
`time: package/libs/gettext-full/host-compile#445.98#77.80#863.79`
`make[2]: Leaving directory '/home/oem/matter/x-wrt'`
`make[1]: *** [package/Makefile:110: /home/oem/matter/x-wrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2`
`make[1]: Leaving directory '/home/oem/matter/x-wrt'`
`make: *** [/home/oem/matter/x-wrt/include/toplevel.mk:230: world] Error 2`

Same for both master code and sdk.

mrninhvn, 2022-01-21:
Build kmod-usb-acm with source solves the problem. Close issue.

closed 2022-01-21 by mrninhvn