x-wrt/x-wrt #216: Rom Stock

simpsss, 2022-01-13:
Hello, the WIFI of my Xiaomi Router 3G only drops, how do I go back to my Stock rom?

Thanks

simpsss, 2022-01-13:
or send me a version with the bootloader unlocked so I can install Breed

ptpt52, 2022-01-13:
download firmware from https://github.com/x-wrt/build-release/releases/download/v21.10_b202201032246/x-wrt-21.10_b202201032246-ramips-mt7621.zip

unzip
and take install the mir3g firmware first

there is a `kmod-mtd-rw*.ipk` you could install it use it to unlock Bootloader

see how to use kmod-mtd-rw: https://github.com/jclehner/mtd-rw

simpsss, 2022-01-13:
it worked, thanks!

closed 2022-01-15 by simpsss

17Chung21, 2022-08-13:
> download firmware from https://github.com/x-wrt/build-release/releases/download/v21.10_b202201032246/x-wrt-21.10_b202201032246-ramips-mt7621.zip
>
> unzip and take install the mir3g firmware first [...]

My Xiaomi R3, use the your above method:
-Download firmware from https://github.com/x-wrt/build-release/releases/download/v21.10_b202201032246/x-wrt-21.10_b202201032246-ramips-mt7620.zip
-Unzip and take upgrade the mir3 firmware.
-Found "kmod-mtd-rw_5.4.188+git-20160214-2_mipsel_24kc.ipk" but can not install .ipk; Aways error "cannot find dependency kernel" (code 255)