x-wrt/x-wrt #222: Can't install kernel modules with opkg

gitnewcomer, 2022-02-11:
I wanted to install Strongswan, but it fails because the kernel version isn't compatible:

`root@X-WRT:~# opkg update
Downloading http://downloads.openwrt.org/snapshots/targets/ramips/mt7620/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_core
Downloading http://downloads.openwrt.org/snapshots/targets/ramips/mt7620/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_base
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_luci
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_packages
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_routing
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_telephony
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
root@X-WRT:~# opkg install strongswan
Installing strongswan (5.9.5-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/strongswan_5.9.5-1_mipsel_24kc.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-crypto-null
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-crypto-aead
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-crypto-manager
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-crypto-authenc
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-authenc found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-crypto-cbc
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-lib-zlib-inflate
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-lib-zlib-inflate found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-lib-zlib-deflate
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-lib-zlib-deflate found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-crypto-acompress
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-acompress found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-crypto-deflate
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-deflate found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-crypto-des
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-crypto-echainiv
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-echainiv found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-crypto-hmac
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-crypto-md5
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-crypto-sha1
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-ipsec
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipsec found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-iptunnel4
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-ipsec4
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipsec4 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-iptunnel6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-ipsec6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipsec6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-nf-ipt
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-ipt-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573) for kmod-ipt-ipsec
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for strongswan:
 *      kernel (= 5.10.96-1-a67146b476b41da0b57084bd696d0573)
 * opkg_install_cmd: Cannot install package strongswan.
`
Is there another package source with compiled kernel modules that are compatible with X-WRT?
Thanks!

ptpt52, 2022-02-11:
it can't.

you have to build your firmware via source

gitnewcomer, 2022-02-13:
> it can't.
> 
> you have to build your firmware via source

Why isn't is possible? Using X-WRT without having the possibility to install kernel modules without needing to compile it ourselfs makes no sense to me. PandoraBox and others had the possibility to install kernel modules because they had their own repository that was compiled at the same time as the firmware, like this example: http://downloads.pangubox.com:6380/pandorabox/19.01/targets/ralink/mt7620/packages/

vhick, 2022-06-08:
I hope they implement their own package source. I need to install sqm and other kernel related software.

Thank you.

kokroo, 2022-07-31:
@ptpt52 Can you implement your own package sources, please?

thanasxda, 2022-10-24:
sometimes for example some devices to use a comparison miwifi mini and router r3 use the same hardware part from storage and sometimes happen to run the same kernel even. however due to the kernelname i have to compile from source. would it be possible in this case to spoof the kernelname and if so how? ive looked around the config files to modify it without luck. it would be quite convenient in cases such as this one.