x-wrt/x-wrt #238: Any support for IPv6 hardware offloading?

jionkitten, 2022-04-17:
The natflow already supports IPv4 hwnat, but it seems that it doesn't support IPv6. I noticed that the official Openwrt has added support for IPv6 hardware offloading https://github.com/openwrt/openwrt/commit/e3166643fbc9f3cf67948c660a76b5cc60bf0ef8 for MT7621, but I‘m not sure if it will also take effect on x-wrt. If that's not the case, is there any possibility to support it? Since IPv6 hardware offloading doesn't involve any NAT stuff and it should be much easier to implement compared to IPv4 hwnat. Thanks for your awesome work.

ptpt52, 2022-04-17:
暂不支持
Not currently supported

padavanonly, 2022-04-24:
> 暂不支持
> Not currently supported

上游op为mtk7622增加的wed加速有考虑吗

ptpt52, 2022-04-24:
暂不考虑，兼容起来比较困难

ptpt52, 2022-05-28:
开始支持ipv6 软件转发了