x-wrt/x-wrt #240: 在线编译的固件无法汉化，部分功能无法启用

lon91ong, 2022-04-20:
**源码分支**：[x-wrt/x-wrt](https://github.com/x-wrt/x-wrt/tree/master)
**路由器**：硬改SPI闪存的HiWifi-B70

编译**配置文件**：[config.buildinfo](https://github.com/x-wrt/x-wrt/files/8518158/config.buildinfo.txt)
参考HC5962和newifi-d2修改得到的**DTS文件**：[hc5962-spi.dts](https://github.com/lon91ong/Actions-OpenWrt/raw/5.4.x/Others/B70/hc5962.dts)

#### 存在的问题

1：界面汉化不全，如下图所示：
[image]
配置完没有效果：
[image]
putty启动界面乱码
[image]

2：Flow Offloading 和 FullCone NAT 无法启用
[image]

请大侠指点

ptpt52, 2022-04-20:
https://blog.x-wrt.com/docs/build/

lon91ong, 2022-04-20:
> https://blog.x-wrt.com/docs/build/

本地编译成功率实在是太低, 网络环境太差, 动不动就资源下载不了...
能否提供[x-wrt/build-release](https://github.com/x-wrt/build-release)项目在线编译时生成的`config.buildinfo`文件, 可以参考着手动修改`config`

lon91ong, 2022-04-21:
@ptpt52 希望参考[P3TERX/Actions-OpenWrt](https://github.com/P3TERX/Actions-OpenWrt)改进一下[x-wrt/build-release](https://github.com/x-wrt/build-release)项目的在线编译逻辑
目前的编译使用[main.yml](https://github.com/x-wrt/build-release/tree/master/.github/workflows/main.yml)初始化环境，通过`build-release/build.sh`调用[com.x-wrt/rom/lede/make.sh](https://github.com/x-wrt/com.x-wrt/blob/master/rom/lede/make.sh)执行编译，再回头调用 `build-release/upload.sh`进行上传

而[P3TERX/Actions-OpenWrt](https://github.com/P3TERX/Actions-OpenWrt)项目几乎所有的动作都是在一个`build-openwrt.yml`文件中完成的，逻辑更加清晰。

以上建议仅供参考。

zhuxiaole, 2022-04-27:
可以参考我的看看
https://github.com/zhuxiaole/Build-OpenWrt

zhuxiaole, 2022-04-27:
中文问题主要是下面的配置
CONFIG_LUCI_LANG_zh_Hans=y
然后每个luci-app-*的插件，都要加个语言的配置，例如
CONFIG_PACKAGE_luci-app-autoreboot=y
CONFIG_PACKAGE_luci-i18n-autoreboot-zh-cn=y

zhuxiaole, 2022-04-27:
Flow Offloading 和 FullCone NAT，我加到x-wrt里也启动不起来。
后来还是改用了x-wrt里面的natcap

lon91ong, 2022-04-27:
> 中文问题主要是下面的配置 CONFIG_LUCI_LANG_zh_Hans=y CONFIG_PACKAGE_luci-i18n-base-zh-cn=y 然后每个luci-app-*的插件，都要加个语言的配置，例如 CONFIG_PACKAGE_luci-app-autoreboot=y CONFIG_PACKAGE_luci-i18n-autoreboot-zh-cn=y
@zhuxiaole 这样能解决SSH中文乱码的问题吗？

zhuxiaole, 2022-04-27:
不知道，没遇到过这问题。看着好像是banner乱码了。或者换个ssh客户端看看？

zhuxiaole, 2022-04-27:
或者跟我一张样，自定义banner试试
rm -rf package/base-files/files/etc/banner
cp $GITHUB_WORKSPACE/xwrt/devices/common/banner package/base-files/files/etc/