x-wrt/x-wrt #242: ipq40xx: 补丁缺失(?)导致性能骤降

PussAzuki, 2022-04-21:
**Describe the bug**
网络性能很差

**To Reproduce**
Steps to reproduce the behavior:
1. 下载20220409的固件
2. 刷到路由器上
3. 网页speedtest测速，仅100Mbps（不是94Mbps）

**Expected behavior**
之前补丁在的时候，pppoe下speedtest测速能有940Mbps

**Additional context**
经过与补丁齐全的202201140818对比，发现缺失了：
https://github.com/x-wrt/x-wrt/commit/e985fe6500093ea637c431f2499f9574527082ae
https://github.com/x-wrt/x-wrt/commit/4feec1c08fea36ebf818565f60865785a934b041

我也尝试手动应用到openwrt主线上，配合natflow后至少pppoe性能上没有问题，但是内网只有680Mbps

ptpt52, 2022-04-21:
@PussAzuki 
不用这两个补丁，对性能影响不是很大呀。理论上。

PussAzuki, 2022-04-21:
那我就不是很懂了，我看到ipq40xx下的不同主要就这两个，其他的应该没啥大不同才对

PussAzuki, 2022-04-22:
后面试了下往主线上只应用了：
https://github.com/x-wrt/x-wrt/commit/d7596fc74bac045634f4bb74bee4a94b90bb1735
https://github.com/x-wrt/x-wrt/commit/01f9b8eb7e7b58f3d426c1eb9c453ac9597dd015
https://github.com/x-wrt/x-wrt/commit/44ee6343c34a60b063d8f9ea1afb20012c298ee7
https://github.com/x-wrt/x-wrt/commit/70b2055a5760c197d63492f4583837d9367cacfa
https://github.com/openwrt/openwrt/pull/9731

然后网络部分只加了6in4和nat64及natflow
然后按照qsdk给edma设置irq：
edma_eth_tx{0-3} 设置为使用CPU2
edma_eth_tx{4-7} 设置为使用CPU3
edma_eth_tx{8-11} 设置为使用CPU0
edma_eth_tx{12-15} 设置为使用CPU1
edma_eth_rx0 设置为使用CPU0
edma_eth_rx2 设置为使用CPU1
edma_eth_rx4 设置为使用CPU2
edma_eth_rx6 设置为使用CPU3

pppoe下speedtest最高950Mbps（不设置irq时CPU0会吃满然后只能跑800+Mbps）
iperf3测速也能跑940Mbps（应用这个[补丁](https://github.com/x-wrt/x-wrt/commit/e985fe6500093ea637c431f2499f9574527082ae)的话，内网只有500Mbps）

所以我也不太懂为什么0409的固件性能大降...

然后464xlat和ds-lite默认会被启用，然后这个设置pppoe之后就有虚拟端口就不断重复 setting up/down （openwrt文档说是这两种网络是自动设置的，想禁用还得手动设置），能不能暂时删除掉呢？
个人觉得留个6in4和nat64就够了

andredpe1, 2022-04-22:
> @PussAzuki 不用这两个补丁，对性能影响不是很大呀。理论上。

您好，我是您编译的忠实粉丝，我来自巴西，我来询问是否可以尝试在您的编译中为 linksys ea6350 v3 开发或应用一些修正补丁或 irq 平衡，因为 pppe 连接中的相同仅饱和一个 irq ，与 wifi 相关，如果可以进行任何修正，因为 2.4ghz wifi 与库存固件相比是不可行的。如果可能的话，留下一些联系，这样我就可以和你谈谈这件事。

PussAzuki, 2022-04-22:
> > @PussAzuki 不用这两个补丁，对性能影响不是很大呀。理论上。
> 
> 您好，我是您编译的忠实粉丝，我来自巴西，我来询问是否可以尝试在您的编译中为 linksys ea6350 v3 开发或应用一些修正补丁或 irq 平衡，因为 pppe 连接中的相同仅饱和一个 irq ，与 wifi 相关，如果可以进行任何修正，因为 2.4ghz wifi 与库存固件相比是不可行的。如果可能的话，留下一些联系，这样我就可以和你谈谈这件事。

但我对edma和pppoe的知识都是基于qsdk源码的，我现在这台路由器也不使用无线了，WiFi相关爱莫能助

ptpt52, 2022-04-22:
@PussAzuki  对的性能问题应该就是个cpu绑定的问题而已，和交换片补丁关系不大

ptpt52, 2022-04-22:
@PussAzuki 
你试试这个 https://github.com/x-wrt/x-wrt/commit/f360272d9f0ec0004368e7828b77e9f38ed79dbc
绑定cpu分配的脚本。

andredpe1, 2022-04-22:
@ptpt52 
这些修复程序是否已包含在您的 lnksys ea6350 v3 编译中？
您能否帮助我了解如何添加此脚本，因为我对脚本不是很有经验，我在哪里添加它们

ptpt52, 2022-04-22:
> @ptpt52 这些修复程序是否已包含在您的 lnksys ea6350 v3 编译中？ 您能否帮助我了解如何添加此脚本，因为我对脚本不是很有经验，我在哪里添加它们

下个发布版本我尝试增加它们。

andredpe1, 2022-04-22:
@ptpt52 
如果可能的话，你可以分析这个 https://github.com/openwrt/openwrt/pull/9731 并包含它，因为我们这个模型的用户在这个设备的 wifi 方面遇到了几个问题。

andredpe1, 2022-04-22:
@ptpt52 
如果我的写作难以理解，请原谅我，因为我的母语是葡萄牙语，而且我正在使用翻译

andredpe1, 2022-04-22:
@ptpt52 
https://github.com/openwrt/openwrt/pull/9731

PussAzuki, 2022-04-22:
> @PussAzuki 你试试这个 [f360272](https://github.com/x-wrt/x-wrt/commit/f360272d9f0ec0004368e7828b77e9f38ed79dbc) 绑定cpu分配的脚本。

我之前一直用的这个，但总觉得和QSDK原版有些出入...于是手动处理了下，加入到 /etc/rc.local 实现自启

ptpt52, 2022-04-22:
@PussAzuki  我试了下，绑定cpu后，pppoe 下行也就跑800那样，上行好点能900

PussAzuki, 2022-04-22:
因为我用128M版本，然后合入如下补丁：

https://github.com/x-wrt/x-wrt/commit/d7596fc74bac045634f4bb74bee4a94b90bb1735
https://github.com/x-wrt/x-wrt/commit/01f9b8eb7e7b58f3d426c1eb9c453ac9597dd015
https://github.com/x-wrt/x-wrt/commit/44ee6343c34a60b063d8f9ea1afb20012c298ee7
https://github.com/x-wrt/x-wrt/commit/70b2055a5760c197d63492f4583837d9367cacfa
https://github.com/openwrt/openwrt/pull/9731
[超频补丁](https://github.com/coolsnowwolf/lede/blob/master/target/linux/ipq40xx/patches-5.4/999-ipq40xx-unlock-cpu-frequency.patch)

然后加入了仅 kmod-natflow ，并按照之前讨论的设置手动自启：
`echo disabled=0 >/dev/natflow_ctl`

手动调整IRQ：
```
echo 4 > /proc/irq/72/smp_affinity
echo 4 > /proc/irq/73/smp_affinity
echo 4 > /proc/irq/74/smp_affinity
echo 4 > /proc/irq/75/smp_affinity
echo 8 > /proc/irq/76/smp_affinity
echo 8 > /proc/irq/77/smp_affinity
echo 8 > /proc/irq/78/smp_affinity
echo 8 > /proc/irq/79/smp_affinity
echo 1 > /proc/irq/80/smp_affinity
echo 1 > /proc/irq/81/smp_affinity
echo 1 > /proc/irq/82/smp_affinity
echo 1 > /proc/irq/83/smp_affinity
echo 2 > /proc/irq/84/smp_affinity
echo 2 > /proc/irq/85/smp_affinity
echo 2 > /proc/irq/86/smp_affinity
echo 2 > /proc/irq/87/smp_affinity
echo 1 > /proc/irq/88/smp_affinity
echo 2 > /proc/irq/90/smp_affinity
echo 4 > /proc/irq/92/smp_affinity
echo 8 > /proc/irq/94/smp_affinity
```
然后实际表现符合预期：
[image]

不过我还注意到一个现象：如果路由器下面还有一个路由器，那么跑speedtest测速的时候，主路由器的CPU负载会很低（至少htop显示如此）

andredpe1, 2022-04-22:
> 然后加入了仅 kmod-natflow ，并按照之前讨论的设置手动自启：

您正在使用特定版本或使用 openwrt 默认值，因为您无法应用任何修复。

andredpe1, 2022-04-23:
@ptpt52 
你能给我一个带有这些更正路径和一个非常好的 irqbalance 的版本的发布日期吗？

ptpt52, 2022-04-23:
@andredpe1  new release is now in progress: https://github.com/x-wrt/build-release/releases/tag/v22.03_b202204240424

andredpe1, 2022-04-23:
> @andredpe1 nova versão está em andamento: https://github.com/x-wrt/build-release/releases/tag/v22.03_b202204240424

謝謝，如果可能的話添加以下補丁 

https://github.com/x-wrt/x-wrt/commit/d7596fc74bac045634f4bb74bee4a94b90bb1735
https://github.com/x-wrt/x-wrt/commit/01f9b8eb7e7b58f3d426c1eb9c453ac9597dd015
https://github.com/x-wrt/x-wrt/commit/44ee6343c34a60b063d8f9ea1afb20012c298ee7
https://github.com/x-wrt/x-wrt/commit/70b2055a5760c197d63492f4583837d9367cacfa
https://github.com/openwrt/openwrt/pull/9731
[超频补丁](https://github.com/coolsnowwolf/lede/blob/master/target/linux/ipq40xx/patches-5.4/999-ipq40xx-unlock-cpu-frequency.patch)

PussAzuki, 2022-04-24:
https://github.com/x-wrt/x-wrt/commit/82b75cf9984ec94064f164b2d2b88d27aea2a3f6
@ptpt52 如何判断这个是否需要？🤔
目前没改过，但看起来没问题啊，qsdk那边也没要求改顺序

ptpt52, 2022-04-24:
> [82b75cf](https://github.com/x-wrt/x-wrt/commit/82b75cf9984ec94064f164b2d2b88d27aea2a3f6) @ptpt52 如何判断这个是否需要？🤔 目前没改过，但看起来没问题啊，qsdk那边也没要求改顺序

这个是优化一下中断处理流程，可选。

PussAzuki, 2022-04-24:
那就结果而言，有没有能区别是否优化了的检测方法？

ptpt52, 2022-04-24:
都是理论上的，尽快ack irq寄存器是更好的选择。

PussAzuki, 2022-04-24:
问个问题：https://github.com/openwrt/openwrt/pull/4721 这里面的DSA补丁是否有些问题？
我测试下来带上 https://github.com/LGA1150/openwrt/commit/89106efc0d25fb356dff221a8f544967ac88c6f4 和natflow及[超频补丁](https://github.com/coolsnowwolf/lede/blob/master/target/linux/ipq40xx/patches-5.4/999-ipq40xx-unlock-cpu-frequency.patch) pppoe下测速最多860Mbps，这是驱动不完善还是？

我看这个PR出来之时你对它有些兴趣的，所以来问问

ptpt52, 2022-04-24:
没细看他们改什么，无论怎么改对咱没什么影响。目前这个ipq40xx的交换片功能暂时废了，不想维护了
他们新的dsa驱动怎么实现的，没有看细节。

PussAzuki, 2022-05-07:
ipq40xx系列好像以nftables的offload性能奇低...（至少我基于官方snapshot有这样的问题）
跑speedtest测试 ：
1.基于nftables 有安装 kmod-nft-offload 和 kmod-natflow
不开natflow和fw4里的offload，下载190Mbps多点；只开fw4的offload，下载220Mbps多点；开natflow，跑满。
2.基于iptables 有安装kmod-ipt-offload 和 kmod-natflow
不开natflow和fw3里的offload，下载260Mbps多点；只开fw3的offload，下载680Mbps多点，开natflow，跑满。

目前只见到openwrt的github这边issue区有个mt7621的报告了切换到nftables之后offload性能下降，我都不知道是不是就我一个人的问题了？

PussAzuki, 2022-05-21:
破案了，昨天看到禁用ethX的gro的commit，就试了下，速度回来了

closed 2022-05-21 by PussAzuki

gw826943555, 2023-04-21:
> 破案了，昨天看到禁用ethX的gro的commit，就试了下，速度回来了

怎么禁用，自家PPPOE 500M都跑不满 换成了mt7621在用

Ais-Danny, 2023-12-28:
> 因为我用竞斗云的128M版本，然后合入如下补丁：
> 
> https://github.com/x-wrt/x-wrt/commit/d7596fc74bac045634f4bb74bee4a94b90bb1735
> https://github.com/x-wrt/x-wrt/commit/01f9b8eb7e7b58f3d426c1eb9c453ac9597dd015
> https://github.com/x-wrt/x-wrt/commit/44ee6343c34a60b063d8f9ea1afb20012c298ee7
> https://github.com/x-wrt/x-wrt/commit/70b2055a5760c197d63492f4583837d9367cacfa
> https://github.com/openwrt/openwrt/pull/9731
> [超频补丁](https://github.com/coolsnowwolf/lede/blob/master/target/linux/ipq40xx/patches-5.4/999-ipq40xx-unlock-cpu-frequency.patch)
> 
> 然后加入了仅 kmod-natflow ，并按照之前讨论的设置手动自启：
> `echo disabled=0 >/dev/natflow_ctl`
> 
> 手动调整IRQ：
> ```
> echo 4 > /proc/irq/72/smp_affinity
> echo 4 > /proc/irq/73/smp_affinity
> echo 4 > /proc/irq/74/smp_affinity
> echo 4 > /proc/irq/75/smp_affinity
> echo 8 > /proc/irq/76/smp_affinity
> echo 8 > /proc/irq/77/smp_affinity
> echo 8 > /proc/irq/78/smp_affinity
> echo 8 > /proc/irq/79/smp_affinity
> echo 1 > /proc/irq/80/smp_affinity
> echo 1 > /proc/irq/81/smp_affinity
> echo 1 > /proc/irq/82/smp_affinity
> echo 1 > /proc/irq/83/smp_affinity
> echo 2 > /proc/irq/84/smp_affinity
> echo 2 > /proc/irq/85/smp_affinity
> echo 2 > /proc/irq/86/smp_affinity
> echo 2 > /proc/irq/87/smp_affinity
> echo 1 > /proc/irq/88/smp_affinity
> echo 2 > /proc/irq/90/smp_affinity
> echo 4 > /proc/irq/92/smp_affinity
> echo 8 > /proc/irq/94/smp_affinity
> ```
> 然后实际表现符合预期：
> [image]
> 
> 不过我还注意到一个现象：如果路由器下面还有一个路由器，那么跑speedtest测速的时候，主路由器的CPU负载会很低（至少htop显示如此）

我现在也是使用这这款设备，竞斗云2.0，

内网WiFi测速可以达到七八百，一上公网测速就寄了（只有240左右，300兆下行带宽，用有线机器测了公网能跑340）。