x-wrt/x-wrt #250: 微调ipq40xx的TX的irq策略

PussAzuki, 2022-05-22:
```
echo 1 > /proc/irq/72/smp_affinity
echo 2 > /proc/irq/73/smp_affinity
echo 4 > /proc/irq/74/smp_affinity
echo 8 > /proc/irq/75/smp_affinity
echo 1 > /proc/irq/76/smp_affinity
echo 2 > /proc/irq/77/smp_affinity
echo 4 > /proc/irq/78/smp_affinity
echo 8 > /proc/irq/79/smp_affinity
echo 1 > /proc/irq/80/smp_affinity
echo 2 > /proc/irq/81/smp_affinity
echo 4 > /proc/irq/82/smp_affinity
echo 8 > /proc/irq/83/smp_affinity
echo 1 > /proc/irq/84/smp_affinity
echo 2 > /proc/irq/85/smp_affinity
echo 4 > /proc/irq/86/smp_affinity
echo 8 > /proc/irq/87/smp_affinity
echo 1 > /proc/irq/88/smp_affinity
echo 2 > /proc/irq/90/smp_affinity
echo 4 > /proc/irq/92/smp_affinity
echo 8 > /proc/irq/94/smp_affinity
```
目前脚本实现的是这个效果，关掉natflow并使用nftables的offload只有620Mbps（单层NAT环境下 speedtest.net 的结果）

```
echo 4 > /proc/irq/72/smp_affinity
echo 4 > /proc/irq/73/smp_affinity
echo 4 > /proc/irq/74/smp_affinity
echo 4 > /proc/irq/75/smp_affinity
echo 8 > /proc/irq/76/smp_affinity
echo 8 > /proc/irq/77/smp_affinity
echo 8 > /proc/irq/78/smp_affinity
echo 8 > /proc/irq/79/smp_affinity
echo 1 > /proc/irq/80/smp_affinity
echo 1 > /proc/irq/81/smp_affinity
echo 1 > /proc/irq/82/smp_affinity
echo 1 > /proc/irq/83/smp_affinity
echo 2 > /proc/irq/84/smp_affinity
echo 2 > /proc/irq/85/smp_affinity
echo 2 > /proc/irq/86/smp_affinity
echo 2 > /proc/irq/87/smp_affinity
```
而像QSDK在qca-edma里的这样，tx{0,1,2,3} 放到CPU2，tx{4,5,6,7}放到CPU3，tx{8,9,10,11}放到CPU0，tx{12,13,14,15}放到CPU1，rx还是CPU0,1,2,3的顺序时，能顺利达到650~670Mbps（单层NAT环境下 speedtest.net 的结果）

虽然使用natflow时都能跑满，但是试了下DSA PR里的驱动，发现它是 直接绑四个tx为一组，共四组 所以或许提前改一下能方便后面切换驱动？

PussAzuki, 2022-05-22:
不过无论是现在在用的还是微调后的irq策略，tx{1,3,5,7,9,11,13,15} 在 /proc/interrupts 里的值都很低，我怀疑仅仅设置个irq还是不足的...

[image]

ptpt52, 2022-05-23:
原则上看 txq 和网口对应关系
比如现在5.10内核的驱动，分为 eth0 eth1 两个
eth0 实际用的是哪些tx { q0,q1,q4,q5,q8,q9,q12,q13 }
最合适的是把 这8个q 分别绑定 4个cpu
比如 
cpu0 --> q0 q1
cpu1 --> q4 q5

andredpe1, 2022-05-24:
@ptpt52 你能用 sqm 和 irqbalance 为 ipq4018 编译一些版本吗？我有缓冲膨胀问题

lin13051275755, 2023-02-17:
新人求教
[image]
这里需要配置吗？
现在遇到个问题是10台设备mesh组网，每台设备iperf3 -b 10M 就基本不通了
请教大神们，需要配置什么？@ptpt52

ptpt52, 2023-02-17:
> 新人求教 [image] 这里需要配置吗？ 现在遇到个问题是10台设备mesh组网，每台设备iperf3 -b 10M 就基本不通了 请教大神们，需要配置什么？@ptpt52

是fakemesh嘛，你用fakemesh吧。 fakemesh.com

lin13051275755, 2023-02-20:
> > 新人求教 [image] 这里需要配置吗？ 现在遇到个问题是10台设备mesh组网，每台设备iperf3 -b 10M 就基本不通了 请教大神们，需要配置什么？@ptpt52
> 
> 是fakemesh嘛，你用fakemesh吧。fakemesh.com

好的，谢谢！去学习一下

closed 2023-03-27 by PussAzuki