x-wrt/x-wrt #251: Mtk_soc_eth watchdog timeout

guoqiang13, 2022-06-02:
[2733116.802884] ------------[ cut here ]------------
[2733116.812473] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:448 0x8047e780
[2733116.826901] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[2733116.841147] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT wireguard pppox ppp_generic nf_nat_pptp nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 libblake2s ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY ts_kmp ts_fsm ts_bm slhc sch_cake poly1305_mips nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libblake2s_generic iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat chacha_mips br_netfilter sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit
[2733116.841336]  act_mirred ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel kpp leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[2733117.117908] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G    B   W         5.10.107 #0
[2733117.132967] Stack : 00000000 80840000 ffffffff 8007d6e0 00000000 00000000 00000000 00000000
[2733117.149937]         00000000 00000000 00000000 00000000 00000000 00000001 87c0dd50 5fc2ce17
[2733117.166908]         87c0dde8 00000000 00000000 00000000 00000038 805e1804 312e342e 23203131
[2733117.183876]         00000000 00021330 00000000 0001cc94 00000000 87c0dd30 00000000 8047e780
[2733117.200851]         00000009 00000000 00200000 00000122 00000001 80359e2c 00000000 80810000
[2733117.217822]         ...
[2733117.223027] Call Trace:
[2733117.223043] [<8007d6e0>] 0x8007d6e0
[2733117.235507] [<805e1804>] 0x805e1804
[2733117.242793] [<8047e780>] 0x8047e780
[2733117.250072] [<80359e2c>] 0x80359e2c
[2733117.257352] [<8000b05c>] 0x8000b05c
[2733117.264633] [<8000b064>] 0x8000b064
[2733117.271907] [<805c6f9c>] 0x805c6f9c
[2733117.279178] [<8007d8ac>] 0x8007d8ac
[2733117.286449] [<8002bfe8>] 0x8002bfe8
[2733117.293729] [<8047e780>] 0x8047e780
[2733117.301004] [<8002c0c0>] 0x8002c0c0
[2733117.308295] [<8047e780>] 0x8047e780
[2733117.315579] [<800a9018>] 0x800a9018
[2733117.322855] [<8047e484>] 0x8047e484
[2733117.330133] [<800965d4>] 0x800965d4
[2733117.337415] [<8007f4c0>] 0x8007f4c0
[2733117.344693] [<80429908>] 0x80429908
[2733117.351966] [<8009681c>] 0x8009681c
[2733117.359245] [<80433bc0>] 0x80433bc0
[2733117.366520] [<80083d2c>] 0x80083d2c
[2733117.373805] [<805e7d1c>] 0x805e7d1c
[2733117.381084] [<80030768>] 0x80030768
[2733117.388355] [<802f8404>] 0x802f8404
[2733117.395626] [<80006c28>] 0x80006c28
[2733117.402895] 
[2733117.406597] ---[ end trace 52e7a6fe65a762ba ]---
[2733117.416180] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[2733117.443658] mtk_soc_eth 1e100000.ethernet eth0: Link is Down
[2733117.483540] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[2733117.499596] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx

使用X-WRT 22.03版本, hwnat开启与关闭都会出现这个交换队列超时的错误

尝试在gsw_mt7621.c里把flow control关闭还是不行
/* (GE1, Force 1000M/FD, FC OFF, MAX_RX_LENGTH 1536) */
mtk_switch_w32(gsw, 0x2305e30b, GSW_REG_MAC_P0_MCR);
mt7530_mdio_w32(gsw, 0x3600, 0x5e30b);

ptpt52, 2022-06-02:
设备型号是哪个？不同型号用的驱动不一样 @guoqiang13

guoqiang13, 2022-06-02:
mt7621_unielec_u7621-06-16m 这块主板，发现在openewrt forum里面也有讨论mt7621这个崩溃错误的主题，不过最终似乎都没能解决

guoqiang13, 2022-06-02:
https://forum.openwrt.org/t/mtk-soc-eth-watchdog-timeout-after-r11573/50000，从19年开始这个问题讨论

ptpt52, 2022-06-02:
```diff
diff --git a/target/linux/ramips/dts/mt7621.dtsi b/target/linux/ramips/dts/mt7621.dtsi
index 665e0332b8..305b2df0fc 100644
--- a/target/linux/ramips/dts/mt7621.dtsi
+++ b/target/linux/ramips/dts/mt7621.dtsi
@@ -514,7 +514,6 @@
                        fixed-link {
                                speed = <1000>;
                                full-duplex;
-                               pause;
                        };
                };
 
@@ -582,7 +581,6 @@
                                                fixed-link {
                                                        speed = <1000>;
                                                        full-duplex;
-                                                       pause;
                                                };
                                        };
                                };
```

你的环境容易出现？你自己编译的吧，改这个，把流控都关掉试试？

guoqiang13, 2022-06-04:
目前发现一个现象，开启HWNAT，使用iperf3推流测试，同时开启有线和WIFI会导致有线速率下跌到15mbit/s左右，关闭WIFI速度会恢复到920mbis/s。
[image]

ptpt52, 2022-06-04:
并不会呀，具体测试步骤怎么做

guoqiang13, 2022-06-04:
两台客户端电脑，一台是有线连接，一台WIFI连接。还有一个服务器电脑连接路由器WAN口，服务器电脑运行两个iperf3服务端模式

guoqiang13, 2022-06-04:
这是两个有线连接并发，速度正常
[image]

ptpt52, 2022-06-04:
并不会呀，我有线 无线 两个一起跑，并不会产生影响

guoqiang13, 2022-06-04:
X-WRT v22.03_b202205290126， natflow 20220608。还有在mt7621 dts文件里把流控关掉了

ptpt52, 2022-06-04:
> X-WRT v22.03_b202205290126， natflow 20220608。还有在mt7621 dts文件里把流控关掉了

理论上都没有任何影响的。

guoqiang13, 2022-06-04:
我是把4个lan口做了一个bridge vlan，然后把wifi接口桥接到这个vlan里面 ，会不会有这方面影响

guoqiang13, 2022-09-30:
不好意思，前段时间忙项目忘了。网口出现这个跟CPU中断分配有关。我用原版OPENWRT修改了网络接口接收中断分配，可以重现这个BUG