x-wrt/x-wrt #254: 可以支持clash链接订阅吗

AnswerX, 2022-06-11:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ptpt52, 2022-06-11:
本身也没有clash吧。

AnswerX, 2022-06-13:
我看到有个openWrt的插件[luci-app-openclash](https://github.com/vernesong/OpenClash)，还有一个什么ShadowSocksR Plus+插件，我在xrt插件列表里没有找到。因为我想使用链接订阅