x-wrt/x-wrt #257: AP mode

godshades, 2022-07-18:
Hi, can we have ap mode in setting
normally I turn off dhcp for each AP, but after reboot it will turn on by default
any ideal for this ?

ptpt52, 2022-07-18:
it should not turn on after reboot
maybe your device cannot save cofnig?

godshades, 2022-07-19:
@ptpt52 any other config can save, only dhcp in lan setting can't
my list ap is: redmi ac2100, mi mini, mi r4ac 100mb
all can't :(

ptpt52, 2022-07-19:
maybe you could use fakemesh to setup your network.

godshades, 2022-07-19:
now my setup is newifi 3d2 (DHCP) => lan - lan each AP (turn off dhcp)
if using fakemesh, i can setup lan-wan each AP and don't need turn off dhcp each ap right ?

ptpt52, 2022-07-19:
fakemesh would handle dhcp, it would help to turn it off, and make wan  work as lan

godshades, 2022-07-23:
Hi, after turn on fake mesh
My router not assign ip (lan & wireless) so now i can't do anything
The wifi list still display new ssid when I create fake mesh but can't connected

ptpt52, 2022-07-23:
your setup must have one AC and one or mor AP
AP.....AP-----AC---

and
AC: set fakemesh role AC,

AP: set fakemesh role wired AP or Agent (if you use wifi to connected to AC)

godshades, 2022-07-23:
Yes, I just set newifi d2 to role ac then this happen like above

godshades, 2022-07-25:
after go to failure mode, I can config and make it work
still don't know what is real problem

godshades, 2022-07-25:
work great
and i found DAWN work better than usteer

closed 2022-07-25 by godshades