x-wrt/x-wrt #259: Newifi D2 not showing WAN on Switch interface

godshades, 2022-07-25:
I have flash laster version twice: [x-wrt-22.03-b202207230649-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade.bin](https://downloads.x-wrt.com/rom/x-wrt-22.03-b202207230649-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade.bin)
and factory reset too

But in my switch interface not showing WAN, please help

Sample image:
[image]

ptpt52, 2022-07-25:
@godshades  sure. for now wan is eth1 on newifi-d2.

ptpt52, 2022-07-25:
wait, you mean not wan port could connect to internet?

godshades, 2022-07-25:
before this is like image below
[image]

now i'm connect to lan1 as wan to have internet access

i will try to change wan in network/interface to eth1 later

ptpt52, 2022-07-25:
I think in next release this would turn back to switch port on wan, since this https://github.com/openwrt/openwrt/pull/10238#issuecomment-1193168848

Romandas-A, 2022-07-29:
Doesn't work Newifi D2 WAN with x-wrt-22.03-b202207280130-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade.bin

ptpt52, 2022-07-29:
> Doesn't work Newifi D2 WAN with x-wrt-22.03-b202207280130-ramips-mt7621-d-team_newifi-d2-squashfs-sysupgrade.bin

do config reset and wan is eth1 now

but in next release wan would be in switch port again.