x-wrt/x-wrt #263: X-WRT R6220 无法无线中继

imzgg, 2022-08-04:
设备型号：Netgear R6220
架构：MediaTek MT7621 ver:1 eco:3

当在正常开启2.4G无线接入点AP情况下，添加WIFI客户端后，WIFI客户端无法连接
[image]
当我禁用了无线接入点AP后，WIFI客户端就可以连接了 PS：并没有更改任何设置的情况下
[image]

2.4G除了带宽和功率修改过，其他均为默认
[image]

ptpt52, 2022-08-04:
设置固定信道。扫描AP的时候注意信道多少，设置好信道

imzgg, 2022-08-05:
> 时间固定信道。扫描AP的注意信道，设置好信道

imzgg, 2022-08-05:
> 时间固定信道。扫描AP的注意信道，设置好信道

你好，请问是否可以根据所连接的客户端的信道来自动设置AP的信道？如果不行也没事，那我手动设置一下，谢谢~

closed 2022-08-05 by imzgg