x-wrt/x-wrt #265: P2W R619AC 128m 无法上网

Leslie-Wong, 2022-08-12:
刷了x-wrt 22.03版， P2W R619AC 128m 无法上网
好像WAN 和 LAN换了，兩个MAC地址换位了，在交换机那WAN不见了。
[image]
[image]
[image]

ptpt52, 2022-08-12:
没错，恢复出厂设置就行了，现在这个版本跟op上游一致了，送一季哦啊幻片功能也丧失了，wan口是eth1

Leslie-Wong, 2022-08-12:
恢复了,还是一样，外网IP还是读不到。

Leslie-Wong, 2022-08-12:
刷Lean版
[image]
[image]

ptpt52, 2022-08-12:
肯定可以的。 说了wan口要是 eth1
恢复默认配置，wan口肯定正常，然后设置向导设置上网。

ptpt52, 2022-08-12:
swtich 不需要任何改动，也不要划分什么vlan，没有这回事了。

ptpt52, 2022-08-12:
就是默认配置就行，你是不是恢复你自己备份的配置？那当然不行。

Leslie-Wong, 2022-08-12:
@ptpt52 谢谢您的回应。
但是我是在opboo用ubi刷，刷完就恢复默认配置，再设置上网。然后不行又在x-wrt升级版面用bin,是沒有保留配置的，刷完又操作恢复默认配置，再设置上网，还是不行。

刚刚也用了openwrt原版试了，也是用上要面方法，结果也是不行，就是外网连不上

ptpt52, 2022-08-12:
x-wrt 和 openwrt 原版是保持一致的，你拔插入网线
dmesg 信息，应该看到 eth1 down up 信息
你试试。

Leslie-Wong, 2022-08-12:
沒有dmesg 
[image]

在logs有，还是没有外网IP。不会是网供应商问题吧。
`Sat Aug  6 13:26:30 2022 kern.info kernel: [  330.411463] ess_edma c080000.edma eth1: Link is Down
Sat Aug  6 13:26:30 2022 daemon.notice netifd: Network device 'eth1' link is down
Sat Aug  6 13:26:30 2022 daemon.notice netifd: Interface 'wan' has link connectivity loss
Sat Aug  6 13:26:30 2022 daemon.notice netifd: Interface 'wan6' has link connectivity loss
Sat Aug  6 13:26:31 2022 daemon.notice netifd: wan (11696): udhcpc: received SIGTERM
Sat Aug  6 13:26:31 2022 daemon.notice netifd: Interface 'wan6' is now down
Sat Aug  6 13:26:31 2022 daemon.notice netifd: wan (11696): udhcpc: entering released state
Sat Aug  6 13:26:31 2022 daemon.notice netifd: wan (11696): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
Sat Aug  6 13:26:31 2022 daemon.notice netifd: Interface 'wan' is now down
Sat Aug  6 13:26:31 2022 user.notice mwan3-hotplug[11717]: mwan3 hotplug on wan6 not called because interface disabled
Sat Aug  6 13:26:32 2022 user.notice upnp daemon: external interface not found, not starting
Sat Aug  6 13:26:32 2022 daemon.err miniupnpd[11941]: STUN: Cannot get IP address for ext interface
Sat Aug  6 13:26:32 2022 daemon.err miniupnpd[11941]: Performing STUN failed. EXITING
Sat Aug  6 13:26:32 2022 user.notice mwan3-hotplug[12031]: mwan3 hotplug on wan not called because interface disabled
Sat Aug  6 13:26:33 2022 user.notice upnp daemon: external interface not found, not starting
Sat Aug  6 13:26:33 2022 daemon.err miniupnpd[12301]: STUN: Cannot get IP address for ext interface
Sat Aug  6 13:26:33 2022 daemon.err miniupnpd[12301]: Performing STUN failed. EXITING
Sat Aug  6 13:26:43 2022 daemon.notice netifd: Network device 'eth1' link is up
Sat Aug  6 13:26:43 2022 daemon.notice netifd: Interface 'wan' has link connectivity
Sat Aug  6 13:26:43 2022 daemon.notice netifd: Interface 'wan' is setting up now
Sat Aug  6 13:26:43 2022 daemon.notice netifd: Interface 'wan6' has link connectivity
Sat Aug  6 13:26:43 2022 daemon.notice netifd: Interface 'wan6' is setting up now
Sat Aug  6 13:26:43 2022 kern.info kernel: [  342.891295] ess_edma c080000.edma eth1: Link is Up - 1Gbps/Full - flow control rx/tx
Sat Aug  6 13:26:43 2022 daemon.notice netifd: wan (12435): udhcpc: started, v1.35.0
Sat Aug  6 13:26:43 2022 daemon.notice netifd: wan (12435): udhcpc: broadcasting discover
Sat Aug  6 13:26:46 2022 daemon.notice netifd: wan (12435): udhcpc: broadcasting discover
Sat Aug  6 13:26:49 2022 daemon.notice netifd: wan (12435): udhcpc: broadcasting discover
Sat Aug  6 13:27:40 2022 daemon.info dnsmasq[8416]: read /etc/hosts - 4 addresses
Sat Aug  6 13:27:40 2022 daemon.info dnsmasq[8416]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sat Aug  6 13:27:40 2022 daemon.info dnsmasq-dhcp[8416]: read /etc/ethers - 0 addresses
Sat Aug  6 13:27:40 2022 daemon.notice ksmbd: Stopping Ksmbd userspace service.
Sat Aug  6 13:27:41 2022 daemon.notice ksmbd: Starting Ksmbd userspace service.`

ptpt52, 2022-08-12:
有 eth1 up down 信息了，不知道你的啥问题了。按道理没问题的，问问其他人的设备。
或者试试断电再开机试试

Leslie-Wong, 2022-08-12:
@ptpt52 谢谢您
我在研究研究吧

Leslie-Wong, 2022-08-12:
天呀！我插到另一台路由器上，雖然有IP了，这是什么会事，真的是网供应商问题？
[image]

closed 2022-08-12 by Leslie-Wong