x-wrt/x-wrt #271: p2w-r619ac-128m - Gdock 2.0 与 mpcie 有问题

Roxy09099, 2022-08-22:
i plugged in wwan lte Quectel EC20 card but x-wrt can't recognize the card
kernel log:
Sun Aug 21 09:31:58 2022 kern.info kernel: [    0.053011] PCI: CLS 0 bytes, default 64
Sun Aug 21 09:31:58 2022 kern.info kernel: [    0.055207] workingset: timestamp_bits=30 max_order=17 bucket_order=0
Sun Aug 21 09:31:58 2022 kern.info kernel: [    0.059127] squashfs: version 4.0 (2009/01/31) Phillip Lougher
Sun Aug 21 09:31:58 2022 kern.info kernel: [    0.059151] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
Sun Aug 21 09:31:58 2022 kern.info kernel: [    0.188960] qcom-pcie 40000000.pci: host bridge /soc/pci@40000000 ranges:
Sun Aug 21 09:31:58 2022 kern.info kernel: [    0.189020] qcom-pcie 40000000.pci:       IO 0x0040200000..0x00402fffff -> 0x0040200000
Sun Aug 21 09:31:58 2022 kern.info kernel: [    0.189051] qcom-pcie 40000000.pci:      MEM 0x0040300000..0x0040ffffff -> 0x0040300000
Sun Aug 21 09:31:58 2022 kern.info kernel: [    1.371397] qcom-pcie 40000000.pci: Phy link never came up
Sun Aug 21 09:31:58 2022 kern.err kernel: [    1.372981] qcom-pcie 40000000.pci: cannot initialize host
Sun Aug 21 09:31:58 2022 kern.warn kernel: [    1.373104] qcom-pcie: probe of 40000000.pci failed with error -110
Sun Aug 21 09:31:58 2022 kern.err kernel: [    1.374812] bam-dma-engine 8e04000.dma: num-channels unspecified in dt
Sun Aug 21 09:31:58 2022 kern.err kernel: [    1.374835] bam-dma-engine 8e04000.dma: num-ees unspecified in dt
Sun Aug 21 09:31:58 2022 kern.info kernel: [    1.376250] tcsr 1949000.tcsr: setting wifi_glb_cfg = 41000000
Sun Aug 21 09:31:58 2022 kern.info kernel: [    1.376353] tcsr 194b000.tcsr: setting usb hs phy mode select = e700e7
Sun Aug 21 09:31:58 2022 kern.info kernel: [    1.376447] tcsr 1953000.ess_tcsr: setting ess interface select = 0
Sun Aug 21 09:31:58 2022 kern.info kernel: [    1.376532] tcsr 1957000.tcsr: setting wifi_noc_memtype_m0_m2 = 2222222
Sun Aug 21 09:31:58 2022 kern.info kernel: [    1.377386] Serial: 8250/16550 driver, 2 ports, IRQ sharing disabled
Sun Aug 21 09:31:58 2022 kern.info kernel: [    1.378269] msm_serial 78af000.serial: msm_serial: detected port #0
Sun Aug 21 09:31:58 2022 kern.info kernel: [    1.378319] msm_serial 78af000.serial: uartclk = 1843200
Sun Aug 21 09:31:58 2022 kern.info kernel: [    1.378378] 78af000.serial: ttyMSM0 at MMIO 0x78af000 (irq = 35, base_baud = 115200) is a MSM
Sun Aug 21 09:31:58 2022 kern.info kernel: [    1.378409] msm_serial: console setup on port #0
Sun Aug 21 09:31:58 2022 kern.info kernel: [    1.974959] printk: console [ttyMSM0] enabled
Sun Aug 21 09:31:58 2022 kern.info kernel: [    1.980284] msm_serial: driver initialized

ptpt52, 2022-08-22:
```
echo "1" >/sys/class/gpio/gpio414/value
sleep 1
echo "0" >/sys/class/gpio/gpio414/value
```
执行这个可以吗

Roxy09099, 2022-08-22:
> 
just run this command?  or have to give this command when starting the device?

Roxy09099, 2022-08-22:
> ```
> echo "1" >/sys/class/gpio/gpio414/value
> sleep 1
> echo "0" >/sys/class/gpio/gpio414/value
> ```
> 
> 执行这个可以吗

mpcie still doesn't work after running the above 2 commands

Roxy09099, 2022-08-22:
give up mpcie

closed 2022-08-22 by Roxy09099