x-wrt/x-wrt #276: please please please fix wifi problem mi router r3

saeidselfjoor, 2022-09-18:
I install latest version of x-wrt

2.4 wifi is too slow and  can't connect at all sometimes

is there any way to revert back to stock firmware? any guide?

or is there any way to use pandorabox driver i heard it's work better

please help me

saeidselfjoor, 2022-09-21:
anyone have version of x-wrt with mtd write unlock ?

Bob-YsPan, 2023-08-27:
I'm not encounter this bug maybe I use 5GHz band mostly.

But I found the problem that if perform a scan at wireless setting page, the 2.4GHz interface will not sending any response and stuck at the scanning page like below screenshot, and I reload the page the webserver will not responding about 1 minutes, and then back to the normal(the 2.4GHz scan still not work)
[image]

The 5GHz band can finish the scan after about 10s and show the available networks:
[image]

Bob-YsPan, 2023-08-27:
And I not see any useful log when I view the system logs page...
(I perform a 2.4GHz scan at 11:11, and only see a log about 2.4GHz device reconnect it)
(The corntab's log is ok, it's a watchdog to see the connection is ok and change the light at front panel)

```
Sun Aug 27 11:10:00 2023 cron.err crond[2646]: USER root pid 28729 cmd if ! ps -e | grep "conn_check" | grep -v grep; then /usr/bin/conn_check > /dev/null 2>&1 < /dev/null & fi
Sun Aug 27 11:11:00 2023 cron.err crond[2646]: USER root pid 29264 cmd if ! ps -e | grep "conn_check" | grep -v grep; then /usr/bin/conn_check > /dev/null 2>&1 < /dev/null & fi
Sun Aug 27 11:11:53 2023 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 24:d7:eb:11:a2:88
Sun Aug 27 11:11:53 2023 daemon.info hostapd: phy1-ap0: STA 24:d7:eb:11:a2:88 IEEE 802.11: disassociated
Sun Aug 27 11:11:53 2023 user.info usteer: station 24:d7:eb:11:a2:88 disconnected from node hostapd.phy1-ap0
Sun Aug 27 11:11:55 2023 daemon.info hostapd: phy1-ap0: STA 24:d7:eb:11:a2:88 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Aug 27 11:12:00 2023 cron.err crond[2646]: USER root pid 29800 cmd if ! ps -e | grep "conn_check" | grep -v grep; then /usr/bin/conn_check > /dev/null 2>&1 < /dev/null & fi
Sun Aug 27 11:13:00 2023 cron.err crond[2646]: USER root pid 30301 cmd if ! ps -e | grep "conn_check" | grep -v grep; then /usr/bin/conn_check > /dev/null 2>&1 < /dev/null & fi
Sun Aug 27 11:14:00 2023 cron.err crond[2646]: USER root pid 30817 cmd if ! ps -e | grep "conn_check" | grep -v grep; then /usr/bin/conn_check > /dev/null 2>&1 < /dev/null & fi
Sun Aug 27 11:14:08 2023 daemon.info hostapd: phy1-ap0: STA 24:d7:eb:11:a2:88 IEEE 802.11: authenticated
Sun Aug 27 11:14:09 2023 daemon.info hostapd: phy1-ap0: STA 24:d7:eb:11:a2:88 IEEE 802.11: associated (aid 1)
Sun Aug 27 11:14:09 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 24:d7:eb:11:a2:88 auth_alg=open
Sun Aug 27 11:14:09 2023 daemon.info hostapd: phy1-ap0: STA 24:d7:eb:11:a2:88 RADIUS: starting accounting session 55F83CA7B6C46506
Sun Aug 27 11:14:09 2023 daemon.info hostapd: phy1-ap0: STA 24:d7:eb:11:a2:88 WPA: pairwise key handshake completed (RSN)
Sun Aug 27 11:14:09 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 24:d7:eb:11:a2:88
Sun Aug 27 11:14:09 2023 daemon.info dnsmasq-dhcp[7575]: DHCPDISCOVER(br-lan) 24:d7:eb:11:a2:88
Sun Aug 27 11:14:09 2023 daemon.info dnsmasq-dhcp[7575]: DHCPOFFER(br-lan) 192.168.5.183 24:d7:eb:11:a2:88
Sun Aug 27 11:14:09 2023 daemon.info dnsmasq-dhcp[7575]: DHCPREQUEST(br-lan) 192.168.5.183 24:d7:eb:11:a2:88
Sun Aug 27 11:14:09 2023 daemon.info dnsmasq-dhcp[7575]: DHCPACK(br-lan) 192.168.5.183 24:d7:eb:11:a2:88 esp32-11A288
```

Bob-YsPan, 2023-08-27:
### Update:
I tried some command on the ssh, when web ui stuck on the scanning and just close the scan window, then input the command line's scan command, it will showing:

```
root@Bob-Mi-WiFi-3:~# time iw phy1-ap0 scan
command failed: Resource busy (-16)
```

After few seconds re-execute it, it can be finished the command, but need to take very long time( > 1 minutes ):

```
(lots of results...)
                 * Pairwise ciphers: CCMP
                 * Authentication suites: PSK
                 * Capabilities: 16-PTKSA-RC 1-GTKSA-RC (0x000c)
BSS ec:ad:e0:cd:d2:f7(on phy1-ap0)
        TSF: 707305165120 usec (8d, 04:28:25)
        freq: 2462
        beacon interval: 100 TUs
        capability: ESS (0x0411)
        signal: -83.00 dBm
        last seen: 11180 ms ago
        SSID: dlink-2.4GHz
        HT capabilities:
                Capabilities: 0x11ee
                        HT20/HT40
                        SM Power Save disabled
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT RX MCS rate indexes supported: 0-15, 32
                HT TX MCS rate indexes are undefined
        HT operation:
                 * primary channel: 11
                 * secondary channel offset: no secondary
                 * STA channel width: 20 MHz
        WPA:     * Version: 1
                 * Group cipher: TKIP
                 * Pairwise ciphers: TKIP CCMP
                 * Authentication suites: PSK
        RSN:     * Version: 1
                 * Group cipher: TKIP
                 * Pairwise ciphers: TKIP CCMP
                 * Authentication suites: PSK
                 * Capabilities: 1-PTKSA-RC 1-GTKSA-RC (0x0000)
real    1m 7.30s
user    0m 0.02s
sys     0m 0.04s
```

The 5GHz band can finished scan in the short time(< 10 seconds )

```
                HE PHY Capabilities: (0x442002c00f419508008c00):
                        HE40/HE80/5GHz
                        242 tone RUs/5GHz
                        LDPC Coding in Payload
                        NDP with 4x HE-LTF and 3.2us GI
                        Rx HE MU PPDU from Non-AP STA
                        SU Beamformer
                        SU Beamformee
                        MU Beamformer
                        Beamformee STS <= 80Mhz: 3
                        Sounding Dimensions <= 80Mhz: 1
                        Ng = 16 SU Feedback
                        Codebook Size SU Feedback
                        Triggered SU Beamforming Feedback
                        Triggered CQI Feedback
                        PPE Threshold Present
                        Max NC: 1
                        TX 1024-QAM
                        RX 1024-QAM
                HE RX MCS and NSS set <= 80 MHz
                        1 streams: MCS 0-11
                        2 streams: MCS 0-11
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                HE TX MCS and NSS set <= 80 MHz
                        1 streams: MCS 0-11
                        2 streams: MCS 0-11
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                PPE Threshold 0x79 0x1c 0xc7 0x71 0x1c 0xc7 0x71
real    0m 3.30s
user    0m 0.02s
sys     0m 0.05s
```