x-wrt/x-wrt #281: kmod-mtd-rw kernel depend mi router r3

saeidselfjoor, 2022-09-21:
hi
i download [X-WRT v22.03_b202209152117](https://github.com/x-wrt/build-release/releases/tag/v22.03_b202209152117)
 and try install kmod-mrt-rw by putting it in /tmp
but get error always

is there any way to install kmod-mtd-rw ?
[image]

ptpt52, 2022-09-21:
you download the wrong one?

saeidselfjoor, 2022-09-21:
hi , no download right asset but get error

try two assets non of them work

saeidselfjoor, 2022-09-21:
> you download the wrong one?

no download it right but doesn't work
[image]

saeidselfjoor, 2022-09-21:
I try every way I know to install kmod but nothing work 
[image]

saeidselfjoor, 2022-09-21:
> you download the wrong one?

[image]

saeidselfjoor, 2022-09-21:
[image]

saeidselfjoor, 2022-09-21:
https://github.com/x-wrt/build-release/releases/download/v22.03_b202209152117/x-wrt-22.03_b202209152117-ramips-mt7620.zip
[image]

saeidselfjoor, 2022-09-21:
> you download the wrong one?

can you please give me a download link of correct one

saeidselfjoor, 2022-09-21:
any x-wrt-22.03-b202209152117-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade version with unlock bootloader?

ptpt52, 2022-09-21:
just untar the ipk and get the kmod.ko file

ptpt52, 2022-09-21:
or install the version in zip file and then try install the ipk again.