x-wrt/x-wrt #291: 10月12日版本的 CSAC固件无法保存设置

mrlong32, 2022-10-16:
ping ttl返回是128
任何配置修改均无效

ptpt52, 2022-10-17:
@mrlong32  登录ssh，执行 df -h 看看空间剩余情况。

mrlong32, 2022-10-17:
无法登录，SSH默认关闭
因无法保存配置
打开操作无效

ptpt52, 2022-10-17:
刷新版本看看。

mrlong32, 2022-10-17:
> 刷新版本看看。

刷了，还是不行，是不是分区不一样了
从挂载点看 /overlay 才300K

ptpt52, 2022-10-17:
df -h 看看

mrlong32, 2022-10-17:
> df -h 看看

打不开SSH，没法进命令行

ptpt52, 2022-10-17:
@mrlong32 
用这个版本试试，固件删了部分软件留出空间
http://downloads.x-wrt.com/rom/Downloads/csac2/

mrlong32, 2022-10-17:
> @mrlong32 用这个版本试试，固件删了部分软件留出空间 http://downloads.x-wrt.com/rom/Downloads/csac2/

OK了，这个恢复正常了，谢谢

ptpt52, 2022-10-17:
df -h 看看。

mrlong32, 2022-10-17:
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 9.0M      9.0M         0 100% /rom
devtmpfs                 58.9M         0     58.9M   0% /rom/dev
tmpfs                    59.5M    444.0K     59.0M   1% /tmp
/dev/mtdblock11           4.4M    368.0K      4.1M   8% /overlay
overlayfs:/overlay        4.4M    368.0K      4.1M   8% /
tmpfs                     1.0M         0      1.0M   0% /mnt
tmpfs                   512.0K         0    512.0K   0% /dev

closed 2022-10-18 by ptpt52