x-wrt/x-wrt #299: 无法安装openconnect  内核版本不兼容

sukianata, 2022-11-25:
尝试安装时遇到以下错误：
```
root@X-WRT:/mnt/sda1# opkg install openconnect
Installing openconnect (9.01-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/openconnect_9.01-1_mipsel_24kc.ipk
Configuring openconnect.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.155-1-8057c73633ced5b3fb73547a26b81aa2) for kmod-tun
```

可以帮忙看一下吗

为了安装openconnect, 我尝试自己编译一个，但是make 时遇到下面的错误

[image]
编译环境：Macos M1 Apple silicon

有哪位同学有在M1 mac下成功编译过吗

ptpt52, 2022-11-25:
强制安装就行 
```
opkg install --force-depends openconnect
```
不需要另外安装所谓kmod-tun
系统自带有。

sukianata, 2022-11-26:
@ptpt52 感谢回复

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2023-12-22 by woaihuangfan