x-wrt/x-wrt #301: add Gl-S1300

manias21, 2022-11-28:
请求支持 Gl-S1300， 谢谢！
Please can you add the Gl-S1300, thanks!!

ptpt52, 2022-11-28:
应该让openwrt官方添加支持。

manias21, 2022-11-28:
I have tried to create a firmware with your manual, but I can't get the fakemesh on the s1300, I have a B1300 and it works great with its x-wrt firmware, for that reason, would you mind making one of the s1300 please?

ptpt52, 2022-11-28:
since s1300 is not converted to dsa, it is now not supported by openwrt. you must request openwrt to add support for it. then we could build x-wrt firmware for it.

manias21, 2022-11-28:
Excuse me if I don't understand you because of the translation that I use in google, but I have located the firmware for openwrt, and with it... could I then?
firmware..
https://openwrt.org/toh/gl.inet/gl.inet_gl-s1300

ptpt52, 2022-11-29:
that is true. but snapshot head is not supported now. must request openwrt add support for it

closed 2023-01-14 by manias21