x-wrt/x-wrt #302: Dear All .. X-WRT Repeator mode

testmodem, 2022-11-28:
Dear all .

I have tried to get X-WRT working as a repeater mode, but failed. I have tried to google it but failed as well. 

Can anyone please help in this respect ? A video tutorial will be very helpful . I am trying to add another router ,as a repeater with my existing home router. 

Main Router = Huawei running propriety software
Repeater Router = X-WRT
Model | Xiaomi Mi Router R3

Architecture | MediaTek MT7620A ver:2 eco:6
Target Platform | ramips/mt7620
Firmware Version | X-WRT 22.10_b202211250846 Kinetic / LuCI (no branch) branch git-22.328.34826-5527785

Thanks

ptpt52, 2022-11-28:
repeater may not possible since huawei may not enable AP wds
you may use x-wrt as sta wifi client.

wireless -> scan -> connect the wifi

testmodem, 2022-11-29:
Thank you for your response. 

I am not connecting wirelessly. I am a cable the connect the both as I have a wall in between the rooms.

Is it possible to have X-Wrt function as a repeater ?

testmodem, 2022-11-29:
> Thank you for your response.
> 
> I am not connecting wirelessly. I am a cable the connect the both as I have a wall in between the rooms.
> 
> Is it possible to have X-Wrt function as a repeater ?

Thank you for your response.

I am not connecting wirelessly. I am a cable the connect the both as I have a wall in between the rooms.

Is it possible to have X-Wrt function as a repeater ?

ptpt52, 2022-11-29:
just disable lan DHCP and then connect lan port to your network.

testmodem, 2022-11-30:
Thanks .. is there any manual on X-Wrt ? or Video tutorials ?