x-wrt/x-wrt #303: x-wrt snapshot: luci cannot login on AX6S

trippleflux, 2022-11-28:
I cannot login into luci on my self-compiled x-wrt snapshot, tried admin:admin root:admin , somehow through ssh I can login with only user root without any password.

trippleflux, 2022-11-28:
From chrome's developer console : 
```
192.168.15.1/:1          GET http://192.168.15.1/cgi-bin/luci/ 403 (Forbidden)
192.168.15.1/:11          GET http://192.168.15.1/cgi-bin/luci/admin/translations/en net::ERR_ABORTED 500 (Internal Server Error)
192.168.15.1/:1 Refused to execute script from 'http://192.168.15.1/cgi-bin/luci/admin/translations/en' because its MIME type ('text/html') is not executable, and strict MIME type checking is enabled.
/luci-static/resources/icons/loading.gif?0.47427385109741094:1          GET https://192.168.15.1/luci-static/resources/icons/loading.gif?0.47427385109741094 net::ERR_CERT_AUTHORITY_INVALID
Image (async)
(anonymous) @ (index):101
(anonymous) @ (index):96
luci.js?v=git-22.330.30560-267ef5b:66          GET http://192.168.15.1/cgi-bin/luci/admin/menu?1669662013693 500 (Internal Server Error)
(anonymous) @ luci.js?v=git-22.330.30560-267ef5b:66
(anonymous) @ luci.js?v=git-22.330.30560-267ef5b:47
Promise.then (async)
request @ luci.js?v=git-22.330.30560-267ef5b:47
get @ luci.js?v=git-22.330.30560-267ef5b:71
load @ ui.js?v=git-22.330.30560-267ef5b:217
__init__ @ menu-openwrt2020.js?v=git-22.330.30560-267ef5b:1
ClassConstructor @ luci.js?v=git-22.330.30560-267ef5b:11
(anonymous) @ luci.js?v=git-22.330.30560-267ef5b:182
Promise.then (async)
compileClass @ luci.js?v=git-22.330.30560-267ef5b:178
Promise.then (async)
require @ luci.js?v=git-22.330.30560-267ef5b:184
(anonymous) @ (index):117
VM230:1 Uncaught (in promise) SyntaxError: Unexpected token '<', "<!--]]>-->"... is not valid JSON
    at JSON.parse (<anonymous>)
    at ClassConstructor.json (luci.js?v=git-22.330.30560-267ef5b:32:24)
    at ClassConstructor.eval (ui.js?v=git-22.330.30560-267ef5b:217:162)
```

ptpt52, 2022-11-28:
you may not use master HEAD code, use tag
see build docs
https://blog.x-wrt.com/docs/build/

closed 2022-11-28 by trippleflux