x-wrt/x-wrt #310: WiFi Client on Zbtlink ZBT-WE826 (16M) very unstable with FW X-WRT 22.10_b202212230054

ruiseixasm, 2022-12-26:
### Describe the bug

Hi,

This device, **WE826-2T (16M)** was always very bad with the WiFi (Wireless), and I already tried many other OpenWRT Firmware versions. This last one works better for a while as WiFi client mode and then after a couple of hours never gets more than 1 Mbps instead of the usual 20Mbps. I know the problem is this router and not the AP because I also test the WiFi quality of the AP with my smartphone directly and the speed of 20Mbps is always there, no drop, it's only the WE826-2T that fails to keep up. It seems though that it's better doing Upload than Download, so my suspicion is that it isn't as sensible to listing as a smartphone.

I also noticed that in the dashboard the square relative to the WiFi sometimes is duplicated in many other squares with the same information of the first square!

Another thing now happening, with this firmware and not with the other ones, is the wireless blue light on the device itself that doesn't light up.

I noticed that many services have been added up to this router, but more important than that is to have a well working WiFi, and that isn't happening. Is it possible to have Wireless Client working all right? Also as an Access Point it isn't stable neither, there are still many problems with this Firmware concerning the WiFi (Wireless) capabilities!

I noticed also that the Telnet service isn't working out of the box!

Can you please solve these issues?

Thank you very much.

PS: System Information:
Hostname	X-WRT
Model	Zbtlink ZBT-WE826 (16M)
Architecture	MediaTek MT7620A ver:2 eco:6
Target Platform	ramips/mt7620
Firmware Version	X-WRT 22.10_b202212230054 Kinetic / LuCI (no branch) branch git-22.356.59859-a3abce8
Kernel Version	5.15.84
Local Time	2022-12-26 12:30:27
Uptime	0h 25m 30s
Load Average	0.21, 0.19, 0.35

### OpenWrt version

X-WRT 22.10_b202212230054 Kinetic / LuCI (no branch) branch git-22.356.59859-a3abce8

### OpenWrt target/subtarget

Wireless / WiFi Speed

### Device

WE826-2T (16M)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2022-12-26:
does it work fine on AP mode only?
if you use it as STA client, please scan wifi, connect, set fixed channel.

ruiseixasm, 2022-12-26:
No, it doesn't work fine in AP mode after a while when doesn't give more than 1Mbps, with or without configured Client WiFi connection. I'm testing the WiFi Client mode via ethernet cable to isolate the behavior of the Client mode without any AP possible limitations. I even did the testing with the AP enabled and disabled.

Before I set STA client it sets the channel automatically (channel 12) and it stays set as auto in the configurations, I only set manually the N (IEEE 802.11n) and 20MHz. Now after setting the channel 12 manually it's achieving more than 20Mbps Upload but is struggling 5 Mbps Download, I suspect that in a couple of hours it will drop even further because that's what happened before. Can it be hardware overheating? I plugged in the router just now. I tested with my smartphone on same location to the same Main AP and I got more than 25Mbps of download! So, once again, it isn't the Main AP that is causing this, it's the ZBT-WE826 instead.

I believe in two scenarios, my ZBT-WE826 router is faulty, or, there is some problem with the hardware design itself. I used a TP-Link TL-WR741ND v4 for comparison, a very cheap router, and as AP it works pretty well, however it struggles a lot to handle as WiFi Client despite the DD-WRT firmware installed in it that lets it to be configured as such. So, overall, the TL-WR741ND works better than the ZBT-WE826, what is surprising given the lower price and memory of the former.

Bellow is a capture of the repetition of the dashboard square concerning the WiFi data. This happens whenever I change the IPv4 gateway IP of my computer ethernet configurations.
[image]