x-wrt/x-wrt #322: 无法关机

phoenixzjh, 2023-02-23:
### Describe the bug

系统在虚拟机下无法自动关机，也无法通过WEB界面进行关机操作，能否加入自适应关机和手机关机操作

### OpenWrt version

x-wrt-22.10-b202302190944-x86-64

### OpenWrt target/subtarget

none

### Device

none

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jieny, 2023-05-11:
遇到同样的问题，pve 下也无法关机，各位大佬有什么解决方案？

jiqing112, 2023-05-12:
> 遇到同样的问题，pve 下也无法关机，各位大佬有什么解决方案？

请教个问题，我在vmware虚拟机里以桥接模式启动了x-wrt ，分配的IP可以从外部ping通，x-wrt内的80端口也有nginx监听，但是通过ip:80无法访问到管理页面。tcping ip的80端口也是不通的，是防火墙的问题吗？

jieny, 2023-05-12:
> > 遇到同样的问题，pve 下也无法关机，各位大佬有什么解决方案？
> 
> 请教个问题，我在vmware虚拟机里以桥接模式启动了x-wrt ，分配的IP可以从外部ping通，x-wrt内的80端口也有nginx监听，但是通过ip:80无法访问到管理页面。tcping ip的80端口也是不通的，是防火墙的问题吗？

百分之八九十是防火墙的问题。主机是Windows? 如果是把windows的防火墙（域网络、专用网络、公用网络）关掉试试看就知道了

jiqing112, 2023-05-12:
> > > 遇到同样的问题，pve 下也无法关机，各位大佬有什么解决方案？
> > 
> > 
> > 请教个问题，我在vmware虚拟机里以桥接模式启动了x-wrt ，分配的IP可以从外部ping通，x-wrt内的80端口也有nginx监听，但是通过ip:80无法访问到管理页面。tcping ip的80端口也是不通的，是防火墙的问题吗？
> 
> 百分之八九十是防火墙的问题。主机是Windows? 如果是把windows的防火墙（域网络、专用网络、公用网络）关掉试试看就知道了

我是从windows 的chrome 访问vmware 内x-wrt的IP:80，虚拟机网络是桥接的，和windows 的ip是同一个网段。
问题好像出在x-wrt上，我在 x-wrt上手动执行iptables放行80端口还是无用。

jiqing112, 2023-05-12:
> > > 遇到同样的问题，pve 下也无法关机，各位大佬有什么解决方案？
> > 
> > 
> > 请教个问题，我在vmware虚拟机里以桥接模式启动了x-wrt ，分配的IP可以从外部ping通，x-wrt内的80端口也有nginx监听，但是通过ip:80无法访问到管理页面。tcping ip的80端口也是不通的，是防火墙的问题吗？
> 
> 百分之八九十是防火墙的问题。主机是Windows? 如果是把windows的防火墙（域网络、专用网络、公用网络）关掉试试看就知道了

就是从windows上ping x-wrt的ip ，可以ping通，

tcping x-wrt的80端口无响应。

jiqing112, 2023-05-12:
> > > 遇到同样的问题，pve 下也无法关机，各位大佬有什么解决方案？
> > 
> > 
> > 请教个问题，我在vmware虚拟机里以桥接模式启动了x-wrt ，分配的IP可以从外部ping通，x-wrt内的80端口也有nginx监听，但是通过ip:80无法访问到管理页面。tcping ip的80端口也是不通的，是防火墙的问题吗？
> 
> 百分之八九十是防火墙的问题。主机是Windows? 如果是把windows的防火墙（域网络、专用网络、公用网络）关掉试试看就知道了

[image]

jieny, 2023-05-12:
@jiqing112 这么神奇，能ssh进x-wrt？里面看看80是否正常，有可能里面冲突了吧，进去看看curl http://127.0.0.1:80 有没有正常返回 html。这个只能自己慢慢排查了。自己编译的固件？

jiqing112, 2023-05-12:
> @jiqing112 这么神奇，能ssh进x-wrt？里面看看80是否正常，有可能里面冲突了吧，进去看看curl http://127.0.0.1:80 有没有正常返回 html。这个只能自己慢慢排查了。自己编译的固件？

curl 有返回， ssh也在运行，但也无法连接。
[image]

jieny, 2023-05-12:
@jiqing112  那x-wrt里算正常，估计外面到里面的问题，估计还是防火墙或者路由、网络什么的有影响，关掉Windows防火墙，关掉VPN，关掉代理等等试试

jiqing112, 2023-05-12:
> @jiqing112 那x-wrt里算正常，估计外面到里面的问题，估计还是防火墙或者路由、网络什么的有影响，关掉Windows防火墙，关掉VPN，关掉代理等等试试

只能说是防火墙的问题，我在xwrt内部把防火墙干掉就可以了。
/etc/init.d/firewall stop

jiqing112, 2023-05-12:
> @jiqing112 那x-wrt里算正常，估计外面到里面的问题，估计还是防火墙或者路由、网络什么的有影响，关掉Windows防火墙，关掉VPN，关掉代理等等试试

解决问题了，防火墙允许wan接口的入站数据就行了

vi /etc/config/firewall 
找到wan区域
[image]