x-wrt/x-wrt #325: H3C Tx1806 upnp无法自动启动

wingssn, 2023-03-01:
### Describe the bug

H3C Tx1806 upnp无法自动启动，每次重启路由器之后都要在页面手动开启一次才能用upnp

### OpenWrt version

x-wrt-22.10-b202302270934

### OpenWrt target/subtarget

x-wrt-22.10-b202302270934

### Device

h3c_tx1806

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-03-02:
开机后看看日志 logread -e upnp

wingssn, 2023-03-09:
> 开机后看看日志 logread -e upnp

 -----------------------------------------------------
root@X-WRT:~# logread -e upnp
Fri Mar 10 02:01:36 2023 daemon.info uwsgi-luci: [info] luci: accepted login on /admin/services/upnp for admin from 192.168.15.173
root@X-WRT:~#

ptpt52, 2023-03-10:
没看见日志，重启一次，立刻查看日志呀

wingssn, 2023-03-13:
root@X-WRT:~# logread -e upnp
Mon Mar 13 09:47:23 2023 daemon.notice procd: /etc/rc.d/S19firewall: sh: can't open '/usr/share/miniupnpd/nft.include': No such file or directory
Mon Mar 13 09:47:36 2023 user.notice ucitrack: Setting up /etc/config/miniupnpd reload dependency on /etc/config/firewall
Mon Mar 13 09:47:42 2023 user.notice upnp daemon: external interface not found, not starting
Mon Mar 13 09:47:42 2023 daemon.err miniupnpd[5988]: STUN: Cannot get IP address for ext interface
Mon Mar 13 09:47:42 2023 daemon.err miniupnpd[5988]: Performing STUN failed. EXITING
Mon Mar 13 09:47:43 2023 user.notice upnp daemon: external interface not found, not starting
Mon Mar 13 09:47:43 2023 daemon.err miniupnpd[6189]: STUN: Cannot get IP address for ext interface
Mon Mar 13 09:47:43 2023 daemon.err miniupnpd[6189]: Performing STUN failed. EXITING
Mon Mar 13 09:47:54 2023 daemon.err miniupnpd[8021]: resolve_stun_host: getaddrinfo(stun.miwifi.com, 3478, ...) failed : Try again
Mon Mar 13 09:47:54 2023 daemon.err miniupnpd[8021]: STUN: Performing STUN failed: Host is unreachable
Mon Mar 13 09:47:54 2023 daemon.err miniupnpd[8021]: Performing STUN failed. EXITING
root@X-WRT:~#
抱歉，这下应该是了

wingssn, 2023-03-13:
logread -e upnp
Mon Mar 13 09:47:23 2023 daemon.notice procd: /etc/rc.d/S19firewall: sh: can't open '/usr/share/miniupnpd/nft.include': No such file or directory
Mon Mar 13 09:47:36 2023 user.notice ucitrack: Setting up /etc/config/miniupnpd reload dependency on /etc/config/firewall
Mon Mar 13 09:47:42 2023 user.notice upnp daemon: external interface not found, not starting
Mon Mar 13 09:47:42 2023 daemon.err miniupnpd[5988]: STUN: Cannot get IP address for ext interface
Mon Mar 13 09:47:42 2023 daemon.err miniupnpd[5988]: Performing STUN failed. EXITING
Mon Mar 13 09:47:43 2023 user.notice upnp daemon: external interface not found, not starting
Mon Mar 13 09:47:43 2023 daemon.err miniupnpd[6189]: STUN: Cannot get IP address for ext interface
Mon Mar 13 09:47:43 2023 daemon.err miniupnpd[6189]: Performing STUN failed. EXITING
Mon Mar 13 09:47:54 2023 daemon.err miniupnpd[8021]: resolve_stun_host: getaddrinfo(stun.miwifi.com, 3478, ...) failed : Try again
Mon Mar 13 09:47:54 2023 daemon.err miniupnpd[8021]: STUN: Performing STUN failed: Host is unreachable
Mon Mar 13 09:47:54 2023 daemon.err miniupnpd[8021]: Performing STUN failed. EXITING
root@X-WRT

ptpt52, 2023-03-13:
你没联网，或者你可以关闭stun选项

wingssn, 2023-03-13:
但我是联网了的，好奇怪啊
[image]

wingssn, 2023-03-13:
这是我手动开关一次之后的日志，甜糖手机服务显示upnp可用，没有手动调整的时候甜糖手机服务提示upnp未开启

logread -e upnp
Mon Mar 13 23:26:23 2023 daemon.notice procd: /etc/rc.d/S19firewall: sh: can't open '/usr/share/miniupnpd/nft.include': No such file or directory
Mon Mar 13 23:26:36 2023 user.notice ucitrack: Setting up /etc/config/miniupnpd reload dependency on /etc/config/firewall
Mon Mar 13 23:26:42 2023 user.notice upnp daemon: external interface not found, not starting
Mon Mar 13 23:26:42 2023 daemon.err miniupnpd[5626]: STUN: Cannot get IP address for ext interface
Mon Mar 13 23:26:42 2023 daemon.err miniupnpd[5626]: Performing STUN failed. EXITING
Mon Mar 13 23:26:42 2023 user.notice upnp daemon: external interface not found, not starting
Mon Mar 13 23:26:42 2023 daemon.err miniupnpd[5820]: STUN: Cannot get IP address for ext interface
Mon Mar 13 23:26:42 2023 daemon.err miniupnpd[5820]: Performing STUN failed. EXITING
Mon Mar 13 23:26:57 2023 daemon.err miniupnpd[8446]: resolve_stun_host: getaddrinfo(stun.miwifi.com, 3478, ...) failed : Try again
Mon Mar 13 23:26:57 2023 daemon.err miniupnpd[8446]: STUN: Performing STUN failed: Host is unreachable
Mon Mar 13 23:26:57 2023 daemon.err miniupnpd[8446]: Performing STUN failed. EXITING
Mon Mar 13 23:31:54 2023 daemon.notice miniupnpd[9544]: perform_stun: 1 response out of 4 received
Mon Mar 13 23:31:54 2023 daemon.notice miniupnpd[9544]: perform_stun: #0 external port changed, but assume 1:1 nat map ok
Mon Mar 13 23:31:54 2023 daemon.warn miniupnpd[9544]: STUN: ext interface pppoe-wan with private IP address 100.78.82.187 is now behind restrictive or symmetric NAT with public IP address 111.49.88.51 which does not support port forwarding
Mon Mar 13 23:31:54 2023 daemon.warn miniupnpd[9544]: NAT on upstream router blocks incoming connections set by miniupnpd
Mon Mar 13 23:31:54 2023 daemon.warn miniupnpd[9544]: Turn off NAT on upstream router or change it to full-cone NAT 1:1 type
Mon Mar 13 23:31:54 2023 daemon.warn miniupnpd[9544]: Port forwarding is now disabled
Mon Mar 13 23:31:54 2023 daemon.notice miniupnpd[9544]: HTTP listening on port 5000
Mon Mar 13 23:31:54 2023 daemon.notice miniupnpd[9544]: Listening for NAT-PMP/PCP traffic on port 5351
Mon Mar 13 23:31:54 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:31:54 2023 daemon.err miniupnpd[9544]: parse_rule_payload: unsupported dreg 9
Mon Mar 13 23:31:54 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:31:54 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:31:54 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:17 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:17 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:17 2023 daemon.err miniupnpd[9544]: parse_rule_payload: unsupported dreg 9
Mon Mar 13 23:32:17 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:17 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:17 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.err miniupnpd[9544]: parse_rule_payload: unsupported dreg 9
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.err miniupnpd[9544]: parse_rule_payload: unsupported dreg 9
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.err miniupnpd[9544]: parse_rule_payload: unsupported dreg 9
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:18 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:19 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:19 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:19 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:19 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:19 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:19 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:19 2023 daemon.err miniupnpd[9544]: parse_rule_payload: unsupported dreg 9
Mon Mar 13 23:32:19 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
[... cut ...]
Mon Mar 13 23:32:43 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:43 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:43 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:43 2023 daemon.err miniupnpd[9544]: parse_rule_payload: unsupported dreg 9
Mon Mar 13 23:32:43 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:43 2023 daemon.warn miniupnpd[9544]: unknown rule type 3
Mon Mar 13 23:32:43 2023 daemon.warn miniupnpd[9544]: unknown rule type 3