x-wrt/x-wrt #330: 大佬，有没有buffalo WZR-HP-G300NH的固件呢？

ghxwzbc, 2023-04-08:
### Describe the bug

下载的WZR-HP-AG300的固件刷上无法启动，请问大佬有没有G300NH的固件呢？

### OpenWrt version

x-wrt-22.10-b202303312336-ath79-generic-buffalo_wzr-hp-ag300h-squashfs-factory

### OpenWrt target/subtarget

x-wrt-22.10-b202303312336-ath79-generic-buffalo_wzr-hp-ag300h-squashfs-factory

### Device

x-wrt-22.10-b202303312336-ath79-generic-buffalo_wzr-hp-ag300h-squashfs-factory

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-04-09:
官方有这个型号，下个版本支持。

ghxwzbc, 2023-04-10:
多谢

closed 2023-04-17 by ptpt52