x-wrt/x-wrt #333: 请问MT7981 eMMC能驱动起来吗

guoxiaoqiao, 2023-05-07:
### Describe the bug

我有个带eMMC的板，用开源的驱动，启用eMMC就会卡死

### OpenWrt version

master

### OpenWrt target/subtarget

filogic

### Device

mt7981

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-05-07:
有dts吗？我发现emmc没有识别出来。

guoxiaoqiao, 2023-05-07:
MTK SDK下能识别吗

ptpt52, 2023-05-07:
可以的。

guoxiaoqiao, 2023-05-08:
我这样配eMMC的
&mmc0 {
	pinctrl-names = "default", "state_uhs";
	pinctrl-0 = <&mmc0_pins_default>;
	pinctrl-1 = <&mmc0_pins_uhs>;
	bus-width = <8>;
	max-frequency = <52000000>;
	cap-mmc-highspeed;
	vmmc-supply = <&reg_3p3v>;
	non-removable;
	status = "okay";
};

&pio {
	mmc0_pins_default: mmc0-pins-default {
		mux {
			groups = "emmc_45";
			function = "flash";
		};
	};

	mmc0_pins_uhs: mmc0-pins-uhs {
		mux {
			groups = "emmc_45";
			function = "flash";
		};
	};
};

运行结果是启动时报错

[   36.886523] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd/data/mrq
[   36.893763] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting mrq=000000005ef460fd cmd=5
[   36.902293] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd=5
[    6.646961] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd/data/mrq
[    6.654192] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting mrq=00000000f275a230 cmd=52
[    6.662721] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd=52
[   11.686578] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd/data/mrq
[   11.693825] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting mrq=00000000f275a230 cmd=52
[   11.702393] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd=52
[   16.726493] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd/data/mrq
[   16.733726] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting mrq=00000000f275a230 cmd=0
[   16.742179] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd=0
[   21.766482] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd/data/mrq
[   21.773715] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting mrq=00000000f275a230 cmd=8
[   21.782157] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd=8
[   26.806508] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd/data/mrq
[   26.813768] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting mrq=000000005ef460fd cmd=5
[   26.822234] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd=5
[   31.846485] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd/data/mrq
[   31.853729] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting mrq=000000005ef460fd cmd=5
[   31.862190] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd=5
[   36.886523] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd/data/mrq
[   36.893763] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting mrq=000000005ef460fd cmd=5
[   36.902293] mtk-msdc 11230000.mmc: msdc_request_timeout: aborting cmd=5

ptpt52, 2023-05-08:
@guoxiaoqiao 
x-wrt/openwrt代码，我这emmc根本不是别。但是sdk的代码，可以正常识别。你用哪个代码？

guoxiaoqiao, 2023-05-08:
我的也是没识别到，代码基于OpenWrt Master

ptpt52, 2023-05-08:
@guoxiaoqiao 
很奇怪，为什么7986的可以识别的emmc，基本上同一个套代码。Bananapi BPi-R3 就是有sdcard或者emmc的设备。它可以。

guoxiaoqiao, 2023-05-26:
23.05修正了，关闭问题

closed 2023-05-26 by guoxiaoqiao