x-wrt/x-wrt #341: miwifi-r3 crash log

ptpt52, 2023-07-11:
### Describe the bug

```
<1>[ 3518.314337] CPU 0 Unable to handle kernel paging request at virtual address 702e7da4, epc == 702e7da4, ra == 80005dbc
<4>[ 3518.325262] Oops[#1]:
<4>[ 3518.327583] CPU: 0 PID: 8300 Comm: ping Not tainted 5.15.120 #0
<4>[ 3518.333633] $ 0   : 00000000 00000001 00000000 00000000
<4>[ 3518.338993] $ 4   : 7fc22f18 840e8d58 00000010 00000000
<4>[ 3518.344345] $ 8   : 00000000 00000000 00000000 81be5fc8
<4>[ 3518.349695] $12   : 81be5f28 7fc22cb0 840e8d58 00000000
<4>[ 3518.355048] $16   : 81be5edc 7ff3c978 7fc22cc8 00409801
<4>[ 3518.360401] $20   : 77e954b0 0000007c 00000000 77e94000
<4>[ 3518.365752] $24   : 8070723c 81be5f28                  
<4>[ 3518.371105] $28   : 81be4000 81be5ea8 00000000 80005dbc
<4>[ 3518.376458] Hi    : 00000000
<4>[ 3518.379394] Lo    : 00000000
<4>[ 3518.382329] epc   : 702e7da4 0x702e7da4
<4>[ 3518.386252] ra    : 80005dbc setup_frame+0x70/0x110
<4>[ 3518.391256] Status: 1100b403	KERNEL EXL IE 
<4>[ 3518.395539] Cause : 50800008 (ExcCode 02)
<4>[ 3518.399629] BadVA : 702e7da4
<4>[ 3518.402566] PrId  : 00019650 (MIPS 24KEc)
<4>[ 3518.406655] Modules linked in: ksmbd rt2800soc rt2800mmio rt2800lib qcserial pppoe ppp_async option cdc_mbim wireguard usb_wwan sierra_net sierra rt2x00soc rt2x00mmio rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nft_fib_inet mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake r8152 poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen
<4>[ 3518.407271]  nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables crc_ccitt compat chacha_mips cdc_wdm br_netfilter natflow natcap nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact
<4>[ 3518.496736]  xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 asn1_decoder nfs msdos ip6_gre ip_gre gre ifb ip6_udp_tunnel udp_tunnel sit ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel tun lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac ecb des_generic libdes cmac arc4 uas usb_storage leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug vfat fat ext4 mbcache jbd2 exfat
<4>[ 3518.585990]  usbcore nls_base usb_common mii crc32c_generic [last unloaded: ksmbd]
<4>[ 3518.682327] Process ping (pid: 8300, threadinfo=b4c25890, task=ce41e66e, tls=77e9ddfc)
<4>[ 3518.690421] Stack : 7fc231d4 00409801 77e954b0 0000007c 08100002 81be5f28 7fc231d4 80006548
<4>[ 3518.698992]         7fc22acc 7fc22b10 0000000e 81be5f28 7fc22cb0 10000000 00411159 00000000
<4>[ 3518.707558]         00000000 00000000 00000000 0000000e 00000080 00000000 00000000 00000000
<4>[ 3518.716120]         00000000 00000000 00000000 0000000e 00409801 00001050 77e96d8c 800025b0
<4>[ 3518.724682]         00000000 00000000 77e1da20 0000040a 7fc230b0 7fc23080 77e9c2b0 00001050
<4>[ 3518.733246]         ...
<4>[ 3518.735746] Call Trace:
<4>[ 3518.735750] 
<4>[ 3518.739848] [<80006548>] do_notify_resume+0x1a8/0x240
<4>[ 3518.745102] [<800025b0>] work_notifysig+0x10/0x18
<4>[ 3518.749941] 
<4>[ 3518.751458] Code: (Bad address in epc)
<4>[ 3518.751458] 
<4>[ 3518.756789] 
<4>[ 3518.758420] ---[ end trace 69591c7716874b6e ]---
```

### OpenWrt version

x

### OpenWrt target/subtarget

x

### Device

x

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-08-01:
crash 69 fixed

ptpt52, 2023-08-01:
crash 70:
mir3g
```
<4>[   81.918836] warning: catch unregister event for dev=xfrmi-test-2065
<4>[   81.925260] {natflow}:natflow_netdev_event(): catch NETDEV_UNREGISTER event for dev=xfrmi-test-2065
<1>[  421.501401] CPU 3 Unable to handle kernel paging request at virtual address 0000048c, epc == 806dacf4, ra == 80059968
<4>[  421.512022] Oops[#1]:
<4>[  421.514291] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 5.15.120 #0
<4>[  421.520362] $ 0   : 00000000 00000001 814511a0 00000001
<4>[  421.525593] $ 4   : 0000048c 00000003 00000000 00000000
<4>[  421.530819] $ 8   : 00045013 0000fa4c 0000029a 80b81360
<4>[  421.536047] $12   : 00000058 000000a1 8080174c 00000001
<4>[  421.541272] $16   : 00000000 00000000 00000003 00000000
<4>[  421.546495] $20   : 0000048c 00000000 00000000 81413a5c
<4>[  421.551735] $24   : 00000000 804c0ef8                  
<4>[  421.556964] $28   : 81480000 81413978 00000001 80059968
<4>[  421.562189] Hi    : 00000032
<4>[  421.565052] Lo    : 2d0e55ce
<4>[  421.567920] epc   : 806dacf4 _raw_spin_lock_irqsave+0x1c/0x74
<4>[  421.573654] ra    : 80059968 try_to_wake_up+0x4c/0x6a0
<4>[  421.578786] Status: 1100fc02	KERNEL EXL 
<4>[  421.582704] Cause : 40800008 (ExcCode 02)
<4>[  421.586692] BadVA : 0000048c
<4>[  421.589559] PrId  : 0001992f (MIPS 1004Kc)
<4>[  421.593633] Modules linked in: ksmbd qcserial pppoe ppp_async option l2tp_ppp cdc_mbim wireguard usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 nft_fib_inet mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_DNETMAP xt_DHCPMAC xt_CT xt_CLASSIFY xfrm_interface usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake r8152 poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue
<4>[  421.594139]  nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables crc_ccitt compat_xtables compat chacha_mips cdc_wdm br_netfilter natflow natcap nf_nat nf_conntrack ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic
<4>[  421.680823]  act_skbedit act_mirred act_gact ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink jool_siit jool nf_defrag_ipv4 jool_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 asn1_decoder nfs msdos bonding tls ip6_gre ip_gre gre ifb nat46 nf_defrag_ipv6 ip6_vti ip_vti sit l2tp_netlink l2tp_core ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag tun ovpn_dco_v2 udp_tunnel ip6_udp_tunnel xfrm_user xfrm_ipcomp af_key xfrm_algo lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg
<4>[  421.768084]  sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac echainiv ecb des_generic libdes cmac chacha20poly1305 cbc authencesn authenc arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd ohci_platform ohci_hcd softdog fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug vfat fat ext4 mbcache jbd2 exfat usbcore nls_base usb_common mii crc32c_generic [last unloaded: ksmbd]
<4>[  421.897782] Process swapper/3 (pid: 0, threadinfo=48032edd, task=0a6b259b, tls=00000000)
<4>[  421.905869] Stack : 80860000 00000000 00000001 00000000 844611a0 00000000 00000000 80059968
<4>[  421.914264]         81413aa0 8160084c 00000002 81600800 00000000 81413a28 fffffff5 805fc8c0
<4>[  421.922631]         844611a0 00000000 8091dd80 00000000 85d07e64 00000000 82f33694 00000000
<4>[  421.930990]         00000003 00000000 00000000 81413a5c 00000001 801dae0c 8272f000 81413a84
<4>[  421.939348]         81413aa0 80a53920 fffffff4 8006f924 8272f000 80860000 846fe400 805fc6e0
<4>[  421.947709]         ...
<4>[  421.950152] Call Trace:
[... cut ...]
<4>[  422.100725] [<8036e03c>] plat_irq_dispatch+0x98/0xcc
<4>[  422.105683] [<80003548>] except_vec_vi_end+0xb8/0xc4
<4>[  422.110643] [<806daa50>] r4k_wait_irqoff+0x1c/0x24
<4>[  422.115417] [<806dab1c>] default_idle_call+0x28/0x4c
<4>[  422.120368] [<8005eae8>] do_idle+0xd0/0x174
<4>[  422.124560] [<8005ee4c>] cpu_startup_entry+0x2c/0x34
<4>[  422.129514] 
<4>[  422.131001] Code: 32100001  000000c0  24030001 <c0850000> 14a00005  00000000  00600825  e0810000  1020fffa 
<4>[  422.140754] 
<4>[  422.142302] ---[ end trace 60f8dae2cae7326d ]---

```

ptpt52, 2023-08-01:
crash 70:
mir3p
```
<3>[ 3704.882740] ntfs3: Unknown parameter 'nls'
<3>[ 3704.923204] squashfs: Unknown parameter 'iocharset'
<4>[ 3784.724872] ubiblock0_0: Can't mount, would change RO state
<4>[ 3784.730852] ubiblock0_0: Can't mount, would change RO state
<4>[ 3924.637594] Unhandled kernel unaligned access[#1]:
<4>[ 3924.642440] CPU: 3 PID: 23024 Comm: modprobe Not tainted 5.15.120 #0
<4>[ 3924.648839] $ 0   : 00000000 00000001 00000000 fcfac5d7
<4>[ 3924.654156] $ 4   : 8fbedde0 8fbee6a0 00000001 8085c3c4
<4>[ 3924.659461] $ 8   : 00000001 0012eee4 00000001 0000000a
<4>[ 3924.664779] $12   : 0000001f 00000391 00000042 018b0567
<4>[ 3924.670105] $16   : 90010078 8fbf0000 80860000 00000000
<4>[ 3924.675408] $20   : 80a20000 00010000 00010001 00402698
<4>[ 3924.680718] $24   : 00000000 00000000                  
<4>[ 3924.686045] $28   : 87428000 87429dd0 00000000 900100a0
<4>[ 3924.691374] Hi    : 00000000
<4>[ 3924.694293] Lo    : 000001f8
<4>[ 3924.697199] epc   : 9a868010 0x9a868010
<4>[ 3924.701076] ra    : 900100a0 init_module+0x28/0x17c [ksmbd]
<4>[ 3924.707064] Status: 1100fc03	KERNEL EXL IE 
<4>[ 3924.711324] Cause : 40800010 (ExcCode 04)
<4>[ 3924.715389] BadVA : 00000002
<4>[ 3924.718321] PrId  : 0001992f (MIPS 1004Kc)
<4>[ 3924.722460] Modules linked in: ksmbd(+) qcserial pppoe ppp_async option l2tp_ppp cdc_mbim wireguard usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 nft_fib_inet mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_DNETMAP xt_DHCPMAC xt_CT xt_CLASSIFY xfrm_interface usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake r8152 poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue
<4>[ 3924.723198]  nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables hwmon crc_ccitt compat_xtables compat chacha_mips cdc_wdm br_netfilter natflow natcap nf_nat nf_conntrack ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow
<4>[ 3924.810157]  cls_basic act_skbedit act_mirred act_gact ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink jool_siit jool nf_defrag_ipv4 jool_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 asn1_decoder nfs msdos bonding tls ip6_gre ip_gre gre ifb nat46 nf_defrag_ipv6 ip6_vti ip_vti sit l2tp_netlink l2tp_core ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag tun ovpn_dco_v2 udp_tunnel ip6_udp_tunnel xfrm_user xfrm_ipcomp af_key xfrm_algo lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 crypto_user algif_skcipher algif_rng algif_hash
<4>[ 3924.897299]  algif_aead af_alg sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac echainiv ecb des_generic libdes cmac chacha20poly1305 cbc authencesn authenc arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd ohci_platform ohci_hcd softdog fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug vfat fat ext4 mbcache jbd2 exfat usbcore nls_base usb_common mii crc32c_generic [last unloaded: ksmbd]
<4>[ 3925.028017] Process modprobe (pid: 23024, threadinfo=c2d039f9, task=7974e2b5, tls=77edfdfc)
<4>[ 3925.036413] Stack : 800b9ef8 8fbee568 8077f60c 8077f62c 00000000 fcfac5d7 90010078 00000000
<4>[ 3925.044851]         80860000 80001638 808c88c4 00006822 81023d80 8077f364 00000000 00000000
<4>[ 3925.053300]         00000000 00000000 8080f2bc 8006792c 000a6465 736b3601 834b34e0 00000000
<4>[ 3925.061768]         6e72656b 00006c65 00000000 00000000 00000000 00000000 00000000 00000000
<4>[ 3925.070262]         00000000 fcfac5d7 8fbee520 00000000 836f0700 c22a9fe0 77dbdff0 800bb3dc
<4>[ 3925.078698]         ...
<4>[ 3925.081196] Call Trace:
<4>[ 3925.081212] 
<4>[ 3925.085174] [<800b9ef8>] module_sect_read+0x0/0xdc
[... cut ...]
<4>[ 3925.095592] [<80001638>] do_one_initcall+0x50/0x1b4
<4>[ 3925.100527] [<8006792c>] pick_next_task_rt+0x90/0x108
<4>[ 3925.105871] [<800bb3dc>] do_init_module+0x48/0x1f4
<4>[ 3925.110757] [<800bdeac>] sys_init_module+0x150/0x18c
<4>[ 3925.115826] [<80014550>] syscall_common+0x34/0x58
<4>[ 3925.120632] 
<4>[ 3925.122165] Code: 07860428  0a113010  02080801 <c3200002> 00040010  40000005  83102020  4a000002  00c40000 
<4>[ 3925.132009] 
<4>[ 3925.134616] ---[ end trace 619f107b50beb728 ]---

```

ptpt52, 2023-08-02:
70:
mt7621-totolink,x5000r
```
<4>[298305.813111] Kernel bug detected[#1]:
<4>[298305.816837] CPU: 1 PID: 7846 Comm: dnsmasq Not tainted 5.15.120 #0
<4>[298305.823120] $ 0   : 00000000 00000001 8085c3c4 00000001
<4>[298305.828460] $ 4   : 87500800 00085087 00000ed4 87500000
<4>[298305.833787] $ 8   : 0023022d 77edfef4 01c2a1e1 01c2a1e1
<4>[298305.839114] $12   : 95ff3333 87e05e68 00000000 0000a8c6
<4>[298305.844439] $16   : 81402700 87500000 80b429a0 84d95e40
<4>[298305.849766] $20   : 87500f00 87e05d3c 00000ed4 77eed000
<4>[298305.855093] $24   : 00000000 77e3fa60                  
<4>[298305.860416] $28   : 87e04000 87e05c90 00000000 804cb540
<4>[298305.865744] Hi    : 00000000
<4>[298305.868702] Lo    : 00000000
<4>[298305.871657] epc   : 80178998 kfree+0xdc/0x26c
<4>[298305.876100] ra    : 804cb540 skb_release_data+0x174/0x1b0
<4>[298305.881597] Status: 1100fc03	KERNEL EXL IE 
<4>[298305.885876] Cause : 50800034 (ExcCode 0d)
<4>[298305.889957] PrId  : 0001992f (MIPS 1004Kc)
<4>[298305.894126] Modules linked in: pppoe ppp_async wireguard pptp pppox ppp_mppe ppp_generic nft_fib_inet mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp
<4>[298305.894871]  nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables hwmon crc_ccitt compat chacha_mips br_netfilter asn1_decoder natflow natcap nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink
<4>[298305.982132]  ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_gre ip_gre gre ifb sit ip6_tunnel tunnel6 tunnel4 ip_tunnel tun ovpn_dco_v2 udp_tunnel ip6_udp_tunnel nls_utf8 nls_iso8859_1 nls_base crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha1_generic seqiv jitterentropy_rng drbg hmac ecb cmac arc4 leds_gpio softdog gpio_button_hotplug crc32c_generic
<4>[298306.106404] Process dnsmasq (pid: 7846, threadinfo=6ccd9c5d, task=b563f648, tls=77ef6dfc)
<4>[298306.114672] Stack : 00000000 804d25dc 87e05d04 fffffff5 77eed000 804d3bb0 00000000 84d95e40
<4>[298306.123149]         00000000 804cb540 00000000 87e05cc0 84d95e40 804ca600 84d95e40 84d95e40
<4>[298306.131621]         00000000 87e05ee0 85d154a0 804cfaf8 00000000 804d285c 87e05d3c 00000000
<4>[298306.140093]         8605f800 804d2e68 00000000 804d2ea4 00000000 00000000 8605f800 8055f82c
<4>[298306.148565]         863ae400 84d95480 fffffffc 804fc98c 80860000 80860000 00000014 00000000
<4>[298306.157038]         ...
<4>[298306.159573] Call Trace:
<4>[298306.162094] [<80178998>] kfree+0xdc/0x26c
<4>[298306.166200] [<804cb540>] skb_release_data+0x174/0x1b0
<4>[298306.171350] [<804cfaf8>] kfree_skb_reason+0x44/0xa8
<4>[298306.176328] [<804d2e68>] skb_free_datagram+0x1c/0x58
<4>[298306.181369] [<8055f82c>] netlink_recvmsg+0x12c/0x36c
<4>[298306.186426] [<804bb66c>] ____sys_recvmsg+0x88/0x118
<4>[298306.191403] [<804bd800>] ___sys_recvmsg+0x7c/0xdc
<4>[298306.196203] [<804be1ac>] sys_recvmsg+0x48/0xb4
<4>[298306.200736] [<80014550>] syscall_common+0x34/0x58
<4>[298306.205538] 
<4>[298306.207112] Code: 02271826  02242021  2c630001 <00030336> 8e030078  7c0428a0  00252c02  00e31826  00651826 
<4>[298306.216990] 
<4>[298306.218722] ---[ end trace fea8971e2dce8852 ]---
```

ptpt52, 2023-08-03:
70: mir3p
```
<4>[  537.183772] ksmbd_smb2_check_message: 44 callbacks suppressed
<3>[  537.183810] ksmbd: cli req too short, len 112 not 116. cmd:10 mid:67342
<3>[  537.200605] ksmbd: cli req too short, len 112 not 116. cmd:10 mid:67343
<3>[  537.260116] ksmbd: cli req too short, len 112 not 116. cmd:10 mid:67374
<3>[  537.271401] ksmbd: cli req too short, len 112 not 116. cmd:10 mid:67382
<3>[  537.381722] ksmbd: cli req too short, len 112 not 116. cmd:10 mid:67456
<3>[  537.391830] ksmbd: cli req too short, len 112 not 116. cmd:10 mid:67460
<3>[  537.421104] ksmbd: cli req too short, len 112 not 116. cmd:10 mid:67482
<3>[  537.431253] ksmbd: cli req too short, len 112 not 116. cmd:10 mid:67486
<3>[  538.584005] ksmbd: cli req too short, len 112 not 116. cmd:10 mid:67973
<3>[  538.603532] ksmbd: cli req too short, len 112 not 116. cmd:10 mid:67988
<1>[  539.996203] CPU 2 Unable to handle kernel paging request at virtual address 44420c20, epc == 801427bc, ra == 801427b0
<4>[  540.006843] Oops[#1]:
<4>[  540.009105] CPU: 2 PID: 36 Comm: kcompactd0 Not tainted 5.15.120 #0
<4>[  540.015349] $ 0   : 00000000 00000001 44420c20 c0180004
<4>[  540.020573] $ 4   : 808ea8e0 00000004 ffffffff 00000000
<4>[  540.025791] $ 8   : 006fb000 8085c3c4 00000000 00000000
<4>[  540.031013] $12   : 00000651 0000000b 00000001 00000000
<4>[  540.036256] $16   : 80dd8ca0 0001c017 00000000 81519e30
<4>[  540.041473] $20   : 808f0000 00000018 00000000 00000000
<4>[  540.046691] $24   : 00000000 806daa34                  
<4>[  540.051909] $28   : 81518000 81519d20 0001c400 801427b0
<4>[  540.057128] Hi    : 0000023b
<4>[  540.059990] Lo    : 00000008
<4>[  540.062852] epc   : 801427bc isolate_migratepages_block+0x764/0xdd0
<4>[  540.069123] ra    : 801427b0 isolate_migratepages_block+0x758/0xdd0
<4>[  540.075366] Status: 1100fc02	KERNEL EXL 
<4>[  540.079277] Cause : 4080000c (ExcCode 03)
<4>[  540.083264] BadVA : 44420c20
<4>[  540.086127] PrId  : 0001992f (MIPS 1004Kc)
<4>[  540.090200] Modules linked in: ksmbd qcserial pppoe ppp_async option l2tp_ppp cdc_mbim wireguard usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 nft_fib_inet mt7615e mt7615_common mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_DNETMAP xt_DHCPMAC xt_CT xt_CLASSIFY xfrm_interface usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake r8152 poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref
<4>[  540.090699]  nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables hwmon crc_ccitt compat_xtables compat chacha_mips cdc_wdm br_netfilter natflow natcap nf_nat nf_conntrack sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit
<4>[  540.177995]  act_mirred act_gact ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink jool_siit jool nf_defrag_ipv4 jool_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 asn1_decoder nfs msdos bonding tls ip6_gre ip_gre gre ifb nat46 nf_defrag_ipv6 ip6_vti ip_vti sit l2tp_netlink l2tp_core ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag tun ovpn_dco_v2 udp_tunnel ip6_udp_tunnel xfrm_user xfrm_ipcomp af_key xfrm_algo lockd sunrpc grace autofs4 dns_resolver nls_iso8859_1 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha256_generic
<4>[  540.265324]  libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac echainiv ecb des_generic libdes cmac chacha20poly1305 cbc authencesn authenc arc4 nls_utf8 nls_cp437 ntfs3 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd ohci_platform ohci_hcd softdog fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug vfat fat ext4 mbcache jbd2 exfat usbcore nls_base usb_common mii crc32c_generic [last unloaded: ksmbd]
[... cut ...]
<4>[  540.449302] [<801427bc>] isolate_migratepages_block+0x764/0xdd0
<4>[  540.455224] [<80144218>] compact_zone+0x934/0xbd0
<4>[  540.459913] [<80144694>] kcompactd_do_work+0x154/0x204
<4>[  540.465034] [<8014492c>] kcompactd+0x1e8/0x268
<4>[  540.469462] [<80050d30>] kthread+0x130/0x154
<4>[  540.473724] [<80003038>] ret_from_kernel_thread+0x14/0x1c
<4>[  540.479109] 
<4>[  540.480587] Code: 8e020008  8e030004  ac620004 <ac430000> 24020100  ae020004  24020122  ae020008  8e030004 
<4>[  540.490331] 
<4>[  540.492394] ---[ end trace ce8a95145e0fba18 ]---
```

ptpt52, 2023-08-12:
7621: crash73
```
<6>[   44.210932] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
<4>[25509.830916] Kernel bug detected[#1]:
<4>[25509.834530] CPU: 0 PID: 197 Comm: napi/mtk_eth-6 Not tainted 5.15.123 #0
<4>[25509.841212] $ 0   : 00000000 00000001 e2038144 00000001
<4>[25509.846441] $ 4   : 86fee240 1dfca504 83935078 e2038144
<4>[25509.851669] $ 8   : 83935700 000005a0 839355f0 41fe06af
<4>[25509.856887] $12   : c30fa8c0 f0215220 b6daf792 7bd1bb01
<4>[25509.862107] $16   : 00000000 00000000 00000000 e20386e4
<4>[25509.867325] $20   : e2038144 000005a0 00000578 80000000
<4>[25509.872543] $24   : 00000000 802f8324                  
<4>[25509.877761] $28   : 81c40000 81c419f8 80a50000 804d21bc
<4>[25509.882980] Hi    : 00000012
<4>[25509.885843] Lo    : 0000313d
<4>[25509.888706] epc   : 804c79e8 skb_copy_and_csum_bits+0x22c/0x2d8
<4>[25509.894627] ra    : 804d21bc skb_segment+0x308/0xcd0
<4>[25509.899584] Status: 11000403	KERNEL EXL IE 
<4>[25509.903765] Cause : 50800034 (ExcCode 0d)
<4>[25509.907765] PrId  : 0001992f (MIPS 1004Kc)
<4>[25509.911839] Modules linked in: pppoe ppp_async wireguard pptp pppox ppp_mppe ppp_generic nft_fib_inet mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp
<4>[25509.912304]  nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables hwmon crc_ccitt compat chacha_mips br_netfilter asn1_decoder natflow natcap nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink
<4>[25509.999197]  ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_gre ip_gre gre ifb sit ip6_tunnel tunnel6 tunnel4 ip_tunnel tun ovpn_dco_v2 udp_tunnel ip6_udp_tunnel nls_utf8 nls_iso8859_1 nls_base crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha1_generic seqiv jitterentropy_rng drbg hmac ecb cmac arc4 leds_gpio softdog gpio_button_hotplug crc32c_generic
<4>[25510.123096] Process napi/mtk_eth-6 (pid: 197, threadinfo=9a2641d3, task=6ede895a, tls=00000000)
<4>[25510.131771] Stack : 804d20b8 804c8b00 8091dd80 00000001 81cc0a6c 807ef358 86da2300 86fee780
<4>[25510.140124]         00000578 00000578 86da2180 00000078 000005a0 869d504c 00000000 804d21bc
<4>[25510.148473]         86fee780 833984a0 83935078 000001c4 00000000 00000000 00000028 00000578
<4>[25510.156822]         00000050 86da2180 00000000 000005a0 00000078 00000000 00000000 00000028
<4>[25510.165169]         00000000 86fee240 00000011 86fee780 ffffffd8 00000000 83398548 83398558
<4>[25510.173520]         ...
<4>[25510.175957] Call Trace:
<4>[25510.178386] [<804c79e8>] skb_copy_and_csum_bits+0x22c/0x2d8
<4>[25510.183957] [<804d21bc>] skb_segment+0x308/0xcd0
<4>[25510.188565] [<805c38c8>] tcp_gso_segment+0x118/0x5a0
<4>[25510.193537] [<805dbcd8>] inet_gso_segment+0x174/0x3c0
<4>[25510.198591] [<804e7820>] skb_mac_gso_segment+0xa8/0x11c
<4>[25510.203802] [<804e7914>] __skb_gso_segment+0x80/0x1b4
<4>[25510.208884] [<82e27a38>] natflow_session_learn+0x1f94/0x70f4 [natflow]
<4>[25510.215419] [<804da8c8>] __skb_get_hash+0x88/0x124
<4>[25510.220205] 
<4>[25510.221684] Code: afa20048  1620ffe1  02549021 <00160336> 8fbf003c  8fbe0038  8fb70034  8fb60030  8fb5002c 
<4>[25510.231430] 
<4>[25510.233037] ---[ end trace f58eda11fff06fac ]---
```

ptpt52, 2023-08-12:
73: rm-ax6000
```
<1>[ 5704.199082] Unable to handle kernel access to user memory outside uaccess routines at virtual address 0000000000000008
<1>[ 5704.209778] Mem abort info:
<1>[ 5704.212558]   ESR = 0x0000000096000005
<1>[ 5704.216290]   EC = 0x25: DABT (current EL), IL = 32 bits
<1>[ 5704.221598]   SET = 0, FnV = 0
<1>[ 5704.224643]   EA = 0, S1PTW = 0
<1>[ 5704.227767]   FSC = 0x05: level 1 translation fault
<1>[ 5704.232651] Data abort info:
<1>[ 5704.235521]   ISV = 0, ISS = 0x00000005
<1>[ 5704.239346]   CM = 0, WnR = 0
<1>[ 5704.242301] user pgtable: 4k pages, 39-bit VAs, pgdp=0000000046a8e000
<1>[ 5704.248727] [0000000000000008] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
<0>[ 5704.257407] Internal error: Oops: 96000005 [#1] SMP
<7>[ 5704.262269] Modules linked in: pppoe ppp_async l2tp_ppp wireguard pptp pppox ppp_mppe ppp_generic nft_fib_inet mt7915e mt76_connac_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT chacha_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_DNETMAP xt_DHCPMAC xt_CT xt_CLASSIFY xfrm_interface ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake poly1305_neon nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6
<7>[ 5704.262421]  nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c libchacha ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables crc_ccitt compat_xtables compat br_netfilter asn1_decoder natflow natcap nf_nat nf_conntrack crypto_safexcel sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark
<7>[ 5704.348602]  ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink jool_siit jool nf_defrag_ipv4 jool_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 bonding tls ip6_gre ip_gre gre ifb nat46 nf_defrag_ipv6 ip6_vti ip_vti sit l2tp_netlink l2tp_core ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel tunnel6 tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag tun ovpn_dco_v2 udp_tunnel ip6_udp_tunnel xfrm_user xfrm_ipcomp af_key xfrm_algo autofs4 nls_utf8 nls_iso8859_1 crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic seqiv md5 echainiv des_generic libdes chacha20poly1305 cbc authencesn authenc arc4 leds_ws2812b leds_gpio gpio_button_hotplug usbcore usb_common
<7>[ 5704.505682] CPU: 3 PID: 2362 Comm: hostapd Not tainted 5.15.123 #0
<7>[ 5704.511845] Hardware name: Xiaomi Redmi Router AX6000 (stock layout) (DT)
<7>[ 5704.518611] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
<7>[ 5704.525553] pc : sta_set_sinfo+0xa1c/0xc40 [mac80211]
<7>[ 5704.530623] lr : sta_set_sinfo+0x5bc/0xc40 [mac80211]
<7>[ 5704.535673] sp : ffffffc00d2f37a0
<7>[ 5704.538972] x29: ffffffc00d2f37a0 x28: 0000000000000001 x27: ffffffc00d2f3dd0
<7>[ 5704.546090] x26: ffffff8000069080 x25: ffffff800a5948c0 x24: ffffff800c240735
<7>[ 5704.553205] x23: ffffffc00d2f3a38 x22: ffffff800a5948c0 x21: 0000058d1783bf82
<7>[ 5704.560321] x20: ffffff800c240000 x19: ffffff8009337400 x18: 0000000000000000
<7>[ 5704.567437] x17: 0000000000001180 x16: ffffffc008e65000 x15: 00000000000008c0
<7>[ 5704.574553] x14: 00013e6049d7efca x13: 0016e8fbd5f5c56a x12: ffffffc008837678
<7>[ 5704.581669] x11: 00000000fa83b2da x10: 00000000000008a0 x9 : ffffffc00d2f3540
<7>[ 5704.588785] x8 : ffffff800014b700 x7 : ffffff8004378880 x6 : 0000000000000001
<7>[ 5704.595900] x5 : ffffff80043783a0 x4 : 0000000000000000 x3 : 0000000000000000
<7>[ 5704.603016] x2 : ffffff8009337444 x1 : 0000000000000001 x0 : 0000000000000004
<7>[ 5704.610133] Call trace:
<7>[ 5704.612566]  sta_set_sinfo+0xa1c/0xc40 [mac80211]
<7>[ 5704.617279]  sta_set_sinfo+0xb58/0xc40 [mac80211]
<7>[ 5704.621983]  sta_info_destroy_addr_bss+0x4c/0x70 [mac80211]
<7>[ 5704.627552]  ieee80211_color_change_finish+0x1c18/0x1e80 [mac80211]
<7>[ 5704.633815]  cfg80211_check_station_change+0x1384/0x4700 [cfg80211]
<7>[ 5704.640075]  genl_family_rcv_msg_doit+0xb4/0x110
<7>[ 5704.644680]  genl_rcv_msg+0xd0/0x1c0
<7>[ 5704.648241]  netlink_rcv_skb+0x58/0x120
<7>[ 5704.652061]  genl_rcv+0x34/0x50
<7>[ 5704.655188]  netlink_unicast+0x1f0/0x2ec
<7>[ 5704.659096]  netlink_sendmsg+0x19c/0x3d0
<7>[ 5704.663003]  ____sys_sendmsg+0x27c/0x2b0
<7>[ 5704.666913]  ___sys_sendmsg+0x80/0xf0
<7>[ 5704.670561]  __sys_sendmsg+0x44/0xa0
<7>[ 5704.674127]  __arm64_sys_sendmsg+0x20/0x30
<7>[ 5704.678208]  invoke_syscall.constprop.0+0x4c/0xe0
<7>[ 5704.682896]  do_el0_svc+0x40/0xd0
<7>[ 5704.686196]  el0_svc+0x14/0x50
<7>[ 5704.689237]  el0t_64_sync_handler+0xe0/0x110
<7>[ 5704.693490]  el0t_64_sync+0x15c/0x160
<0>[ 5704.697141] Code: d3441c63 12000c00 8b030ca3 f9409c63 (f9400465) 
<4>[ 5704.703214] ---[ end trace 0aaf8c2684f19205 ]---

```

BioXeon, 2023-12-04:
How can I switch KERNEL from 5.10.* to 5.15.* for my Xiaomi Mi Router R3 so that build my own builds with this kernel? I use your repo for cloning changes, updates and so on from it. TY for your answer

ptpt52, 2023-12-05:
> How can I switch KERNEL from 5.10.* to 5.15.* for my Xiaomi Mi Router R3 so that build my own builds with this kernel? I use your repo for cloning changes, updates and so on from it. TY for your answer

you do not need the kernel 5.15 I think.
if use kernel 5.15 you would suffer the crash issue
why you need 5.15?

BioXeon, 2023-12-05:
You see, I use for connecting the mesh option on both sides of my APs (routers - Xiaomi Mi Router R3) and how the practice seemed I have had better experience when I used firmware with kernel 5.15.* (the latest was 5.15.125 before you had switched it in your repo).

BioXeon, 2023-12-05:
When I used the latest build with 5.10.*, I had issue with the mesh option after rebooting my AP. It was like my options (standard options - 802.11s & WPA3, and so on) couldn't be accepted for the mesh connection, but when I use the firmware with 5.15.125 they are working fine.*

ptpt52, 2023-12-05:
@BioXeon  you may use Fakemesh on x-wrt.
https://fakemesh.com/
7620 on x-wrt would likely use kernel 5.10

BioXeon, 2023-12-05:
No, I didn't use Fakemesh, I use -> IEEE 802.11s

BioXeon, 2023-12-05:
You can use as example this video - YouTube /watch?v=vVoZppb_FR0

ptpt52, 2023-12-05:
@BioXeon I know 80211s mesh but...

X-WRT's built-in FakeMesh feature is a robust mesh networking method that offers greater stability and higher performance compared to 802.11s. It's user-friendly and straightforward. I strongly recommend using the FakeMesh network setup. Additionally, it's important to note that the kernel for X-WRT's 7620 series devices will remain fixed at version 5.10 for an extended period and won't receive further updates.

Bob-YsPan, 2023-12-20:
> you do not need the kernel 5.15 I think. if use kernel 5.15 you would suffer the crash issue why you need 5.15?

Have you found the reason why linux 5.15 causes the mt7620 to crash?

I using the mi wifi r3, but the linux 5.10 kernel seems doesn't have some kernel modules matches the kernel version like ksmbd so it can't use some functions like the samba shares.

Or is it already append to the recent version that uses linux 510?