x-wrt/x-wrt #342: 用默认配置自编译固件, 接口里面无4G qmi协议选项. 同一天您编译的固件却有

668168, 2023-07-14:
### Describe the bug

同一台设备:  R619AC 【竞斗云】 IPQ40XX with 宽翼B806u 4g 模块 BM806U
用同一个版本:x-wrt-23.06-b202307110013 

同一天您编译的固件有4G qmi 协议. x-wrt-23.06-b202307110013-ipq40xx-generic-p2w_r619ac-128m-squashfs-factory.ubi
用默认配置自编译固件, 接口里面无4G qmi协议选项. openwrt-ipq40xx-generic-p2w_r619ac-128m-squashfs-factory.ubi

刷机方式: factory

### OpenWrt version

OpenWrt SNAPSHOT r23720+2-60b7e77f72 / LuCI Master git-23.191.58219-a5de4c1

### OpenWrt target/subtarget

ipq40xx/generic  

### Device

 P&W R619AC 128M

### Image kind

Self-built image

### Steps to reproduce

我抓取official build 版本, 在接口-> 协议 里面 有qmi 协议.
但
self-build , 同一天版本, 刷机后无qmi协议.

抓取的master分支, 如下时间点的codebase.
```
commit da7d9be3ce34e5e133cf8a0574b26a1f71b6666f (origin/master, origin/HEAD)
Author: Chen Minqiang <ptpt52@gmail.com>
Date:   Mon Jun 12 13:09:38 2023 +0800

    armsr: reisze vmdk/vdi to 8G

```

### Diffconfig

```text
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_ipq40xx_generic=y
CONFIG_TARGET_ipq40xx_generic_DEVICE_p2w_r619ac-128m=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_ARIA2_ASYNC_DNS=y
CONFIG_ARIA2_BITTORRENT=y
CONFIG_ARIA2_COOKIE=y
CONFIG_ARIA2_LIBXML2=y
CONFIG_ARIA2_METALINK=y
CONFIG_ARIA2_OPENSSL=y
CONFIG_ARIA2_SFTP=y
CONFIG_ARIA2_WEBSOCKET=y
CONFIG_BASE_CONFIG_SETTING_LUCI_LOGIN="admin"
CONFIG_BUILD_PATENTED=y
CONFIG_BUSYBOX_CONFIG_BASE64=y
CONFIG_BUSYBOX_CONFIG_CKSUM=y
CONFIG_BUSYBOX_CONFIG_DIFF=y
CONFIG_BUSYBOX_CONFIG_NOHUP=y
CONFIG_BUSYBOX_CONFIG_TIMEOUT=y
CONFIG_BUSYBOX_CONFIG_VCONFIG=y
# CONFIG_DRIVER_11AX_SUPPORT is not set
# CONFIG_FEED_x is not set
CONFIG_HTOP_LMSENSORS=y
[... 1124 lines cut ...]
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
# CONFIG_PACKAGE_dnsmasq_full_auth is not set
# CONFIG_PACKAGE_dnsmasq_full_conntrack is not set
# CONFIG_PACKAGE_dnsmasq_full_dhcpv6 is not set
# CONFIG_PACKAGE_dnsmasq_full_dnssec is not set
# CONFIG_PACKAGE_dnsmasq_full_noid is not set
# CONFIG_PACKAGE_dnsmasq_full_tftp is not set
CONFIG_PACKAGE_e2fsprogs=m
CONFIG_PACKAGE_ethtool=m
CONFIG_PACKAGE_kmod-nf-conntrack-netlink=m
CONFIG_PACKAGE_libcomerr=m
CONFIG_PACKAGE_libext2fs=m
CONFIG_PACKAGE_libgmp=m
# CONFIG_PACKAGE_libnetfilter-conntrack is not set
# CONFIG_PACKAGE_libnettle is not set
CONFIG_PACKAGE_libsmartcols=m
CONFIG_PACKAGE_libss=m
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-07-14:
除非你直接拷贝模版配置config编译，如果自己make menuconfig配置，你得把qmi相关的模块和应用自己选上
或者你可以拷贝config来后，先把config =m的改成 =y的，再make menuconfig。

closed 2023-07-14 by ptpt52

668168, 2023-07-14:
谢谢大佬这么快帮忙答疑. 
目前我的确是这么做的, 自己make menuconfig,手动把您默认模板里面这些m 的模块全设置为y. 
因为不确定哪些模块和qmi相关, 只好用笨办法, 先全加上,测试ok ,再慢慢删.
```
CONFIG_TARGET_DEVICE_PACKAGES_ipq40xx_generic_DEVICE_p2w_r619ac-128m="aria2 ath10k-board-qca4019 ath10k-firmware-qca4019-ct blockd block-mount chat comgt comgt-ncm dawn ethtool gre iptables-mod-dhcpmac iptables-mod-dnetmap iptables-mod-ipsec jool-tools-netfilter kmod-ath10k-ct kmod-bonding kmod-crypto-authenc kmod-crypto-cbc kmod-crypto-chacha20poly1305 kmod-crypto-deflate kmod-crypto-des kmod-crypto-echainiv kmod-crypto-md4 kmod-crypto-md5 kmod-crypto-sha256 kmod-dnsresolver kmod-fs-autofs4 kmod-fs-exfat kmod-fs-ext4 kmod-fs-ksmbd kmod-fs-msdos kmod-fs-nfs kmod-fs-nfs-common kmod-fs-nfs-v4 kmod-fs-ntfs3 kmod-fs-smbfs-common kmod-fs-vfat kmod-ifb kmod-inet-diag kmod-ip6-vti kmod-ipsec kmod-ipsec4 kmod-ipsec6 kmod-ipt-compat-xtables kmod-ipt-dhcpmac kmod-ipt-dnetmap kmod-ipt-ipsec kmod-ip-vti kmod-jool-netfilter kmod-l2tp kmod-lib-crc16 kmod-lib-zlib-deflate kmod-lib-zlib-inflate kmod-mii kmod-nat46 kmod-nf-conntrack-netlink kmod-nls-cp437 kmod-oid-registry kmod-pppol2tp kmod-usb2 kmod-usb3 kmod-usb-ehci kmod-usb-net kmod-usb-net-cdc-ether kmod-usb-net-cdc-mbim kmod-usb-net-cdc-ncm kmod-usb-net-huawei-cdc-ncm kmod-usb-net-ipheth kmod-usb-net-qmi-wwan kmod-usb-net-rndis kmod-usb-net-rtl8152 kmod-usb-net-sierrawireless kmod-usb-ohci kmod-usb-printer kmod-usb-serial kmod-usb-serial-option kmod-usb-serial-pl2303 kmod-usb-serial-qualcomm kmod-usb-serial-sierrawireless kmod-usb-serial-wwan kmod-usb-storage kmod-usb-storage-extras kmod-usb-storage-uas kmod-usb-wdm kmod-usb-xhci-hcd kmod-xfrm-interface ksmbd-server libblkid libcares libevdev libgcrypt libgmp libgpg-error libimobiledevice libimobiledevice-utils liblzo libnl libnl-core libnl-genl libnl-nf libnl-route libopenssl libopenssl-conf libpcre libplist libqrencode libsqlite3 libssh2 libstdcpp libudev-zero libusb-1.0 libusbmuxd libwrap libxml2 lte-modem-p2w-r619ac luci-app-aria2 luci-app-dawn luci-app-ksmbd luci-app-openvpn luci-app-p910nd luci-app-sqm luci-i18n-aria2-zh-cn luci-i18n-dashboard-zh-cn luci-i18n-ksmbd-zh-cn luci-i18n-openvpn-zh-cn luci-i18n-p910nd-zh-cn luci-i18n-sqm-zh-cn luci-lib-json luci-mod-dashboard luci-proto-3g luci-proto-bonding luci-proto-gre luci-proto-mbim luci-proto-ncm luci-proto-qmi luci-proto-relay luci-proto-vti luci-proto-wireguard luci-proto-xfrm luci-ssl-nginx minicom nfs-utils nginx-mod-luci nginx-mod-luci-ssl nginx-mod-ubus nginx-ssl nginx-ssl-util nginx-util openssl-util openvpn-openssl p910nd ppp-mod-pppol2tp proto-bonding qrencode r8152-firmware relayd sendip sqm-scripts strongswan strongswan-charon strongswan-charon-cmd strongswan-default strongswan-ipsec strongswan-libnttfft strongswan-libtls strongswan-mod-addrblock strongswan-mod-aes strongswan-mod-af-alg strongswan-mod-attr strongswan-mod-bliss strongswan-mod-blowfish strongswan-mod-ccm strongswan-mod-chapoly strongswan-mod-cmac strongswan-mod-connmark strongswan-mod-constraints strongswan-mod-coupling strongswan-mod-ctr strongswan-mod-curve25519 strongswan-mod-des strongswan-mod-dhcp strongswan-mod-dnskey strongswan-mod-drbg strongswan-mod-duplicheck strongswan-mod-eap-identity strongswan-mod-eap-md5 strongswan-mod-eap-mschapv2 strongswan-mod-eap-radius strongswan-mod-eap-tls strongswan-mod-farp strongswan-mod-fips-prf strongswan-mod-forecast strongswan-mod-gcm strongswan-mod-gcrypt strongswan-mod-gmp strongswan-mod-gmpdh strongswan-mod-hmac strongswan-mod-kernel-netlink strongswan-mod-md4 strongswan-mod-md5 strongswan-mod-mgf1 strongswan-mod-newhope strongswan-mod-ntru strongswan-mod-openssl strongswan-mod-pem strongswan-mod-pgp strongswan-mod-pkcs1 strongswan-mod-pkcs11 strongswan-mod-pkcs12 strongswan-mod-pkcs7 strongswan-mod-pkcs8 strongswan-mod-pubkey strongswan-mod-random strongswan-mod-rc2 strongswan-mod-resolve strongswan-mod-revocation strongswan-mod-sha1 strongswan-mod-sha2 strongswan-mod-sha3 strongswan-mod-socket-default strongswan-mod-socket-dynamic strongswan-mod-sshkey strongswan-mod-stroke strongswan-mod-test-vectors strongswan-mod-unity strongswan-mod-updown strongswan-mod-vici strongswan-mod-whitelist strongswan-mod-x509 strongswan-mod-xauth-eap strongswan-mod-xauth-generic strongswan-mod-xcbc strongswan-pki strongswan-swanctl tc-tiny umbim uqmi usb-modeswitch usbmuxd usbutils uwsgi uwsgi-cgi-plugin uwsgi-luci-support uwsgi-syslog-plugin vti webui-aria2 wget-ssl wireguard-tools wpad-openssl wsdd2 wwan xfrm xl2tpd"

```