x-wrt/x-wrt #344: Netgear R8500

Fiddl3, 2023-07-25:
### Describe the bug

I wonder if anyone has managed to run X-WRT on the Netgear R8500 router. The latest compilations don't work, and version [v23.06_b202307110013](https://github.com/x-wrt/build-release/releases/tag/v23.06_b202307110013) starts up but the WiFi doesn't work, or when it is showing in wireless networks I can't connect to the network;
I did try to use 

### OpenWrt version

v23.06

### OpenWrt target/subtarget

bcm53xx/generic

### Device

Netgear R8500

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-07-26:
could you try openwrt snapshots version from here https://downloads.openwrt.org/snapshots/targets/bcm53xx/generic/

does it work?

x-wrt code is based on openwrt snapshots.

Fiddl3, 2023-07-26:
this router is not supported in "official" openwrt builds

Fiddl3, 2023-07-26:
I did try https://github.com/XOBAY/OpenWrt-R8500-firmware and in this build WIFI works (I don know how it perform)
but lan ports behave like in lastest x-wrt builds (I can not connect to router via LAN or WAN, computer does not recognize network I did try to set IP manual and DHCP)

Sherman-Liu, 2023-10-31:
Can confirm, exact same issue, I've tested on both V1&V2 revision hardware, same bug exists.

sbwml, 2024-04-07:
**Try my build, Lan gateway: http://10.0.0.1  or http://openwrt.lan**

[https://github.com/sbwml/builder/releases](https://github.com/sbwml/builder/releases/download/v23.05.3/openwrt-23.05.3-bcm53xx-generic-netgear_r8500-squashfs.chk)

[image]
[image]
[image]