x-wrt/x-wrt #352: Ruijie RG-MTFi-M520 LAN port not working

mrkenz9595, 2023-11-25:
### Describe the bug

Ruijie RG-MTFi-M520 LAN port not working, please help

### OpenWrt version

23.10_b202311242121

### OpenWrt release

23.10_b202311242121

### OpenWrt target/subtarget

23.10_b202311242121

### Device

Ruijie RG-MTFi-M520

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-11-25:
@mrkenz9595 great, I am gona fix the issue.

ptpt52, 2023-11-25:
@mrkenz9595 
try the firmware and test again
`https://downloads.x-wrt.com/rom/Downloads/ipq806x/`

mrkenz9595, 2023-11-25:
@ptpt52 working ! Thanks you very much.

closed 2023-11-25 by mrkenz9595