x-wrt/x-wrt #353: xiaomi mini no 2.4 wireless

godshades, 2023-12-02:
### Describe the bug

Hi i just flash latest version for xiaomi mini but 2.4 band has gone. 5 ghz band max tx power is 3dbm, please help
[image]

### OpenWrt version

latest

### OpenWrt release

latest

### OpenWrt target/subtarget

mips

### Device

xiaomi mini

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ptpt52, 2023-12-03:
@godshades 
I am gona post a new firmware here, please test.

ptpt52, 2023-12-03:
try:
https://downloads.x-wrt.com/rom/Downloads/x-wrt-23.10-b202312030808-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade.bin

godshades, 2023-12-03:
@ptpt52 breaked (steady orange) :(

ptpt52, 2023-12-03:
again, try the new one: https://downloads.x-wrt.com/rom/Downloads/x-wrt-23.10-b202312031209-ramips-mt7620-xiaomi_miwifi-mini-squashfs-sysupgrade.bin

godshades, 2023-12-04:
@ptpt52 after breaked, when come safe mode to flash miwifi, my router is red light, i think it's dead...
i'm finding way to fix this

godshades, 2023-12-14:
i have buy new mini and the lastest fw work
anyone know how to fix the steady red light please ping me

closed 2023-12-14 by godshades