x-wrt/x-wrt #358: mi router 3 log crash

musakilic, 2023-12-26:
Tue Dec 26 00:08:01 2023 kern.err kernel: [99308.275194] ieee80211 phy1: rt2800_wait_bbp_rf_ready: Error - BBP/RF register access failed, aborting
Tue Dec 26 00:08:01 2023 kern.warn kernel: [99308.284629] ieee80211 phy1: rt2800_config_alc_rt6352: Warning - RF busy while configuring ALC
Tue Dec 26 00:09:47 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 00:14:29 2023 kern.err kernel: [99474.355367] ieee80211 phy1: rt2800_wait_bbp_rf_ready: Error - BBP/RF register access failed, aborting
Tue Dec 26 00:14:29 2023 kern.warn kernel: [99474.364817] ieee80211 phy1: rt2800_config_alc_rt6352: Warning - RF busy while configuring ALC
Tue Dec 26 00:14:45 2023 kern.err kernel: [99481.317654] ieee80211 phy1: rt2800_wait_bbp_rf_ready: Error - BBP/RF register access failed, aborting
Tue Dec 26 00:14:45 2023 kern.warn kernel: [99481.327837] ieee80211 phy1: rt2800_config_alc_rt6352: Warning - RF busy while configuring ALC
Tue Dec 26 00:15:17 2023 kern.err kernel: [99495.398250] ieee80211 phy1: rt2800_wait_bbp_rf_ready: Error - BBP/RF register access failed, aborting
Tue Dec 26 00:15:17 2023 kern.warn kernel: [99495.407927] ieee80211 phy1: rt2800_config_alc_rt6352: Warning - RF busy while configuring ALC
Tue Dec 26 00:16:05 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED d8:3b:bf:4e:01:c9
Tue Dec 26 00:16:05 2023 user.info usteer: station d8:3b:bf:4e:01:c9 disconnected from node hostapd.phy0-ap0
Tue Dec 26 00:16:26 2023 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 3c:6a:a7:b5:34:52
Tue Dec 26 00:16:26 2023 user.info usteer: station 3c:6a:a7:b5:34:52 disconnected from node hostapd.phy1-ap0
Tue Dec 26 00:17:37 2023 kern.err kernel: [99553.037082] ieee80211 phy1: rt2800_wait_bbp_rf_ready: Error - BBP/RF register access failed, aborting
Tue Dec 26 00:17:37 2023 kern.warn kernel: [99553.046628] ieee80211 phy1: rt2800_config_alc_rt6352: Warning - RF busy while configuring ALC
Tue Dec 26 00:18:52 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 00:19:19 2023 kern.warn kernel: [99593.200965] warning: (PPI)[PRE](natcap_peer_pre_in_hook:2946)[8.218.31.137:5467->192.168.0.45:3436|ID:D4E1,IL:1104|...A..S.|S=2039744495|A=2620713517|]: peer_server_node pmi user=00000000,84353c40 mismatch
Tue Dec 26 00:26:54 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 00:30:09 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 00:30:22 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 00:31:43 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 00:34:50 2023 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 00:08:22:f0:49:96
Tue Dec 26 00:34:50 2023 daemon.info hostapd: phy1-ap0: STA 00:08:22:f0:49:96 IEEE 802.11: disassociated due to inactivity
Tue Dec 26 00:34:50 2023 user.info usteer: station 00:08:22:f0:49:96 disconnected from node hostapd.phy1-ap0
Tue Dec 26 00:34:51 2023 daemon.info hostapd: phy1-ap0: STA 00:08:22:f0:49:96 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Dec 26 00:37:21 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 00:44:55 2023 kern.warn kernel: [100184.727579] warning: (PPI)[PRE](natcap_peer_pre_in_hook:2946)[8.218.31.137:47503->192.168.0.45:49821|ID:2D3F,IL:80|...A..S.|S=1903709206|A=834431481|]: peer_server_node pmi user=84344a80,843448c0 mismatch
Tue Dec 26 00:46:05 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 00:47:33 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 00:50:47 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED c6:29:a5:e4:7d:9a
Tue Dec 26 00:50:47 2023 user.info usteer: station c6:29:a5:e4:7d:9a disconnected from node hostapd.phy0-ap0
Tue Dec 26 00:50:48 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 00:54:40 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 00:59:57 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 01:07:53 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 01:13:35 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 01:17:32 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 01:26:00 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 01:28:01 2023 daemon.info hostapd: phy1-ap0: STA 00:08:22:f0:49:96 IEEE 802.11: authenticated
Tue Dec 26 01:28:01 2023 daemon.info hostapd: phy1-ap0: STA 00:08:22:f0:49:96 IEEE 802.11: associated (aid 1)
Tue Dec 26 01:28:01 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 00:08:22:f0:49:96 auth_alg=open
Tue Dec 26 01:28:01 2023 daemon.info hostapd: phy1-ap0: STA 00:08:22:f0:49:96 RADIUS: starting accounting session 0688EC158FA5C8DF
Tue Dec 26 01:28:01 2023 daemon.info hostapd: phy1-ap0: STA 00:08:22:f0:49:96 WPA: pairwise key handshake completed (RSN)

musakilic, 2023-12-26:
Tue Dec 26 03:44:42 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 03:46:52 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 03:56:36 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 04:01:34 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 04:11:09 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 04:20:56 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 04:24:59 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 04:30:39 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 04:38:05 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 04:45:26 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 04:50:11 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 04:54:16 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 05:00:19 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 05:09:52 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 05:16:47 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 05:21:20 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 05:27:44 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 05:32:55 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 05:42:40 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 05:51:52 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 05:57:42 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 06:03:05 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 06:11:28 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 06:20:42 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 06:25:21 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 06:34:10 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
Tue Dec 26 06:43:19 2023 kern.warn kernel: [108452.629104] warning: (PPI)[PRE](natcap_peer_pre_in_hook

ptpt52, 2023-12-26:
If you need a stable version, please use the older release at https://github.com/x-wrt/build-release/releases/tag/v23.10_b202312150123. This older version still utilizes the 5.10 kernel. The new version, which uses the kernel 5.15, frequently experiences crashes, and a solution for this issue is currently unavailable.

DragonBluep, 2023-12-26:
> Tue Dec 26 03:44:42 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!
> Tue Dec 26 03:46:52 2023 daemon.warn odhcpd[2583]: No default route present, overriding ra_lifetime!

These logs are harmless warnings. Disabling ipv6 can silence them.

@ptpt52 I found the LNA pinctrl is missing on mi router 3. Perhaps configuring them correctly would be helpful.
```
&wmac {
	pinctrl-names = "default", "pa_gpio";
	pinctrl-0 = <&pa_pins>;
	pinctrl-1 = <&pa_gpio_pins>;
};
```

musakilic, 2023-12-26:
use stabil version

[image]

X-WRT 23.10_b202312150123 Lunar / LuCI (no branch) branch git-23.347.82030-a4472af
--

musakilic, 2023-12-26:
alo add log 
[image]

DragonBluep, 2023-12-27:
And this pin group can be removed now.
https://github.com/x-wrt/x-wrt/blob/baef3802e55e7c7fa6b766389e0fcc45dd418c4e/target/linux/ramips/dts/mt7620a_xiaomi_miwifi-r3.dts#L185-L188

ptpt52, 2023-12-27:
@DragonBluep 
great! I will try it out.

musakilic, 2023-12-27:
how can i do dts

musakilic, 2023-12-27:
X-WRT 23.10_b202312150123 Lunar again installl

ptpt52, 2023-12-27:
The latest version, 23.10_b202312270857, incorporates the mentioned modifications. Feel free to give it a shot, though I'm skeptical that these changes will address the recurring crashes.
https://github.com/x-wrt/build-release/releases/tag/v23.10_b202312270857

musakilic, 2023-12-27:
Yes install but all apps remove also install again
How can apps image

musakilic, 2023-12-28:
flash last version
[image]

musakilic, 2023-12-28:
[image]

ptpt52, 2023-12-29:
@DragonBluep  tested and nothing help with the crash issue. the same crash log happens on mt7620 (miwifi-r3 or mini)

ptpt52, 2023-12-29:
the log: https://github.com/x-wrt/x-wrt/issues/354

DragonBluep, 2023-12-30:
@ptpt52 I cannot reproduce this issue on my Phicomm K2 (custom build image with natflow module, based on official OpenWrt). Perhaps some of the packages/modifications in x-wrt caused the conflicts? BTW, I have an extra patch on my custom build image.
https://patchwork-proxy.ozlabs.org/project/openwrt/patch/TYAP286MB0315ABFBF338B0CBAC69E110BC28A@TYAP286MB0315.JPNP286.PROD.OUTLOOK.COM/

musakilic, 2023-12-31:
> @ptpt52 I cannot reproduce this issue on my Phicomm K2 (custom build image with natflow module, based on official OpenWrt). Perhaps some of the packages/modifications in x-wrt caused the conflicts? [...]

how can i do pls help me.. but ever 24 hours my internet goes down remove cable and insert 
pls help me

ptpt52, 2024-01-01:
@musakilic  use old version x-wrt
https://github.com/x-wrt/build-release/releases/tag/v23.10_b202312150123

musakilic, 2024-01-01:
same 24 hour later internet goes down

nathanielvarona, 2024-02-19:
Thank you @ptpt52 

After a few weeks of observations of my router `Xiaomi Mi Router R3 (MediaTek MT7620A ver:2 eco:6)`
It solves the issues of intermittent restart by downgrading from the latest releases to later stable  [X-WRT v23.10_b202312150123](https://github.com/x-wrt/build-release/releases/tag/v23.10_b202312150123) release.

I hope for future releases to solve the issues. I keep checking it and watching future releases.

Thanks Again!