x-wrt/x-wrt #363: Unknown UDP Connection

TobiKi400, 2024-01-05:
[image]

I am noticing strange connection to a Server from the Router.
Router is running as AP over eth0.

[image]

The Connection cannot be seen in netstat!
It is blocked by my Firewall because the random Higports are not supported in my Network,
As you can see from my Firewall log. The Source and Destination Ports Change from time to time. 
[image]
The Connection can be seen in tcpdmup on if eth0.
[image]
Running ps w | grep '8.218.31.137' from time to time gives following result
[image]
This seems to be Ping to the same IP Adress originating from Busybox.
I cannot track easy because the process disappears. 
Not sure if this is the same Request as in the Firewall Log. No Cron Jobs found triggering this ping.
How to stop it ? 
What is this UDP Connection ?

ptpt52, 2024-01-05:
it is connection by natcap, is part of remote web luci.
you need to stop and disable all natcapd to stop them.
```
/etc/init.d/natcapd disable
reboot
```

TobiKi400, 2024-01-09:
Can confirm requests are not there anymore after disdabling natcapd.
Closing Issue. 
Thanks!

closed 2024-01-09 by TobiKi400