x-wrt/x-wrt #368: 由于删了桥接接口，无法登陆路由器管理界面，成了一个交换机，请教有什么办法挽救？

andrewlxf, 2024-02-04:
我的小米路由器3刷了对应的固件，由于误删了桥接接口，现在无法登陆路由器管理界面。按reset键无法恢复出产设置，也无法重新刷系统。有没有大神，用简单的办法，恢复过来。

andrewlxf, 2024-02-04:
> 我的小米路由器3刷了对应的固件，由于误删了桥接接口，现在无法登陆路由器管理界面。按reset键无法恢复出产设置，也无法重新刷系统。有没有大神，用简单的办法，恢复过来。

我写错了，应该我是停了桥接端口。

ptpt52, 2024-02-04:
按reset当然可以，
开机运行状态下（可以开机跑5分钟先），按下reset保持摁压 6秒放开（要一气呵成）就重置恢复默认配置
@andrewlxf

andrewlxf, 2024-02-05:
> 按reset当然可以， 开机运行状态下（可以开机跑5分钟先），按下reset保持摁压 6秒放开（要一气呵成）就重置恢复默认配置 @andrewlxf

 原来是中国人哪，那就好办了。我照你的方法是试了，黄灯转蓝灯。还是没有恢复过来，无法登陆192.168.5.1，但是可以上网（变成交换机，由主路由分配的IP），说明没有恢复出厂设置（如果恢复了，旁路由会给我一个192.168.5..0/24的IP，我都连再LAN的网口上，原先我是关闭旁路由的DHCP的）。

现在这种情况还能刷机吗，有没有具体的教程。谢谢啦，大神！

andrewlxf, 2024-02-05:
> > 按reset当然可以， 开机运行状态下（可以开机跑5分钟先），按下reset保持摁压 6秒放开（要一气呵成）就重置恢复默认配置 @andrewlxf
> 
> 原来是中国人哪，那就好办了。我照你的方法是试了，黄灯转蓝灯。还是没有恢复过来，无法登陆192.168.5.1，但是可以上网（变成交换机，由主路由分配的IP），说明没有恢复出厂设置（如果恢复了，旁路由会给我一个192.168.5..0/24的IP，我都连再LAN的网口上，原先我是关闭旁路由的DHCP的）。
> 
> 现在这种情况还能刷机吗，有没有具体的教程。谢谢啦，大神！

 我刷机的时候遇到点小问题。我按照教程把一下两个文件放在临时文件夹里
x-wrt-24.01-b202401270830-ramips-mt7620-xiaomi_miwifi-r3-squashfs-kernel1.bin
x-wrt-24.01-b202401270830-ramips-mt7620-xiaomi_miwifi-r3-squashfs-rootfs0.bin

我只执行第一个文件的mtd命令后，系统就自动重启了。重复操作，执行不了第二个文件。但是能进系统，我就不纠结了，用有upgtate的文件升级了一下，想总没问题了。不知是不是这个原因。

这种情况，还能重新刷机吗？请大神详解！