x-wrt/x-wrt #374: TTL刷机遇到一些问题。

andrewlxf, 2024-02-23:
我误把桥接端口删，尝试用ttl刷机，但是遇到一些问题。

问题一如下所示：
 ##### The CPU freq = 580 MHZ #### 
 estimate memory 8 Mbytes

Poose the operation: 
   1: Load system code to SDRAM via TFTP. Load system code then write to Flash via TFTP. 
   3: Boot sys via Flash (default).
   4: Entr boot command line interface.
ad Boot Loader code then write to Flash via TFTP. 
   3: Syt system code via Flash.
它自动选择3，我无法选择其他选项。

问题二如下所示：
 X-WRT 24.01_b202401270830, Mantic
 -----------------------------------------------------
================= FAILSAFE MODE active ================
special commands:
* firstboot          reset settings to factory defaults
* mount_root     mount root-partition with config files

after mount_root:
* passwd                         change root's password
* /etc/config               directory with config files

for more help see:
https://openwrt.org/docs/guide-user/troubleshooting/
- failsafe_and_factory_reset
- root_password_reset
=======================================================

root@(none):
下面无法输入命令，我还原出厂设置也好的，也算救砖了。

请大神指示！

ptpt52, 2024-02-23:
一般等系统启动完毕，串口 root / admin 登录，执行 `firstboot -y -r`

andrewlxf, 2024-02-23:
我的问题就是不能打命令，不知道为什么

ptpt52, 2024-02-23:
整个串口boot log发出来看看

andrewlxf, 2024-02-23:
U-Boot 1.1.3 (Sep 13 2000:44)

Board: Ralink APSoC DRAM:  128 MB
Pmemory test. Memory size= 128 MB...OK!
relocate_code Pointer at: 87fb8000
enable ephy clock...done. 9 = 5
SSC disabled.
!!! nand page size = 204len=4
..============================================  UBoot Version: 4.2.S.1
------------------------------------ 
ASIC 7620_MP (Port5<->None)_CONF_FROM: Auto-detection 
DRAM_TYPE: DDR2 
DRnent: 1024 Mbits
DRAM bus: 16 bit
Total memoMBytes
Flash component: NAND Flash
Date:Sep  Time:09:00:44
==================================== 
icache: sets:512, ways:4, linesz:32 ,536
dcache: sets:256, ways:4, linesz:32 ,tot 

 ##### The CPU freq = 580 MHZ #### 
 estmory size =128 Mbytes

Please choose the operation: 
   system code to SDRAM via TFTP. 
   2: Load sde then write to Flash via TFTP. 
   3: Boot ode via Flash (default).
   4: Entr boot comm interface.
   9: Load Boot Loader code then  Flash via TFTP. 

   
3: System Boot systeia Flash.
Booting System 2
..ranand_erase_write: offs:40000, count:200.Done!## Booting image at bc600000 ...
   Image Name:   MIPS X-WRT Linux-5.15.147
   pe:   MIPS Linux Kernel Image (lzma compressedta Size:    2462435 Bytes =  2.3 MB
   Load A80000000
   Entry Point:  80000000
......................   Verifying Checksum ... OK
   Uncompressil Image ... OK
commandline u factory_mode=0 mem=128m
No initrd
## Transfontrol to Linux (at address 80000000) ...
## inux memsize in MB, 128

Starting kernel ...[    ] Linux version 5.15.147 (ptpt52@SC-GAME) (mipwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r25123+ff88f) 12.3.0, GNU ld (GNU Binutils) 2.40.0) an 27 00:31:50 2024
[    0.000000] Board has    0.000000] Analog PMU set to hw control
[  00] Digital PMU set to hw control
[    0.0000Type: MediaTek MT7620A ver:2 eco:6
[    0.000ntk: bootconsole [early0] enabled
[    0.0000 revision is: 00019650 (MIPS 24KEc)
[    0.000000] Mhine is Xiaomi Mi Router R3
[    0.0nitrd not found or empty - disabling initrd
[0000] Primary instruction cache 64kB, VIPT, 4-esize 32 bytes.
 0] Zone ranges:rimary data cB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000]   Normal   [me000000000000-0x0000000007ffffff]
[    0.00000le zone start for each node
[    0.000000] Eary node ranges
[    0.000000]   node   0: [me000000000000-0x0000000007ffffff]
[    0.00000em setup node 0 [mem 0x0000000000000000-0x0000ffff]
[    0.00ݠBuilt 1 zonelists, mobility grouping on.  Total 2512
[    0.000000] Kernel command line: cons0,115200 rootfstype=squashfs,jffs2
.000000] Dentry cache hash table entries: 1638: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 81r: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=0
[    0.000000] Readback ErrCtl register=0006   0.000000] mem auto-init: stack:off, heap al heap free:off
[    0.000000] Memory: 120542K available (5912K kernel code, 588K rwdata, ˠrodata, 1240K init, 217K bss, 10532K reserved, 0ˠreserved)
[    0.000000] SLUB: HWalOrder=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_I
[... 293 lines cut ...]
[   61.110357] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=eth0.1(tso=1,1,hw_csum=1,1), add ingress hook
[   61.172800] br-lan: port 1(eth0.1) entered blocking state
[   61.178355] br-lan: port 1(eth0.1) entered disabled state
[   61.184355] device eth0.1 entered promiscuous mode
[   61.389146] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=eth0.2(tso=1,1,hw_csum=1,1), add ingress hook
[   82.623716] warning: (zone_netdev_event:379)dev=phy1-ap0 set zone=127 type=0
[   82.631030] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=phy1-ap0(tso=0,0,hw_csum=0,0), add ingress hook
[   82.647255] br-lan: port 2(phy1-ap0) entered blocking state
[   82.653070] br-lan: port 2(phy1-ap0) entered disabled state
[   82.659158] device phy1-ap0 entered promiscuous mode
[   84.912204] warning: (zone_netdev_event:379)dev=phy0-ap0 set zone=127 type=0
[   84.919506] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=phy0-ap0(tso=0,0,hw_csum=0,0), add ingress hook
[   84.935539] br-lan: port 3(phy0-ap0) entered blocking state
[   84.941365] br-lan: port 3(phy0-ap0) entered disabled state
[   84.947431] device phy0-ap0 entered promiscuous mode
[   91.794978] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
[   91.802042] br-lan: port 2(phy1-ap0) entered blocking state
[   91.807758] br-lan: port 2(phy1-ap0) entered forwarding state
[   91.932434] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[  100.926962] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
[  100.934058] br-lan: port 3(phy0-ap0) entered blocking state
[  100.939818] br-lan: port 3(phy0-ap0) entered forwarding state

Error reading data on serial port:
  Access is denied.

andrewlxf, 2024-02-23:
发给你了，加个QQ聊吧，一句两句说不清楚

ptpt52, 2024-02-23:
你看，现在启动到这个位置，摁Entry，输入 root 密码 admin 登录，执行 `firstboot -y -r`

ptpt52, 2024-02-23:
另外
在U盘根目录下创建一个空白文件，并将其重命名为 x-wrt-reset.txt。
将U盘插入路由器。
开机等待自动重启一次，恢复默认配置成功。

andrewlxf, 2024-02-23:
嗯，启动完，按回车是可以打字了，但是打完命令，指示图标回到第一格，并没有执行。不知道为什么。方便加个qq吗？我略懂些技术，只是对这个不熟，想搞懂它。

andrewlxf, 2024-02-23:
我想在这里选择，但是系统太快了，或者自己选择了
 e CPU freq = 580 MHZ #### 
 estimate memory 8 Mbytes

Pleas the operation: 
   1: Load system code to S TFTP. 
   2: Load system code then write to a TFTP. 
   4: Entr boot command line interface.
   Boot Loader code then write to Flash via TFTP 
3: System Boot system code via Flash.
..ranand_erase_write0000, count:20000
.Done!
done

这个有没有办法？

ptpt52, 2024-02-23:
哦 米3 串口是锁的应该无法输入吧。
你还是用usb 重置配置的方法。

ptpt52, 2024-02-23:
qq 你可以进群 131750901