x-wrt/x-wrt #375: [Kernel crash] x-wrt running on PVE(VM), Kernel BUG at skb_pull_rcsum+0x94/0xa0

228peace, 2024-02-25:
crash version: x-wrt-docker-24.01-b202402242121-x86-64-generic-ext4-combined-efi.img

PVE host, cpu: intel n5000 
KVM (x-wrt) cpu: 2 core (host), ram: 1GB, net: virtIO,  

我還需要提供些什麼資料方便追查嗎？

---- log below 

[   78.417553] warning: catch unregister event for dev=pppoe-wan
[   78.428986] {natflow}:natflow_netdev_event(): catch NETDEV_UNREGISTER event for dev=pppoe-wan
[   79.650553] warning: (zone_netdev_event:379)dev=pppoe-wan set zone=127 type=0
[   79.674754] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=pppoe-wan : set flags IFF_PPPOE
[   79.713337] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=pppoe-wan(tso=0,0,hw_csum=0,0), add ingress hook
[   80.068480] warning: catch unregister event for dev=pppoe-wan
[   80.081994] {natflow}:natflow_netdev_event(): catch NETDEV_UNREGISTER event for dev=pppoe-wan
[   81.736149] warning: (zone_netdev_event:379)dev=pppoe-wan set zone=127 type=0
[   81.752943] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=pppoe-wan : set flags IFF_PPPOE
[   81.776569] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=pppoe-wan(tso=0,0,hw_csum=0,0), add ingress hook
[   81.923098] warning: catch unregister event for dev=pppoe-wan
[   81.929426] {natflow}:natflow_netdev_event(): catch NETDEV_UNREGISTER event for dev=pppoe-wan
[   82.474143] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[   82.481158] NFSD: Using legacy client tracking operations.
[   82.486390] NFSD: starting 10-second grace period (net f0000000)
grep: /tmp/hosts/fakemesh: No such file or directory
grep: /tmp/hosts/fakemesh.idx: No such file or directory
[   89.548194] warning: (zone_netdev_event:379)dev=pppoe-wan set zone=127 type=0
[   89.554505] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=pppoe-wan : set flags IFF_PPPOE
[   89.562561] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=pppoe-wan(tso=0,0,hw_csum=0,0), add ingress hook
Command failed: Not found
[   97.948429] ------------[ cut here ]------------
[   97.953117] Kernel BUG at skb_pull_rcsum+0x94/0xa0 [verbose debug info unavailable]
[   97.959871] invalid opcode: 0000 [#1] SMP PTI
[   97.964183] CPU: 1 PID: 16403 Comm: iptables Not tainted 6.1.78 #0
[   97.970011] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 3.20230228-4 06/06/2023
[   97.984707] RIP: 0010:skb_pull_rcsum+0x94/0xa0
[   97.989744] Code: eb d6 8b 93 88 00 00 00 f7 da e8 f7 47 1f 00 f7 d8 89 83 88 00 00 00 48 8b 83 c8 00 00 00 48 8b 5d f8 c9 c3 cc cc cc cc 0f 0b <0f> 0b 66 2e 0f 1f 84 00 00 00 00 00 8b 87 f0 00 00 00 8b 96 d0 00
[   98.004240] RSP: 0000:ffffc900000ccc58 EFLAGS: 00010293
[   98.009148] RAX: 00000000000005c4 RBX: ffff888002c5f9c0 RCX: 0000000000000000
[   98.015102] RDX: ffff88801acafc00 RSI: 0000000000000008 RDI: ffff88801acafc4e
[   98.022611] RBP: ffffc900000ccc60 R08: 00000000c1bc2201 R09: 0000000000000100
[   98.028450] R10: ffffc900000ccda8 R11: 80bb917586c5c92c R12: 0000000000000001
[   98.034533] R13: 0000000000000005 R14: ffffc900000cce10 R15: 0000000000000000
[   98.040708] FS:  00007f69a93cab48(0000) GS:ffff88803e500000(0000) knlGS:0000000000000000
[   98.047147] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   98.052466] CR2: 00007f69a936a190 CR3: 0000000006a22000 CR4: 0000000000350ee0
[   98.058669] Call Trace:
[   98.061813]  <IRQ>
[   98.064687]  ? show_regs.part.0+0x1e/0x24
[   98.068981]  ? __die+0x55/0x99
[   98.072126]  ? die+0x2a/0x50
[   98.075365]  ? do_trap+0x108/0x110
[   98.078842]  ? do_error_trap+0x6c/0x90
[   98.082592]  ? skb_pull_rcsum+0x94/0xa0
[   98.086387]  ? exc_invalid_op+0x4f/0x70
[   98.090220]  ? skb_pull_rcsum+0x94/0xa0
[   98.093940]  ? asm_exc_invalid_op+0x1b/0x20
[   98.099938]  ? skb_pull_rcsum+0x94/0xa0
[   98.103663]  natflow_path_pre_ct_in_hook+0x863/0x42e0 [natflow]
[   98.115008]  ? __skb_get_hash+0x5a/0x1e0
[   98.118483]  ? __skb_get_hash+0x94/0x1e0
[   98.123851]  ? _raw_spin_unlock_irqrestore+0x9/0x20
[   98.128391]  ? enqueue_to_backlog+0x1a9/0x2a0
[   98.140555]  nf_hook_slow+0x3c/0xc0
[   98.159005]  ? nf_hook_slow+0x3c/0xc0
[   98.165112]  __netif_receive_skb_core.constprop.0+0xa20/0xdb0
[   98.174904]  __netif_receive_skb_one_core+0x2a/0x70
[   98.183535]  process_backlog+0x89/0x1a0
[   98.187884]  __napi_poll+0x29/0x110
[   98.191092]  net_rx_action+0x130/0x230
[... 26 lines cut ...]
[   98.482690] R10: 00007f69a93ca940 R11: 00007f69a93c5fba R12: 000000008683fbf8
[   98.497241] R13: 00007f69a93ca940 R14: 0000000000000000 R15: 0000000000000000
[   98.511968]  </TASK>
[   98.517131] Modules linked in: ksmbd ath9k ath9k_common rt2800usb rt2800lib qcserial pppoe ppp_async option mt76x0u mt76x0_common l2tp_ppp iwlmvm iwldvm cdc_mbim brcmsmac brcmfmac ath9k_hw ath6kl_usb ath6kl_core ath10k_pci ath10k_core ath wireguard vhost_net vhost usb_wwan sierra_net sierra rt2x00usb rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 nft_fib_inet mt7915e mt76x02_usb mt76x02_lib mt7663u mt7663_usb_sdio_common mt7615_common mt7601u mt76_usb mt76_connac_lib mt76 mmc_spi mlx5_core mac80211 libchacha20poly1305 iwlwifi ipt_REJECT huawei_cdc_ncm curve25519_x86_64 cp210x chacha_x86_64 cfg80211 cdc_ncm cdc_ether zstd xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_quota xt_policy xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_HL
[   98.517284]  xt_DSCP xt_DNETMAP xt_DHCPMAC xt_CLASSIFY xfrm_interface vhost_iotlb usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sfp sch_cake rtl8150 r8169 r8152 poly1305_x86_64 pcnet32 of_mmc_spi nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323
[   98.699844]  nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount mlxfw mlx4_en mlx4_core mdio_i2c macvlan lzo_rle lzo libcurve25519_generic libchacha ipvlan iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables igc i40e forcedeth crc8 crc7 crc_itu_t crc_ccitt cordic compat_xtables compat cdc_wdm brcmutil br_netfilter bnx2x bnx2 atl2 atl1e atl1c atl1 alx natflow natcap ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact configs evdev i2c_piix4 i2c_i801 i2c_smbus i2c_dev ledtrig_usbport cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink jool_siit jool
[   98.865328]  jool_common dwmac_intel ip6table_nat nf_nat nf_conntrack nf_defrag_ipv4 ip6t_NPT ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 cifs_arc4 asn1_decoder dwmac_generic stmmac_platform stmmac nfsd nfs msdos bonding ip6_gre ip_gre gre ixgbe amd_xgbe mdio_devres ifb dummy nat46 nf_defrag_ipv6 l2tp_ip6 l2tp_ip l2tp_eth ip6_vti ip_vti sit mdio l2tp_netlink l2tp_core ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel tunnel6 tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag rpcsec_gss_krb5 auth_rpcgss oid_registry dm_raid veth tun ovpn_dco_v2 xfrm_user xfrm_ipcomp af_key xfrm_algo virtiofs fuse lockd sunrpc grace autofs4 dns_resolver multipath bcma raid456 async_raid6_recov async_pq async_xor async_memcpy async_tx raid10 raid1 raid0 linear md_mod nls_utf8 pcs_xpcs eeprom_93cx6 vxlan udp_tunnel ip6_udp_tunnel ena crypto_user algif_skcipher algi [... cut ...]
[   99.494131] R10: ffffc900000ccda8 R11: 80bb917586c5c92c R12: 0000000000000001
[   99.507332] R13: 0000000000000005 R14: ffffc900000cce10 R15: 0000000000000000
[   99.521496] FS:  00007f69a93cab48(0000) GS:ffff88803e500000(0000) knlGS:0000000000000000
[   99.537613] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   99.548222] CR2: 00007f69a936a166 CR3: 0000000006a22000 CR4: 0000000000350ee0
[   99.561698] Kernel panic - not syncing: Fatal exception in interrupt
[   99.574749] Kernel Offset: disabled
[   99.582553] Rebooting in 3 seconds..

----

ptpt52, 2024-02-26:
这个日志产生的日期是什么时间，这个不一定是当前版本产生的宕机日志。

ptpt52, 2024-02-26:
23年12月14之后的版本 已经修复这个bug了的。

228peace, 2024-02-26:
> 这个日志产生的日期是什么时间，这个不一定是当前版本产生的宕机日志。

這是我從console即時複製下來的。

228peace, 2024-02-26:
> 23年12月14之后的版本 已经修复这个bug了的。

我能提供什麼資訊嗎？
這些log 都是我從console 即時複製下來的

ptpt52, 2024-02-26:
内核版本 `uname -r`
查看natflow版本 `opkg list | grep natflow`

228peace, 2024-02-26:
> 内核版本 `uname -r` 查看natflow版本 `opkg list | grep natflow`

 -----------------------------------------------------
 X-WRT-docker 24.01_b202402242121, Mantic
 -----------------------------------------------------
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# uname -r
6.1.78
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# opkg list |grep natflow
kmod-natflow - 6.1.78+20240210-1
luci-app-natflow-users - 1.0.1-1
natflow-auth - 20240210
natflow-boot - 20240210
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~#

228peace, 2024-02-26:
目前 WAN 拔掉網線的話不會觸發 crash reboot
我這邊 WAN 是 PPPoE 的設定

228peace, 2024-02-26:
我手上有一台運行較舊版本的沒有遇到這個 crash 狀況 
版本如下

---

root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# cat /etc/os-release 
NAME="X-WRT-docker"
VERSION="23.10_b202311242121"
ID="x-wrt-docker"
ID_LIKE="lede openwrt"
PRETTY_NAME="X-WRT-docker 23.10_b202311242121"
VERSION_ID="23.10_b202311242121"
HOME_URL="https://x-wrt.com/"
BUG_URL="https://x-wrt.com/"
SUPPORT_URL="https://x-wrt.com/"
BUILD_ID="r24663+1-ac1ef1ccf09"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="X-WRT-TEAM"
OPENWRT_DEVICE_MANUFACTURER_URL="https://x-wrt.com/"
OPENWRT_DEVICE_PRODUCT="X-Router"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="X-WRT-docker 23.10_b202311242121 Lunar"
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# uname -r
5.15.139
root@X-WRT-DOCKER:~# opkg list |grep natflow
kmod-natflow - 5.15.139+20231125-1
luci-app-natflow-users - 1.0.1-1
natflow-auth - 20231125
natflow-boot - 20231125
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~#

228peace, 2024-02-26:
這一版也會有 crash 
"X-WRT-docker 23.10_b202312111425 Lunar"

---

root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# opkg list |grep natflow
kmod-natflow - 6.1.65+20231204-1
luci-app-natflow-users - 1.0.1-1
natflow-auth - 20231204
natflow-boot - 20231204
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# uname -r
6.1.65
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# cat /etc/os-release 
NAME="X-WRT-docker"
VERSION="23.10_b202312111425"
ID="x-wrt-docker"
ID_LIKE="lede openwrt"
PRETTY_NAME="X-WRT-docker 23.10_b202312111425"
VERSION_ID="23.10_b202312111425"
HOME_URL="https://x-wrt.com/"
BUG_URL="https://x-wrt.com/"
SUPPORT_URL="https://x-wrt.com/"
BUILD_ID="r0-f3abfbc"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="X-WRT-TEAM"
OPENWRT_DEVICE_MANUFACTURER_URL="https://x-wrt.com/"
OPENWRT_DEVICE_PRODUCT="X-Router"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="X-WRT-docker 23.10_b202312111425 Lunar"
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~# 
root@X-WRT-DOCKER:~#

ptpt52, 2024-02-27:
我预期最新版本 比如 b202402242121 是不存在问题的，如果出现问题，可以给出完整的console日志，用code标记把日志发到这里
```
code 标记 : "```日志```"
```

228peace, 2024-02-27:
code 标记 : "```日志```"

`[    0.835684] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.840992] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.845620] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.851766] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.855614] x86/fpu: xstate_offset[3]:  576, xstate_sizes[3]:   64
[    0.861224] x86/fpu: xstate_offset[4]:  640, xstate_sizes[4]:   64
[    0.865611] x86/fpu: Enabled xstate features 0x1b, context size is 704 bytes, using 'compacted' format.
[    0.919679] Freeing SMP alternatives memory: 48K
[    0.923063] pid_max: default: 32768 minimum: 301
[    0.928549] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.935645] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.945776] smpboot: CPU0: Intel(R) Pentium(R) Silver N5000 CPU @ 1.10GHz (family: 0x6, model: 0x7a, stepping: 0x1)
[    0.956217] cblist_init_generic: Setting adjustable number of callback queues.
[    0.962005] cblist_init_generic: Setting shift to 1 and lim to 1.
[    0.965742] Performance Events: Goldmont plus events, full-width counters, Intel PMU driver.
[    0.972471] ... version:                2
[    0.975476] ... bit width:              48
[    0.975628] ... generic registers:      4
[    0.978611] ... value mask:             0000ffffffffffff
[    0.985627] ... max period:             00007fffffffffff
[    0.989953] ... fixed-purpose events:   3
[    0.995629] ... event mask:             000000070000000f
[    1.001733] signal: max sigframe size: 1616
[    1.005765] rcu: Hierarchical SRCU implementation.
[    1.013231] rcu:     Max phase no-delay instances is 1000.
[    1.016765] smp: Bringing up secondary CPUs ...
[    1.026455] x86: Booting SMP configuration:
[    1.034311] .... node  #0, CPUs:      #1
[    1.036346] smp: Brought up 1 node, 2 CPUs
[    1.045647] smpboot: Max logical packages: 1
[    1.052999] smpboot: Total of 2 processors activated (4377.60 BogoMIPS)
[    1.062970] devtmpfs: initialized
[    1.070853] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    1.075650] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    1.080481] pinctrl core: initialized pinctrl subsystem
[    1.085947] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    1.091829] thermal_sys: Registered thermal governor 'step_wise'
[    1.091834] thermal_sys: Registered thermal governor 'user_space'
[    1.095851] cpuidle: using governor ladder
[    1.103468] acpiphp: ACPI Hot Plug PCI Controller Driver version: 0.5
[    1.106036] PCI: Using configuration type 1 for base access
[    1.138271] HugeTLB: registered 2.00 MiB page size, pre-allocated 0 pages
[    1.142846] HugeTLB: 0 KiB vmemmap can be freed for a 2.00 MiB page
[    1.146691] cryptd: max_cpu_qlen set to 1000
[    1.151363] ACPI: Added _OSI(Module Device)
[    1.154489] ACPI: Added _OSI(Processor Device)
[    1.155710] ACPI: Added _OSI(3.0 _SCP Extensions)
[    1.159172] ACPI: Added _OSI(Processor Aggregator Device)
[    1.236781] ACPI: 2 ACPI AML tables successfully acquired and loaded
[    1.250490] ACPI: Interpreter enabled
[    1.252990] ACPI: PM: (supports S0 S5)
[    1.255419] ACPI: Using IOAPIC for interrupt routing
[    1.255868] PCI: Using host bridge windows from ACPI; if necessary, use "pci=nocrs" and report a bug
[    1.261806] PCI: Ignoring E820 reservations for host bridge windows
[    1.268522] ACPI: Enabled 3 GPEs in block 00 to 3F
[    1.365160] ACPI: PCI Root Bridge [PCI0] (domain 0000 [bus 00-ff])
[    1.365639] acpi PNP0A08:00: _OSC: OS supports [ASPM ClockPM Segments MSI HPX-Type3]
[    1.369783] acpi PNP0A08:00: _OSC: not requesting OS control; OS requires [ExtendedConfig ASPM ClockPM MSI]
[    1.378081] PCI host bridge to bus 0000:00
[    1.380319] pci_bus 0000:00: root bus resource [io  0x0000-0x0cf7 window]
[    1.384217] pci_bus 0000:00: root bus resource [io  0x0d00-0xffff window]
[    1.385627] pci_bus 0000:00: root bus resource [mem 0x000a0000-0x000bffff window]
[    1.389816] pci_bus 0000:00: root bus resource [mem 0x40000000-0xafffffff window]
[    1.394116] pci_bus 0000:00: root bus resource [mem 0xc0000000-0xfebfffff window]
[    1.395621] pci_bus 0000:00: root bus resource [mem 0x7000000000-0x77ffffffff window]
[    1.400305] pci_bus 0000:00: root bus resource [bus 00-ff]
[    1.403759] pci 0000:00:00.0: [8086:29c0] type 00 class 0x060000
[    1.407624] pci 0000:00:01.0: [1234:1111] type 00 class 0x030000
[... 826 lines cut ...]
[   38.448960] RDX: 0000000000000008 RSI: 0000000000000080 RDI: 0000000000000088
[   38.461696] RBP: 0000000000000007 R08: 00007f78bb85c1f0 R09: 0000000000000003
[   38.471874] R10: 00007f78bb890dec R11: 0000000000000000 R12: 0000000000000007
[   38.477679] R13: 00000000000006e1 R14: 0000000000000008 R15: 000000000000003d
[   38.483741]  </TASK>
[   38.486062] Modules linked in: ath9k ath9k_common rt2800usb rt2800lib qcserial pppoe ppp_async option mt76x0u mt76x0_common l2tp_ppp iwlmvm iwldvm cdc_mbim brcmsmac brcmfmac ath9k_hw ath6kl_usb ath6kl_core ath10k_pci ath10k_core ath wireguard vhost_net vhost usb_wwan sierra_net sierra rt2x00usb rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 nft_fib_inet mt7915e mt76x02_usb mt76x02_lib mt7663u mt7663_usb_sdio_common mt7615_common mt7601u mt76_usb mt76_connac_lib mt76 mmc_spi mlx5_core mac80211 libchacha20poly1305 iwlw [... cut ...]
[   39.130563] R10: ffffc90004367cd8 R11: 0000000000000001 R12: 0000000000000001
[   39.136373] R13: 0000000000000005 R14: ffffc90004367d40 R15: 0000000000000000
[   39.142284] FS:  00007f78bb9fcb48(0000) GS:ffff88803e400000(0000) knlGS:0000000000000000
[   39.150144] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   39.155296] CR2: 00007f78bb83a1d0 CR3: 0000000003438000 CR4: 0000000000350ef0
[   39.160947] Kernel panic - not syncing: Fatal exception in interrupt
[   39.166538] Kernel Offset: disabled
[   39.170649] Rebooting in 3 seconds..`

ptpt52, 2024-02-27:
how to create your virtual machine?

228peace, 2024-02-27:
> how to create your virtual machine?

[image]

I create VM on PVE, and trans X-WRT.img into X-WRT.qcow2 then load on VM as a virt-driver, boot on and set PPPoE.

img轉qcow2格式 匯入

$ qemu-img convert -O qcow2 X-WRT.img X-WRT.qcow2 

Now, import the QCOW2 image into the Proxmox server using command:

$ qm importdisk 102 X-WRT.qcow2  local-lvm

228peace, 2024-03-14:
這img在PVE做VM是有問題，我就不研究下去了，我後來改用LXC虛擬化了，LXC底下使用兩週過去完全沒問題，但就是抽取rootfs有點麻煩，官網下載區能在之後的編譯釋出rootfs的檔案嘛？

ptpt52, 2024-03-14:
you need `rootfs.img.gz`?

228peace, 2024-03-14:
> you need `rootfs.img.gz`?
希望釋出的檔案是 “rootfs.**tar.gz**”

我看官方的打包名稱是 “rootfs.**tar.gz**”，而不是 ”＊-rootfs.img.gz“
LXC執行openwrt的話是這個檔案就可以跑了，非常的輕巧，資源佔用也非常的少。

LXC教程我看也很少人撰寫，找到下面這位老兄的文，還不錯。
https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=8252802

ptpt52, 2025-02-09:
> 用最新的版本重新測試
> 確認是WAN設定 PPPoE 加上VM的網卡使用net: virtIO 網卡 才會發生的狀況。
> 能夠穩定復現。
> ＶＭ的網卡若是選別種就不會發生問題。

感谢，这个信息很有用，我将调查这个 bug

ptpt52, 2025-02-14:
是有拨号（pppoe） 才出现宕机吗？

ptpt52, 2025-02-14:
麻烦执行一下 ethtool -i eth0
ethtool -i eth1

另外主机host上的网卡执行 关闭tso 看看，具体如何关闭tso 去google一下对应系统如何操作

ptpt52, 2025-02-14:
另外你kvm，试试把vm网卡类型改成 e1000 看看