x-wrt/x-wrt #380: Cannot access LAN or the internet when connected to the OpenVPN

bachvnnvn, 2024-03-31:
I am using this build for x86: Powered by [LuCI Master (git-24.081.29684-bb81d65)](https://github.com/openwrt/luci) / [X-WRT-docker 24.01_b202403222222 (r25789+1-f1bfb87ec16)](https://x-wrt.com/)

I enabled the OpenVPN Server and from my phone I can connect to my router using 4G.
But I cannot access the interret or LAN machine.

Could you guide me where should I check?

Thanks!

ptpt52, 2024-03-31:
hei, you enable openVPN by `One Key VPN`? try it.

bachvnnvn, 2024-04-01:
Hi,

Yes, it is enabled.

[image]
[image]
[image]

bachvnnvn, 2024-04-01:
And I am using this image: https://downloads.x-wrt.com/rom/x-wrt-docker-24.01-b202403222222-x86-64-generic-ext4-combined-efi.img.gz

Thank you!

ptpt52, 2024-04-01:
why u disable natcapovpn_masq_tcp?
and I think every is woring if u enable VPN via `one key vpn`.

bachvnnvn, 2024-04-01:
Oh, yes, my bad. I am using OpenVPN over UDP only, so I accidentally turned off the natcapovpn_masq_tcp.
I turned it on and it is working fine now.

Thank you very much!

1 more question. I pulled the Ubuntu docker but there is no internet inside the docker container.
Do I need to change any setting?

Thanks again!

closed 2024-04-01 by bachvnnvn